intel/media-driver #1258: Unexpected h264 FHD encoding perf for i7-10510U (UHD 620) vs i5-1135G7 (Iris Xe) -> 372fps vs 276fps

nach, 2021-09-06:
## System information
- CPU information: Intel(R) Core(TM) i7-10510U CPU @ 1.80GHz; 11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz vs 
- GPU information:
00:02.0 VGA compatible controller [0300]: Intel Corporation TigerLake-LP GT2 [Iris Xe Graphics] [8086:9a49] (rev 01)
00:02.0 VGA compatible controller [0300]: Intel Corporation CometLake-U GT2 [UHD Graphics] [8086:9b41] (rev 02)
- Display server if rendering to display(X or wayland): X for i7-10510, Wayland for i5-1135g7

## Issue behavior

Hi, I was doing some encoding benchmark on these 2 platforms expecting 11th
gen i7-1135G7 (Lenovo T14 Gen2) to beat 10th gen i7-10510U (Dell XPS13 7390)
thanks to the Iris Xe GPU but it seems not the case.

For the benchmark I used the libva-utils `h264encode` tools packed from
archlinux libva-utils (v 2.12.0-1) with the following command and result:

## i7-10510U

```
❯ sudo intel_gpu_time h264encode -w 1920 -h 1080 -f 30 --bitrate=5000000 --rcmode=CBR --srcyuv=ref_1920x1080_1800.nv12 --fourcc=NV12 -o out.h264 --low_power 0 -n0 --profile BP --syncmode
Source YUV file ref_1920x1080_1800.nv12 with 1800 frames
Source frame is 1920x1080 and will code clip to 1920x1088 with crop

INPUT:Try to encode H264...
INPUT: RateControl  : CBR
INPUT: Resolution   : 1920x1080, 1800 frames
INPUT: FrameRate    : 30
INPUT: Bitrate      : 5000000
INPUT: Slieces      : 1
INPUT: IntraPeriod  : 30
INPUT: IDRPeriod    : 60
INPUT: IpPeriod     : 1
INPUT: Initial QP   : 26
INPUT: Min QP       : 0
INPUT: Source YUV   : FILE:ref_1920x1080_1800.nv12 (fourcc NV12)
INPUT: Coded Clip   : out.h264
INPUT: Rec   Clip   : Not save reconstructed frame

error: XDG_RUNTIME_DIR not set in the environment.
error: can't connect to X server!
Using EntryPoint - 6 
Use profile VAProfileH264ConstrainedBaseline
Support rate control mode (0xcde):CBR VBR VCM CQP 
Support VAConfigAttribEncPackedHeaders
Support packed sequence headers
Support packed picture headers
Support packed slice headers
Support packed misc headers
Support VAConfigAttribEncInterlaced
Support VA_ENC_INTERLACED_FIELD
Support 8 RefPicList0 and 2 RefPicList1
Support 256 slices
Support VAConfigAttribEncSliceStructure
Support VA_ENC_SLICE_STRUCTURE_ARBITRARY_MACROBLOCKS
      \00001799(021820 bytes coded)

PERFORMANCE:   Frame Rate           : 270.35 fps (1800 frames, 6658 ms (3.70 ms per frame))
PERFORMANCE:   Compression ratio    : 148:1
PERFORMANCE:     UploadPicture      : 1801 ms (1.00, 27.05% percent)
PERFORMANCE:     vaBeginPicture     : 6 ms (0.00, 0.09% percent)
PERFORMANCE:     vaRenderHeader     : 23 ms (0.01, 0.35% percent)
PERFORMANCE:     vaEndPicture       : 711 ms (0.40, 10.68% percent)
PERFORMANCE:     vaSyncSurface      : 3965 ms (2.20, 59.55% percent)
PERFORMANCE:     SavePicture        : 118 ms (0.07, 1.77% percent)
PERFORMANCE:     Others             : 34 ms (0.02, 0.51% percent)
user: 1.963729s, sys: 0.835289s, elapsed: 6.704250s, CPU: 41.7%, GPU: 35.8%

````

without `--syncmode`
[... 241 lines cut ...]
      VAProfileVP9Profile2            :	VAEntrypointEncSliceLP
      VAProfileVP9Profile3            :	VAEntrypointVLD
      VAProfileVP9Profile3            :	VAEntrypointEncSliceLP
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointEncSlice
      VAProfileHEVCMain422_10         :	VAEntrypointVLD
      VAProfileHEVCMain422_10         :	VAEntrypointEncSlice
      VAProfileHEVCMain422_12         :	VAEntrypointVLD
      VAProfileHEVCMain422_12         :	VAEntrypointEncSlice
      VAProfileHEVCMain444            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         :	VAEntrypointVLD
      VAProfileHEVCMain444_10         :	VAEntrypointEncSliceLP
      VAProfileHEVCMain444_12         :	VAEntrypointVLD
      VAProfileHEVCSccMain            :	VAEntrypointVLD
      VAProfileHEVCSccMain            :	VAEntrypointEncSliceLP
      VAProfileHEVCSccMain10          :	VAEntrypointVLD
      VAProfileHEVCSccMain10          :	VAEntrypointEncSliceLP
      VAProfileHEVCSccMain444         :	VAEntrypointVLD
      VAProfileHEVCSccMain444         :	VAEntrypointEncSliceLP
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCSccMain444_10      :	VAEntrypointVLD
      VAProfileHEVCSccMain444_10      :	VAEntrypointEncSliceLP

```

XinfengZhang, 2021-10-17:
hi @nach ,  
for the performance , I suppose the most important part of the perf data is frequency. could you share the frequency data  of each platform when you run the encode command?

another issue is about low_power mode, you modified the h264encode.c to replace VAEntrypointEncSlice to VAEntrypointEncSliceLP?

OttoYang, 2023-03-28:
@nach Since it's been too long without a response. So, if you still have this issue?