intel/media-driver #1259: [regression] libva conformance failures since #1229

uartie, 2021-09-07:
## System information
- Impacted GPU platform:  <= Gen9 (CML, CFL, KBL, SKL,...)
- Host machine if it's descreted card: NA
- Good libva/libva-utils/gmmlib/media-driver version? 91cf6de96b87c30c2a4d340897ee967963812478
- Bad libva/libva-utils/gmmlib/media-driver version? 4ce77a648065205711f16dccd374ab83464aa911

## Issue behavior
### Describe the current behavior
Since PR #1229, libva conformance tests fail...

```
commit 4ce77a648065205711f16dccd374ab83464aa911
Author: Yuankun Zhang <yuankunx.zhang@intel.com>
Date:   Thu Jul 22 10:44:44 2021 +0800

    [VP] Fix loss of Y210 and Y410 format in vaQuerySurfaceAttributes.
    
    Fix loss of Y210 and Y410 format in vaQuerySurfaceAttributes.
    
    Signed-off-by: Yuankun Zhang <yuankunx.zhang@intel.com>
```

```
$ test_va_api
<snip>
==========] 17415 tests from 14 test cases ran. (33675 ms total)
[  PASSED  ] 17381 tests.
[  FAILED  ] 34 tests, listed below:
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesWithConfigAttribs/136, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 16x16)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesWithConfigAttribs/137, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 32x32)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesWithConfigAttribs/138, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 176x120)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesWithConfigAttribs/139, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 352x240)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesWithConfigAttribs/140, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 352x288)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesWithConfigAttribs/141, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 480x320)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesWithConfigAttribs/142, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 720x480)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesWithConfigAttribs/143, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 720x576)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesWithConfigAttribs/144, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 768x576)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesWithConfigAttribs/145, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 800x480)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesWithConfigAttribs/146, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 854x480)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesWithConfigAttribs/147, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 1280x720)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesWithConfigAttribs/148, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 1920x1080)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesWithConfigAttribs/149, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 2048x1080)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesWithConfigAttribs/150, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 4096x2160)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesWithConfigAttribs/151, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 8192x8192)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesWithConfigAttribs/152, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 10240x10240)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesNoConfigAttrib/136, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 16x16)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesNoConfigAttrib/137, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 32x32)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesNoConfigAttrib/138, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 176x120)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesNoConfigAttrib/139, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 352x240)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesNoConfigAttrib/140, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 352x288)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesNoConfigAttrib/141, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 480x320)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesNoConfigAttrib/142, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 720x480)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesNoConfigAttrib/143, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 720x576)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesNoConfigAttrib/144, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 768x576)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesNoConfigAttrib/145, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 800x480)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesNoConfigAttrib/146, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 854x480)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesNoConfigAttrib/147, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 1280x720)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesNoConfigAttrib/148, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 1920x1080)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesNoConfigAttrib/149, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 2048x1080)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesNoConfigAttrib/150, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 4096x2160)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesNoConfigAttrib/151, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 8192x8192)
[  FAILED  ] CreateSurfaces/VAAPICreateSurfaces.CreateSurfacesNoConfigAttrib/152, where GetParam() = (-1:VAProfileNone, 10:VAEntrypointVideoProc, 10240x10240)
```

### Describe the expected behavior

```
$ test_va_api
<snip>
[==========] 17415 tests from 14 test cases ran. (34005 ms total)
[  PASSED  ] 17415 tests.
```

cc: @zhangyuankun-star

zhangyuankun-star, 2021-09-08:
I submitted PR: https://github.com/intel/media-driver/pull/1260 to try to fix this issue.

uartie, 2021-09-08:
fixed by #1260

closed 2021-09-08 by uartie