intel/media-driver #1262: Fails to build

simonbagley, 2021-09-09:
I'm probably making a newby mistake, but the media-driver build is failing at the last step, when I issue the make -j8 command.
I am getting lots of these sort of errors:

```
simon@simon-desktop:~/Documents/workspace/build_media$ make -j8
[  1%] Building CXX object media_driver/linux/ult/libdrm_mock/CMakeFiles/drm_mock.dir/xf86drmHash_mock.c.o
[  1%] Built target iHD_drv_video_SSE2
cc1: warning: command line option ‘-Wreorder’ is valid for C++/ObjC++ but not for C
[  1%] Building CXX object media_driver/linux/ult/libdrm_mock/CMakeFiles/drm_mock.dir/mos_bufmgr_api_mock.c.o
cc1: warning: command line option ‘-Wsign-promo’ is valid for C++/ObjC++ but not for C
cc1: warning: command line option ‘-Wnon-virtual-dtor’ is valid for C++/ObjC++ but not for C
cc1: warning: command line option ‘-Wno-invalid-offsetof’ is valid for C++/ObjC++ but not for C
cc1: warning: command line option ‘-Wno-delete-incomplete’ is valid for C++/ObjC++ but not for C
cc1: warning: ‘-Werror=’ argument ‘-Werror=non-virtual-dtor’ is not valid for C
cc1: warning: command line option ‘-std=c++14’ is valid for C++/ObjC++ but not for C
cc1: warning: command line option ‘-std=c++11’ is valid for C++/ObjC++ but not for C
```

and

```
[  4%] Building CXX object media_driver/linux/ult/libdrm_mock/CMakeFiles/drm_mock.dir/__/__/common/os/mos_vma.c.o
/home/simon/Documents/workspace/media-driver/media_driver/linux/ult/libdrm_mock/xf86drmHash_mock.c: In function ‘drmHashCreate’:
/home/simon/Documents/workspace/media-driver/media_driver/linux/ult/libdrm_mock/xf86drmHash_mock.c:116:24: error: ‘nullptr’ undeclared (first use in this function)
  116 |     if (!table) return nullptr;
      |                        ^~~~~~~
/home/simon/Documents/workspace/media-driver/media_driver/linux/ult/libdrm_mock/xf86drmHash_mock.c:116:24: note: each undeclared identifier is reported only once for each function it appears in
```

XinfengZhang, 2021-09-15:
hi @simonbagley , could you share more information , compiler version, OS version etc.

Sherry-Lin, 2022-01-07:
what's the libva version you used? Could you try with latest libva release 2.13.0?

zrus, 2022-01-07:
> what's the libva version you used? Could you try with latest libva release 2.13.0?

Yes. When I rollback to latest version I could build it successfully. I was using git clone the master of libva so it might lead to failure.

Thank you.

XinfengZhang, 2022-01-07:
latest master should not have such issue. VADisplayAttribCopy is added in this patch https://github.com/intel/libva/pull/513 . hi @zrus, so this issue could be closed?

zrus, 2022-01-07:
> latest master should not have such issue. VADisplayAttribCopy is added in this patch [intel/libva#513](https://github.com/intel/libva/pull/513) . hi @zrus, so this issue could be closed?

It's alright with me. Idk has the OP fixed yet.

closed 2022-01-07 by XinfengZhang