intel/media-driver #1265: VAAPI VP9 Encode on Rocket Lake Crashes - GPU HANG

ajgnet, 2021-09-19:
- 11th Gen Intel(R) Core(TM) i7-11700K @ 3.60GHz
- 00:02.0 VGA compatible controller [0300]: Intel Corporation Device [8086:4c8a] (rev 04)

GPU hangs when HW VP9 encoding using ffmpeg + vp9_vaapi:
```
ffmpeg -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -hwaccel_output_format vaapi -i input.mp4 -c:v vp9_vaapi -b:v 2M -maxrate 2M output.mp4
```

dmesg shows:
```
[15607.512076] i915 0000:00:02.0: [drm] Resetting vcs0 for preemption time out
[15607.512175] i915 0000:00:02.0: [drm] ffmpeg[125068] context reset due to GPU hang
[15607.514726] i915 0000:00:02.0: [drm] GPU HANG: ecode 12:4:28ffff7d, in ffmpeg [125068]
```

The current packages are installed:

- intel-media-va-driver-non-free:amd64 21.1.1+ds1-1
- intel-gpu-tools                      1.25-2.1
- intel-media-va-driver-non-free:amd64 21.1.1+ds1-1
- libdrm-intel1:amd64                  2.4.104-1
- intel-microcode                      3.20210608.2

LIBVA_TRACE shows:

```
[34956.839240][ctx       none]=========vaMapBuffer ret = VA_STATUS_ERROR_ENCODING_ERROR, internal encoding error
```

```
ls /dev/dri
by-path  card0	renderD128
```

```
cat vainfo.log
error: XDG_RUNTIME_DIR not set in the environment.
error: can't connect to X server!
libva info: Open new log file /tmp/libva_trace.log.004941.thd-0x0001fb72 for the thread 0x0001fb72
libva info: LIBVA_TRACE is on, save log into /tmp/libva_trace.log.004941.thd-0x0001fb72
libva info: VA-API version 1.10.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_10
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.10 (libva 2.10.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 21.1.1 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointFEI
      VAProfileHEVCMain               :	VAEntrypointEncSliceLP
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSlice
      VAProfileHEVCMain10             :	VAEntrypointEncSliceLP
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointEncSliceLP
      VAProfileVP9Profile1            :	VAEntrypointVLD
      VAProfileVP9Profile1            :	VAEntrypointEncSliceLP
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointEncSliceLP
      VAProfileVP9Profile3            :	VAEntrypointVLD
      VAProfileVP9Profile3            :	VAEntrypointEncSliceLP
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointEncSlice
      VAProfileHEVCMain422_10         :	VAEntrypointVLD
      VAProfileHEVCMain422_10         :	VAEntrypointEncSlice
      VAProfileHEVCMain422_12         :	VAEntrypointVLD
      VAProfileHEVCMain422_12         :	VAEntrypointEncSlice
      VAProfileHEVCMain444            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         :	VAEntrypointVLD
      VAProfileHEVCMain444_10         :	VAEntrypointEncSliceLP
      VAProfileHEVCMain444_12         :	VAEntrypointVLD
      VAProfileHEVCSccMain            :	VAEntrypointVLD
      VAProfileHEVCSccMain            :	VAEntrypointEncSliceLP
      VAProfileHEVCSccMain10          :	VAEntrypointVLD
      VAProfileHEVCSccMain10          :	VAEntrypointEncSliceLP
      VAProfileHEVCSccMain444         :	VAEntrypointVLD
      VAProfileHEVCSccMain444         :	VAEntrypointEncSliceLP
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCSccMain444_10      :	VAEntrypointVLD
      VAProfileHEVCSccMain444_10      :	VAEntrypointEncSliceLP
```

I have tried kernels 5.10 - 5.14, tried compiling media-driver from source, tried gstreamer and also intel_qsv instead of vaapi. All encoding works for H.264 and H.265 - only VP9 fails. 1 out of 5 times it works successfully, the remaining times it begins and randomly fails.

aidan2020sh, 2021-10-20:
@ajgnet , have you updated the IFWI to the latest? Recently a random GPU hang issue found in HEVC/VP9 Vdenc cases on RKL has been fixed with ww33 or the latest IFWI.

georgnistor, 2021-11-03:
Hi,
I cannot encode at all with vp9_vaapi encoder selected.

[vp9_vaapi @ 0x56388f8c0000] No usable encoding entrypoint found for profile VAProfileVP9Profile0 (19).
Error initializing output stream 0:0 -- Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height

Here is my vainfo output:
george@localhost-live:~> vainfo
libva info: VA-API version 1.13.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib64/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_13
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.13 (libva 2.13.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 21.3.4 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointFEI
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointFEI
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointFEI
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointFEI
      VAProfileHEVCMain               : VAEntrypointEncSliceLP
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSlice
      VAProfileHEVCMain10             : VAEntrypointEncSliceLP
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile1            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileVP9Profile3            : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointEncSlice
      VAProfileHEVCMain422_10         : VAEntrypointVLD
      VAProfileHEVCMain422_10         : VAEntrypointEncSlice
      VAProfileHEVCMain422_12         : VAEntrypointVLD
      VAProfileHEVCMain422_12         : VAEntrypointEncSlice
      VAProfileHEVCMain444            : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         : VAEntrypointVLD
      VAProfileHEVCMain444_10         : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_12         : VAEntrypointVLD
      VAProfileHEVCSccMain            : VAEntrypointVLD
      VAProfileHEVCSccMain            : VAEntrypointEncSliceLP
      VAProfileHEVCSccMain10          : VAEntrypointVLD
      VAProfileHEVCSccMain10          : VAEntrypointEncSliceLP
      VAProfileHEVCSccMain444         : VAEntrypointVLD
      VAProfileHEVCSccMain444         : VAEntrypointEncSliceLP
      VAProfileAV1Profile0            : VAEntrypointVLD
      VAProfileHEVCSccMain444_10      : VAEntrypointVLD
      VAProfileHEVCSccMain444_10      : VAEntrypointEncSliceLP

XinfengZhang, 2021-11-03:
@georgnistor , please share your platform information, huc firmware was not loaded?

aidan2020sh, 2021-11-04:
you can try below commands to check the guc&huc status, maybe the directory has some tiny difference in your system:
sudo cat /sys/kernel/debug/dri/0/i915_guc_load_status
sudo cat /sys/kernel/debug/dri/0/i915_huc_load_status 
if guc/huc is not on, please try adding the paramter "i915.enable_guc=2" when loading i915.

ajgnet, 2021-11-25:
> @ajgnet , have you updated the IFWI to the latest? Recently a random GPU hang issue found in HEVC/VP9 Vdenc cases on RKL has been fixed with ww33 or the latest IFWI. (BIOS is part of IFWI)
Yes: latest IFWI. Does not make a difference. We were forced to upgrade to Alder Lake which runs perfectly.

aidan2020sh, 2021-12-09:
@ajgnet We do not observe such kind of GPU hang issue in internal VP9 encoder test. What you reported looks to be similar with one GPU hang resolved by a IFWI upgrade several months ago but i am not sure whether your IFWI contain the fix.  Are you still going to work on RKL with this issue resolved?  if yes, could you share the IFWI version and the source in your test?

eshiryae, 2022-02-15:
Hi @ajgnet, do you still see this issue? if yes, could you share the IFWI version?

vom22, 2022-03-01:
Hi,
I have the same crashes on Rocket Lake using electron I built with 'use_vaapi=true' option, too.
I can't encode VP9 with '--enable-features VaapiVideoEncoder.'

dmesg shows:
`GPU HANG: ecode 12:4:28ffff7d.`

vainfo shows:
```
VAProfileVP9Profile0 : VAEntrypointVLD
VAProfileVP9Profile0 : VAEntrypointEncSliceLP
VAProfileVP9Profile1 : VAEntrypointVLD
VAProfileVP9Profile1 : VAEntrypointEncSliceLP
VAProfileVP9Profile2 : VAEntrypointVLD
VAProfileVP9Profile2 : VAEntrypointEncSliceLP
VAProfileVP9Profile3 : VAEntrypointVLD
VAProfileVP9Profile3 : VAEntrypointEncSliceLP
```

How can I confirm the IFWI version?

Xiaogangli-intel, 2022-06-16:
Hi @vom22, you can get the IFWI version in BIOS, like:
[image]

And I think you can update to new version BIOS to update IFWI firmware.

vom22, 2022-06-16:
Hi, @Xiaogangli-intel. Thank you for your reply.

I couldn't get IFWI version in BIOS in [H470M-STX](https://download.asrock.com/Manual/H470M-STX.pdf).
Could I confirm the IFWI version in OS, such as linux?

But in either case, the BIOS version has already been the latest.
Should I ask a bios maker support?

Xiaogangli-intel, 2022-06-17:
OK, you can also get it by 
`# dmidecode -s "bios-version"`
I think you can check if have new version BIOS with OEM, they should integrate the latest IFWI firmware in BIOS.

vom22, 2022-06-17:
Thank you.
I checked BIOS version but no IFWI version.
I've already updated the latest BIOS version I can download from ASRock.

Xiaogangli-intel, 2022-06-20:
> Thank you. I checked BIOS version but no IFWI version. I've already updated the latest BIOS version I can download from ASRock.

great!
@vom22, could you update status after updating BIOS? Thanks!

vom22, 2022-06-20:
Even after updating BIOS, I couldn't.

Xiaogangli-intel, 2022-06-20:
could you share your '# dmidecode -s "bios-version"' result?

vom22, 2022-06-20:
```
~$ sudo dmidecode -s "bios-version"
P2.20
```
The BIOS version I downloaded from ASRock is 2.20, too.
https://www.asrock.com/MB/Intel/H470M-STX/index.jp.asp#BIOS

Xiaogangli-intel, 2022-06-20:
Hi @vom22, I tried to ask ASRock for IFWI version at https://event.asrock.com/tsd.asp?ln=en, but I don't have `Serial Number`.
Could you try to get IFWI version in this page? Then I can confirm if Known issue fixes are in the IFWI they are using. If not, I will try to trigger a process to release new version IFWI. Thanks in advance.

vom22, 2022-06-21:
Thank you for your confirmation.
I asdked ASRock for IFWI version at https://event.asrock.com/tsd.asp?ln=en.
As soon as I got an answer, I will share it.

vom22, 2022-06-21:
Hi @Xiaogangli-intel, I got it from ASRock.
They said that 'the IFWI version of the BIOS 2.20 is "14.1.51.1528".'

Xiaogangli-intel, 2022-06-22:
Thanks @vom22, I will check it.

Xiaogangli-intel, 2022-09-30:
@vom22, seems ASRock still doesn't update BIOS. Now it's should not driver issue, it's should a firmware issue.

vom22, 2022-09-30:
@Xiaogangli-intel, thank you for your checking.

If ASRock won't update BIOS forever, can't we encode forever?
Is there any other way?

Xiaogangli-intel, 2022-11-08:
Sorry @vom22, I did some investigation but didn't find other ways to update the firmware. I think you should ask ASRock for help.

vom22, 2022-11-18:
Thank you so much, @Xiaogangli-intel.
Though I asked ASRcok, they said that we should use Windows instead of Linux because they support only Windows.

m911t, 2022-12-26:
This also happens on my intel tigerlake laptop...