intel/media-driver #1270: Intel Tiger Lake, Xe Graphics, vainiliate failed

bbbear2002, 2021-09-30:
## System information
- CPU information(`cat /proc/cpuinfo | grep "model name" | uniq`):
model name      : 11th Gen Intel(R) Core(TM) i5-11300H @ 3.10GHz

- GPU information(`lspci -nn | grep -E 'VGA|isplay`):
00:02.0 VGA compatible controller [0300]: Intel Corporation Device [8086:9a49] (rev 01)

- Display server if rendering to display(X or wayland):
Don't know how

## Issue behavior
Cannot use vaapi

### Describe the current behavior
vainitialize failed

### Describe the expected behavior
vainitialize success

## Debug information
- What's libva/libva-utils/gmmlib/media-driver version?
libva info: VA-API version 1.12.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri//iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_12
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.12 (libva 2.4.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 21.3.4 (18a9bad1)
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointFEI
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointFEI
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointFEI
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointFEI
      VAProfileHEVCMain               : VAEntrypointEncSliceLP
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSlice
      VAProfileHEVCMain10             : VAEntrypointEncSliceLP
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile1            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileVP9Profile3            : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointEncSlice
      VAProfileHEVCMain422_10         : VAEntrypointVLD
      VAProfileHEVCMain422_10         : VAEntrypointEncSlice
      VAProfileHEVCMain422_12         : VAEntrypointVLD
      VAProfileHEVCMain422_12         : VAEntrypointEncSlice
[... 200 lines cut ...]
2021-09-30, 08:51:16.914 [Info ] [gles2_cmd_decoder.cc 3850] : GL_EXT_packed_depth_stencil supported.
2021-09-30, 08:52:14.174 [Info ] [gl_surface_glx.cc 658] : [NPAPI] NativeViewGLSurfaceGLX Initialize XCreateWindow:680001d
2021-09-30, 08:52:16.472 [Warning] [x11_util.cc 1620] : X error received: serial 14550, error_code 170 (GLXBadWindow), request_code 152, minor_code 32 (X_GLXDestroyWindow)
2021-09-30, 08:52:16.472 [Warning] [x11_util.cc 1620] : X error received: serial 14554, error_code 3 (BadWindow (invalid Window parameter)), request_code 4, minor_code 0 (X_DestroyWindow)
2021-09-30, 08:52:34.757 [Info ] [gl_surface_glx.cc 658] : [NPAPI] NativeViewGLSurfaceGLX Initialize XCreateWindow:6800028
2021-09-30, 08:52:59.610 [Info ] [gl_surface_glx.cc 658] : [NPAPI] NativeViewGLSurfaceGLX Initialize XCreateWindow:680002d
2021-09-30, 08:53:01.359 [Warning] [x11_util.cc 1620] : X error received: serial 16796, error_code 170 (GLXBadWindow), request_code 152, minor_code 32 (X_GLXDestroyWindow)
2021-09-30, 08:53:01.359 [Warning] [x11_util.cc 1620] : X error received: serial 16800, error_code 3 (BadWindow (invalid Window parameter)), request_code 4, minor_code 0 (X_DestroyWindow)
2021-09-30, 08:53:21.803 [Info ] [gl_surface_glx.cc 658] : [NPAPI] NativeViewGLSurfaceGLX Initialize XCreateWindow:6800035
2021-09-30, 08:54:44.820 [Info ] [gl_surface_glx.cc 658] : [NPAPI] NativeViewGLSurfaceGLX Initialize XCreateWindow:680003d
2021-09-30, 08:54:46.789 [Warning] [x11_util.cc 1620] : X error received: serial 36131, error_code 170 (GLXBadWindow), request_code 152, minor_code 32 (X_GLXDestroyWindow)
2021-09-30, 08:54:46.789 [Warning] [x11_util.cc 1620] : X error received: serial 36135, error_code 3 (BadWindow (invalid Window parameter)), request_code 4, minor_code 0 (X_DestroyWindow)
2021-09-30, 08:58:19.833 [Info ] [gl_surface_glx.cc 658] : [NPAPI] NativeViewGLSurfaceGLX Initialize XCreateWindow:6800042
2021-09-30, 08:58:21.451 [Warning] [x11_util.cc 1620] : X error received: serial 51818, error_code 170 (GLXBadWindow), request_code 152, minor_code 32 (X_GLXDestroyWindow)
2021-09-30, 08:58:21.451 [Warning] [x11_util.cc 1620] : X error received: serial 51822, error_code 3 (BadWindow (invalid Window parameter)), request_code 4, minor_code 0 (X_DestroyWindow)
2021-09-30, 08:58:40.679 [Info ] [gl_surface_glx.cc 658] : [NPAPI] NativeViewGLSurfaceGLX Initialize XCreateWindow:6800047
2021-09-30, 09:00:14.236 [Info ] [gl_surface_glx.cc 658] : [NPAPI] NativeViewGLSurfaceGLX Initialize XCreateWindow:680004c
2021-09-30, 09:00:15.990 [Warning] [x11_util.cc 1620] : X error received: serial 55762, error_code 170 (GLXBadWindow), request_code 152, minor_code 32 (X_GLXDestroyWindow)
2021-09-30, 09:00:15.990 [Warning] [x11_util.cc 1620] : X error received: serial 55766, error_code 3 (BadWindow (invalid Window parameter)), request_code 4, minor_code 0 (X_DestroyWindow)
2021-09-30, 09:01:55.580 [Info ] [gl_surface_glx.cc 658] : [NPAPI] NativeViewGLSurfaceGLX Initialize XCreateWindow:6800051
2021-09-30, 09:01:57.001 [Warning] [x11_util.cc 1620] : X error received: serial 57319, error_code 170 (GLXBadWindow), request_code 152, minor_code 32 (X_GLXDestroyWindow)
2021-09-30, 09:01:57.002 [Warning] [x11_util.cc 1620] : X error received: serial 57323, error_code 3 (BadWindow (invalid Window parameter)), request_code 4, minor_code 0 (X_DestroyWindow)
2021-09-30, 09:02:33.790 [Info ] [gles2_cmd_decoder.cc 3850] : GL_EXT_packed_depth_stencil supported.
2021-09-30, 09:02:38.062 [Info ] [gles2_cmd_decoder.cc 3850] : GL_EXT_packed_depth_stencil supported.
2021-09-30, 09:02:52.443 [Info ] [gl_surface_glx.cc 658] : [NPAPI] NativeViewGLSurfaceGLX Initialize XCreateWindow:6800064

bbbear2002, 2021-10-09:
I saw in the vainfo that, the libva version is (libva 2.4.0), but i have installed the latest v2.13. Is this caused the issue?

XinfengZhang, 2021-10-17:
and the libva log you shared should just include the vainfo, not from the failed cases of your browser usage

XinfengZhang, 2021-10-17:
duplicate with #1231 ?

Xiaogangli-intel, 2022-09-06:
Close this because duplicate with #1231

closed 2022-09-06 by Xiaogangli-intel