intel/media-driver #1283: Crash on Xe graphics with seccomp

nirik, 2021-10-30:
## System information
- CPU information(`cat /proc/cpuinfo | grep "model name" | uniq`): 
model name      : 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz
- GPU information(`lspci -nn | grep -E 'VGA|isplay`):
00:02.0 VGA compatible controller [0300]: Intel Corporation TigerLake-LP GT2 [Iris Xe Graphics] [8086:9a49] (rev 01)
- Display server if rendering to display(X or wayland):
Wayland
## Issue behavior
### Describe the current behavior

Firefox tabs with videos crash

### Describe the expected behavior

Tabs don't crash

## Debug information
- What's libva/libva-utils/gmmlib/media-driver version?

libva-2.13.0-1.fc36.x86_64
libva-utils-2.13.0-1.fc36.x86_64
intel-gmmlib-21.3.1-1.fc36.x86_64
intel-media-driver-21.3.5-1.fc36.x86_64

- Could you confirm whether GPU hardware exist or not by `ls /dev/dri`?

by-path  card1  renderD128

- Could you provide vainfo log if possible by `vainfo >vainfo.log 2>&1`?

libva info: VA-API version 1.13.0                                                                    
libva info: Trying to open /usr/lib64/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_13
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.13 (libva 2.12.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 21.3.5 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointFEI
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointFEI
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointFEI
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointFEI
      VAProfileHEVCMain               : VAEntrypointEncSliceLP
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSlice
      VAProfileHEVCMain10             : VAEntrypointEncSliceLP
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointEncSliceLP
[... 50 lines cut ...]
bxul.so + 0x1a2ec57)                                                                                 
                #7  0x00007fb80e043d6c _ZN7mozilla18FFmpegVideoDecoderILi58EE4InitEv (libxul.so + 0x1
a2fd6c)                                                                                              
                #8  0x00007fb80e02474f _ZN7mozilla18MediaChangeMonitor4InitEv (libxul.so + 0x1a1074f)
                #9  0x00007fb80fb0fe4f _ZN7mozilla6detail21ProxyFunctionRunnableIZNS_21MediaDataDecod
erProxy4InitEvEUlvE_NS_10MozPromiseINS_9TrackInfo9TrackTypeENS_11MediaResultELb1EEEE3RunEv (libxul.so
 + 0x34fbe4f)                                                                                        
                #10 0x00007fb80f34a589 _ZN7mozilla9TaskQueue6Runner3RunEv (libxul.so + 0x2d36589)    
                #11 0x00007fb80f34a28e _ZN12nsThreadPool3RunEv (libxul.so + 0x2d3628e)               
                #12 0x00007fb80f0f7233 _ZN8nsThread16ProcessNextEventEbPb (libxul.so + 0x2ae3233)    
                #13 0x00007fb80f0f6ce0 _Z19NS_ProcessNextEventP9nsIThreadb (libxul.so + 0x2ae2ce0)   
                #14 0x00007fb80f114c06 _ZN7mozilla3ipc28MessagePumpForNonMainThreads3RunEPN4base11Mes
sagePump8DelegateE (libxul.so + 0x2b00c06)                                                           
                #15 0x00007fb80f4cff1c _ZN11MessageLoop3RunEv (libxul.so + 0x2ebbf1c)                
                #16 0x00007fb80f348eaf _ZN8nsThread10ThreadFuncEPv (libxul.so + 0x2d34eaf)           
                #17 0x00007fb818ba3524 _pt_root (libnspr4.so + 0x2c524)                              
                #18 0x00007fb818c55657 start_thread (libc.so.6 + 0x8d657)

launching firefox with 'MOZ_DISABLE_CONTENT_SANDBOX=1 firefox' and it works. 

- Could you attach dmesg log if it's GPU hang by `dmesg >dmesg.log 2>&1`?
- Do you want to contribute a PR? (yes/no): no, not sure where to

I have no idea if this is a intel-media-driver issue, a libva issue, a firefox issue or a seccomp issue.
Feel free to close and point me to the right place if this isn't it. ;)

eero-t, 2021-11-11:
Seems like media-driver does NULL pointer (method) access, when seccomp sandbox rejects system call "64", and returns error instead.

Instead media-driver should check for the error [1] and return failure back to libva.

Media still won't work though.  That you need to take up with Firefox sandbox authors.

---

[1] Easier way for media-driver devs to experiment with seccomp filters could be using docker, as it allows one to specify which syscalls get rejected: https://docs.docker.com/engine/security/seccomp/

What the underlying container runtime returns for the application may not match what the Firefox sandbox returns: https://github.com/opencontainers/runc/issues/2151

But if they differ, I think either of the sandboxes is in error.  Blocking of syscalls should be done similarly by different sandboxes so that middleware code can act reasonably on the restrictions...

XinfengZhang, 2021-12-05:
suppose the issue is caused by the IPC failure then failed to exit with correct return value. so, I think maybe we need fix both. 
1. could you help to try latest master with commit 4e70cd6? it remove IPC call for TGL. 
2. @cqian2 , could you help to add some protection code to avoid crash?

ps, we have another PR #1293  for older platform

eero-t, 2021-12-07:
First ticket about this seems to be: https://github.com/intel/media-driver/issues/854