intel/media-driver #1284: gmmlib cmake error

bbbear2002, 2021-10-31:
cmake [-DCMAKE_BUILD_TYPE= Release] [-DARCH= 64] ..
-- API version: 11.3.0
-- Package version: 11.3.0
*BUILD_TYPE not defined, default to: release
platform: Linux
source_dir: /home/bbbear2002/gmmlib/Source/GmmLib
arch: 64
build type: Release
SourceConfiguration:
CommonDir: /home/bbbear2002/gmmlib/Source/Common
IncDir: /home/bbbear2002/gmmlib/Source/inc
MOCS table: Static
CMake Error at /usr/share/cmake-3.13/Modules/CheckIncludeFile.cmake:50 (configure_file):
configure_file attempted to configure a file:
/home/bbbear2002/gmmlib/CMakeFiles/CMakeTmp/CheckIncludeFile.c into a
source directory.
Call Stack (most recent call first):
/usr/share/cmake-3.13/Modules/FindThreads.cmake:121 (CHECK_INCLUDE_FILE)
Source/GmmLib/CMakeLists.txt:444 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/bbbear2002/gmmlib/CMakeFiles/CMakeOutput.log".
See also "/home/bbbear2002/gmmlib/CMakeFiles/CMakeError.log".

bbbear2002, 2021-10-31:
Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND
Build flags:
Id flags:

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND
Build flags:
Id flags: -c

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND
Build flags:
Id flags: --c++

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND
Build flags:
Id flags: --ec++

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND
Build flags:
Id flags:

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND
Build flags:
Id flags: -c

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND
Build flags:
Id flags: --c++

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND
Build flags:
Id flags: --ec++

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND
Build flags:
Id flags:

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND
Build flags:
Id flags: -c

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND
Build flags:
Id flags: --c++

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND
Build flags:
Id flags: --ec++

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND
Build flags:
Id flags:

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND
Build flags:
Id flags: -c

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND
Build flags:
Id flags: --c++

The output was:
No such file or directory

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND
Build flags:
Id flags: --ec++

The output was:
No such file or directory

bbbear2002, 2021-10-31:
The system is: Linux - 5.12.14-amd64-desktop - x86_64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /usr/bin/cc
Build flags:
Id flags:

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/home/bbbear2002/gmmlib/CMakeFiles/3.13.4/CompilerIdC/a.out"

Determining if the C compiler works passed with the following output:
Change Dir: /home/bbbear2002/gmmlib/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_e63b3/fast"
/usr/bin/make -f CMakeFiles/cmTC_e63b3.dir/build.make CMakeFiles/cmTC_e63b3.dir/build
make[1]: 进入目录“/home/bbbear2002/gmmlib/CMakeFiles/CMakeTmp”
Building C object CMakeFiles/cmTC_e63b3.dir/testCCompiler.c.o
/usr/bin/cc -o CMakeFiles/cmTC_e63b3.dir/testCCompiler.c.o -c /home/bbbear2002/gmmlib/CMakeFiles/CMakeTmp/testCCompiler.c
Linking C executable cmTC_e63b3
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_e63b3.dir/link.txt --verbose=1
/usr/bin/cc -rdynamic CMakeFiles/cmTC_e63b3.dir/testCCompiler.c.o -o cmTC_e63b3
make[1]: 离开目录“/home/bbbear2002/gmmlib/CMakeFiles/CMakeTmp”

Detecting C compiler ABI info compiled with the following output:
Change Dir: /home/bbbear2002/gmmlib/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_ab6a1/fast"
/usr/bin/make -f CMakeFiles/cmTC_ab6a1.dir/build.make CMakeFiles/cmTC_ab6a1.dir/build
make[1]: 进入目录“/home/bbbear2002/gmmlib/CMakeFiles/CMakeTmp”
Building C object CMakeFiles/cmTC_ab6a1.dir/CMakeCCompilerABI.c.o
/usr/bin/cc -o CMakeFiles/cmTC_ab6a1.dir/CMakeCCompilerABI.c.o -c /usr/share/cmake-3.13/Modules/CMakeCCompilerABI.c
Linking C executable cmTC_ab6a1
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_ab6a1.dir/link.txt --verbose=1
/usr/bin/cc -v -rdynamic CMakeFiles/cmTC_ab6a1.dir/CMakeCCompilerABI.c.o -o cmTC_ab6a1
Using built-in specs.
COLLECT_GCC=/usr/bin/cc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/8/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Uos 8.3.0.3-3+rebuild' --with-bugurl=file:///usr/share/doc/gcc-8/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-8 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 8.3.0 (Uos 8.3.0.3-3+rebuild)
COMPILER_PATH=/usr/lib/gcc/x86_64-linux-gnu/8/:/usr/lib/gcc/x86_64-linux-gnu/8/:/usr/lib/gcc/x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/8/:/usr/lib/gcc/x86_64-linux-gnu/
LIBRARY_PATH=/usr/lib/gcc/x86_64-linux-gnu/8/:/usr/lib/gcc/x86_64-linux-gnu/8/../../../x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/8/../../../../lib/:/lib/x86_64-linux-gnu/:/lib/../lib/:/usr/lib/x86_64-linux-gnu/:/usr/lib/../lib/:/usr/lib/gcc/x86_64-linux-gnu/8/../../../:/lib/:/usr/lib/
COLLECT_GCC_OPTIONS='-v' '-rdynamic' '-o' 'cmTC_ab6a1' '-mtune=generic' '-march=x86-64'
/usr/lib/gcc/x86_64-linux-gnu/8/collect2 -plugin /usr/lib/gcc/x86_64-linux-gnu/8/liblto_plugin.so -plugin-opt=/usr/lib/gcc/x86_64-linux-gnu/8/lto-wrapper -plugin-opt=-fresolution=/tmp/ccOCIUpd.res -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_s --build-id --eh-frame-hdr -m elf_x86_64 --hash-style=gnu -export-dynamic -dynamic-linker /lib64/ld-linux-x86-64.so.2 -o cmTC_ab6a1 /usr/lib/gcc/x86_64-linux-gnu/8/../../../x86_64-linux-gnu/crt1.o /usr/lib/gcc/x86_64-linux-gnu/8/../../../x86_64-linux-gnu/crti.o /usr/lib/gcc/x86_64-linux-gnu/8/crtbegin.o -L/usr/lib/gcc/x86_64-linux-gnu/8 -L/usr/lib/gcc/x86_64-linux-gnu/8/../../../x86_64-linux-gnu -L/usr/lib/gcc/x86_64-linux-gnu/8/../../../../lib -L/lib/x86_64-linux-gnu -L/lib/../lib -L/usr/lib/x86_64-linux-gnu -L/usr/lib/../lib -L/usr/lib/gcc/x86_64-linux-gnu/8/../../.. CMakeFiles/cmTC_ab6a1.dir/CMakeCCompilerABI.c.o -lgcc --push-state --as-needed -lgcc_s --pop-state -lc -lgcc --push-state --as-needed -lgcc_s --pop-state /usr/lib/gcc/x86_64-linux-gnu/8/crtend.o /usr/lib/gcc/x86_64-linux-gnu/8/../../../x86_64-linux-gnu/crtn.o
COLLECT_GCC_OPTIONS='-v' '-rdynamic' '-o' 'cmTC_ab6a1' '-mtune=generic' '-march=x86-64'
make[1]: 离开目录“/home/bbbear2002/gmmlib/CMakeFiles/CMakeTmp”

Parsed C implicit link information from above output:
link line regex: [^( |.[/])(ld|CMAKE_LINK_STARTFILE-NOTFOUND|([^/]+-)?ld|collect2)[^/\]*( |$)]
ignore line: [Change Dir: /home/bbbear2002/gmmlib/CMakeFiles/CMakeTmp]
ignore line: []
ignore line: [Run Build Command:"/usr/bin/make" "cmTC_ab6a1/fast"]
ignore line: [/usr/bin/make -f CMakeFiles/cmTC_ab6a1.dir/build.make CMakeFiles/cmTC_ab6a1.dir/build]
[... cut ...]
Feature record: CXX_FEATURE:1cxx_template_template_parameters
Feature record: CXX_FEATURE:0cxx_thread_local
Feature record: CXX_FEATURE:0cxx_trailing_return_types
Feature record: CXX_FEATURE:0cxx_unicode_literals
Feature record: CXX_FEATURE:0cxx_uniform_initialization
Feature record: CXX_FEATURE:0cxx_unrestricted_unions
Feature record: CXX_FEATURE:0cxx_user_literals
Feature record: CXX_FEATURE:0cxx_variable_templates
Feature record: CXX_FEATURE:0cxx_variadic_macros
Feature record: CXX_FEATURE:0cxx_variadic_templates

XinfengZhang, 2021-11-01:
sorry, but why not file the issue against https://github.com/intel/gmmlib?

XinfengZhang, 2021-11-01:
@johnmach

Sherry-Lin, 2021-11-29:
Build error when compiling GmmLib, submitter moved to https://github.com/intel/gmmlib/issues/90, so close this issue.

closed 2021-11-29 by Sherry-Lin