intel/media-driver #1287: Webkit fails to play the video correctly.

HeJunyan, 2021-11-11:
## System information
- Impacted GPU platform: Mesa Intel® HD Graphics 620 (KBL GT2)

## Issue behavior
### Describe the current behavior
The video output has garbage and artifact.

### Describe the expected behavior
Should play the video well.

You can see
https://gitlab.freedesktop.org/gstreamer/gstreamer/-/issues/877#note_1154550
for more details.

21.3.2-1 is OK but 21.3.5-1 is not good.

HeJunyan, 2021-11-11:
It's a bug found by the gstreamer community.

Sherry-Lin, 2021-11-29:
Is it related to format support issue in gst-inspect-1.0 vaapipostproc? Could you provide the cmdline to reproduce this issue?

Jexu, 2021-12-08:
>>>21.3.2-1 is OK but 21.3.5-1 is not good.
It is umd driver regression?

HeJunyan, 2021-12-08:
Please wait for more info from the community, I just get it from the community but I can not produce it locally. We need other maintainers root cause it.

Jexu, 2022-12-02:
So far, any update from community?

HeJunyan, 2022-12-02:
I think we can close it now.

closed 2022-12-02 by HeJunyan