intel/media-driver #1288: guys! I need someone help me! I use media-driver in vainfo got this: Sku/Wa/GtSystemInfo initialization!~

JohyC, 2021-11-13:
## System information
- CPU information(`cat /proc/cpuinfo | grep "model name" | uniq`):
- GPU information(`lspci -nn | grep -E 'VGA|isplay`):
- Display server if rendering to display(X or wayland):

CPU : model name      : Intel(R) Core(TM) i7-4785T CPU @ 2.20GHz
GPU: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor Integrated Graphics Controller
LINUX: Debian GNU/Linux 11 (bullseye) x86_64

## Issue behavior
### Describe the current behavior

vainfo cann`t use iHD_drv_video!

in debug use media driver, i got error;
Failed to get the device info for Device

[LIBVA]:CRITICAL - DdiMedia_InitMediaContext:1868: Fatal error - unsuccesfull Sku/Wa/GtSystemInfo initialization
libva error: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 1
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_8
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.14 (libva 2.10.0)
vainfo: Driver version: Intel i965 driver for Intel(R) Haswell Desktop - 2.4.1`

### Describe the expected behavior

## Debug information
- What's libva/libva-utils/gmmlib/media-driver version?
 I use the latest git repository to compile; maybe release 21.3.5 ?
- Could you confirm whether GPU hardware exist or not by `ls /dev/dri`?
I got this
 ` ~ ツ ls /dev/dri
by-path  card0  renderD128`

- Could you provide vainfo log if possible by `vainfo >vainfo.log 2>&1`?

[vainfo.log](https://github.com/intel/media-driver/files/7531822/vainfo.log)

- Could you provide libva trace log if possible? Run cmd `export LIBVA_TRACE=/tmp/libva_trace.log` first then execute the case.

sorry! Ican`t understand this...

- Could you attach dmesg log if it's GPU hang by `dmesg >dmesg.log 2>&1`?

[dmesg.log](https://github.com/intel/media-driver/files/7531838/dmesg.log)

- Do you want to contribute a PR? (yes/no): 
 no,I don't have this level！

XinfengZhang, 2021-11-15:
sorry about it, but media-driver start the support from BDW, your platform is HSW which is not in please refer https://github.com/intel/media-driver#supported-platforms
please use i965 driver https://github.com/intel/intel-vaapi-driver , thanks

JohyC, 2021-11-15:
> sorry about it, but media-driver start the support from BDW, your platform is HSW which is not in please refer https://github.com/intel/media-driver#supported-platforms please use i965 driver https://github.com/intel/intel-vaapi-driver , thanks

😭😭😭i got it. thanks you reply for this ques.

closed 2021-11-15 by JohyC