intel/media-driver #1294: Accesses to `/etc/libva.conf` and `/etc/igfx_user_feature.txt`

paulmenzel, 2021-11-23:
Using Debian sid/unstable intel-media-va-driver-non-free:amd64 21.4.1+ds1-1, and Mozilla Firefox (Nightly) with VA-API, running `sudo opensnoop.bt`, it shows a lot of accesses to `/etc/libva.conf` and `/etc/igfx_user_feature.txt`.

```
$ sudo opensnoop.bt
Attaching 6 probes...
Tracing open syscalls... Hit Ctrl-C to end.
PID    COMM               FD ERR PATH
6287   DNS Res~ver #33   168   0 /etc/hosts
414    systemd-resolve    -1   2 /run/systemd/netif/links/2
414    systemd-resolve    -1   2 /run/systemd/netif/links/2
414    systemd-resolve    -1   2 /run/systemd/netif/links/2
414    systemd-resolve    -1   2 /run/systemd/netif/links/2
6287   FS Broker 9185     -1   2 /etc/libva.conf
6287   FS Broker 9185     -1   2 /etc/libva.conf
6287   FS Broker 9185     -1   2 /etc/libva.conf
6287   FS Broker 9185     -1   2 /etc/libva.conf
6287   FS Broker 9185     -1   2 /etc/libva.conf
6287   FS Broker 9185     -1   2 /etc/libva.conf
6287   FS Broker 9185     -1   2 /etc/libva.conf
6287   FS Broker 9185     -1   2 /etc/libva.conf
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/libva.conf
6287   FS Broker 9185     -1   2 /etc/libva.conf
6287   FS Broker 9185     -1   2 /etc/libva.conf
6287   FS Broker 9185     -1   2 /etc/libva.conf
6287   FS Broker 9185     -1   2 /etc/libva.conf
6287   FS Broker 9185     -1   2 /etc/libva.conf
6287   FS Broker 9185     -1   2 /etc/libva.conf
6287   FS Broker 9185     -1   2 /etc/libva.conf
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/libva.conf
6287   FS Broker 9185     -1   2 /etc/libva.conf
6287   FS Broker 9185     -1   2 /etc/libva.conf
6287   FS Broker 9185     -1   2 /etc/libva.conf
6287   FS Broker 9185     -1   2 /etc/libva.conf
6287   FS Broker 9185     -1   2 /etc/libva.conf
6287   FS Broker 9185     -1   2 /etc/libva.conf
6287   FS Broker 9185     -1   2 /etc/libva.conf
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
6287   FS Broker 9185     -1   2 /etc/igfx_user_feature.txt
```

According to [Video Processing Debug Tool](https://github.com/intel/media-driver/wiki/Video-Processing-Debug-Tool), at least `/etc/igfx_user_feature.txt`, is mainly used for debugging and tracing. As I am using the release version, should the files be queried that often.

XinfengZhang, 2021-12-05:
@pingliwa , suppose it is an requirement for our next step refactor

pingliwa, 2021-12-06:
It accesses the /etc/igfx_user_feature.txt twice for each media context, one is to open for read at the beginning of the media context creation, and the other is to open for write at the end of the media context termination.
Does it run in multiple processes? Does it create and destroy media context multiple times in one process?

paulmenzel, 2021-12-06:
Thank you for your responses:

> and the other is to open for write at the end of the media context termination

Why is that? What is written to it?

> Does it run in multiple processes? Does it create and destroy media context multiple times in one process?

Sorry, no idea. With Mozilla Firefox/Nightly 96.0a1 (20211204215420) and I am currently not able to reproduce that anymore, when playing one video from YouTube (H264). Maybe beforehand there was some advertisement video played in some tab.

But, both of the files are not present on my system and not documented on my system. Reading the Wiki, as noted in my original report, both files shouldn’t be needed for systems using the normal Intel media driver releases?

pingliwa, 2021-12-06:
>>>Why is that? What is written to it?
It is a configure file  mainly for debug for example the [Video-Processing-Debug-Tool](https://github.com/intel/media-driver/wiki/Video-Processing-Debug-Tool). 

Write back the driver status.

>>>>for systems using the normal Intel media driver releases?
Do not need it for the normal Intel media driver releases.

paulmenzel, 2021-12-06:
> > Why is that? What is written to it?
> 
> It is a configure file mainly for debug for example the [Video-Processing-Debug-Tool](https://github.com/intel/media-driver/wiki/Video-Processing-Debug-Tool).

Yes, that is what I wrote.

> Write back the driver status.

Well, if it’s not there in the beginning, can code be added to only write to it, if it was there in the beginning.

> > for systems using the normal Intel media driver releases?
> 
> Do not need it for the normal Intel media driver releases.

Can an environment variable be added to enable the debug mode, and if it’s not set, do not access these files at all?

Jexu, 2022-12-02:
Could you try latest media driver since it should be refined now.

XinfengZhang, 2023-05-08:
@pingliwa , it should be a good idea to have a mode selection block the write operation in registry file, and was disabled by default. then there will no such issue anymore.