intel/media-driver #1295: segfault error 4 in iHD_drv_video.so

fti-ahande, 2021-11-23:
## System information
- CPU information: 11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz
- GPU information: Intel Corporation TigerLake-LP GT2 [Iris Xe Graphics] [8086:9a49] (rev 01)
- Display server: X

## Issue
iHD driver segfault while running my gstreamer pipeline (vaapih264enc)
```
[1072319.913878] queue1:src[8162]: segfault at 7f313c23a000 ip 00007f318b49e79c sp 00007f315df5e170 error 4 in iHD_drv_video.so[7f318af4c000+20f2000]
```
Pipline as follows:
```
'v4l2src' 'device=/dev/video0' 'do-timestamp=true' '!' 'video/x-raw,width=1280,height=720' '!' 'videoconvert' '!' 'video/x-raw,format=I420' '!' 'videorate' '!' 'video/x-raw,framerate=15/1' '!' 'queue' 'leaky=downstream' '!' 'videoconvert' '!' 'video/x-raw,format=NV12' '!' 'vaapih264enc' 'rate-control=1' 'bitrate=1036' 'keyframe-period=45' '!' 'video/x-h264,profile=constrained-baseline' '!' 'h264parse' '!' 'video/x-h264,stream-format=avc' '!' 'mp4mux' '!' 'filesink' 'location=./out.mp4'
```

## Debug information
- gst-inspect vaapi
```
Plugin Details:
  Name                     vaapi
  Description              VA-API based elements
  Filename                 /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstvaapi.so
  Version                  1.16.2
  License                  LGPL
  Source module            gstreamer-vaapi
  Source release date      2019-12-03
  Binary package           gstreamer-vaapi
  Origin URL               http://bugzilla.gnome.org/enter_bug.cgi?product=GStreamer

  vaapih264feienc: VA-API H264 FEI Advanced encoder (Experimental)
  vaapih264enc: VA-API H264 encoder
  vaapijpegenc: VA-API JPEG encoder
  vaapih265enc: VA-API H265 encoder
  vaapimpeg2enc: VA-API MPEG-2 encoder
  vaapisink: VA-API sink
  vaapidecodebin: VA-API Decode Bin
  vaapipostproc: VA-API video postprocessing
  vaapih265dec: VA-API H265 decoder
  vaapivp9dec: VA-API VP9 decoder
  vaapivp8dec: VA-API VP8 decoder
  vaapivc1dec: VA-API VC1 decoder
  vaapih264dec: VA-API H264 decoder
  vaapimpeg2dec: VA-API MPEG2 decoder
  vaapijpegdec: VA-API JPEG decoder

  15 features:
  +-- 15 elements

```
- What's libva/libva-utils/gmmlib/media-driver version:
```
libva info: VA-API version 1.13.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_13
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.13 (libva 2.13.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 21.3.5 (ec4dc5d6)
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointFEI
      VAProfileHEVCMain               :	VAEntrypointEncSliceLP
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSlice
      VAProfileHEVCMain10             :	VAEntrypointEncSliceLP
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile1            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileVP9Profile3            :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointEncSlice
      VAProfileHEVCMain422_10         :	VAEntrypointVLD
      VAProfileHEVCMain422_10         :	VAEntrypointEncSlice
      VAProfileHEVCMain422_12         :	VAEntrypointVLD
      VAProfileHEVCMain422_12         :	VAEntrypointEncSlice
      VAProfileHEVCMain444            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         :	VAEntrypointVLD
      VAProfileHEVCMain444_10         :	VAEntrypointEncSliceLP
      VAProfileHEVCMain444_12         :	VAEntrypointVLD
      VAProfileHEVCSccMain            :	VAEntrypointVLD
      VAProfileHEVCSccMain            :	VAEntrypointEncSliceLP
      VAProfileHEVCSccMain10          :	VAEntrypointVLD
      VAProfileHEVCSccMain10          :	VAEntrypointEncSliceLP
      VAProfileHEVCSccMain444         :	VAEntrypointVLD
      VAProfileHEVCSccMain444         :	VAEntrypointEncSliceLP
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCSccMain444_10      :	VAEntrypointVLD
      VAProfileHEVCSccMain444_10      :	VAEntrypointEncSliceLP
```
- `ls /dev/dri`
card0  renderD128
- `libva trace`
[libva_trace.log.1211.zip](https://github.com/intel/media-driver/files/7587965/libva_trace.log.1211.zip)
- `dmesg >dmesg.log 2>&1`
[dmesg.log](https://github.com/intel/media-driver/files/7587955/dmesg.log)

fti-ahande, 2021-11-23:
I found below in dmesg:
`uvcvideo: Failed to resubmit video URB`
kernel issue?

schnulller, 2021-11-28:
I had a similar issue (segfault with error 4) and resolved it temporarily by using the non-free version of the driver. I know it's just a workaround, but maybe it helps.

fti-ahande, 2021-11-29:
Thanks @schnulller for your suggestion. I forgot to mention that I have compiled libva and intel-media-driver from source.

Jexu, 2022-12-02:
Could share the reproduce steps, call stack log or crash point in media driver?

zlice, 2022-12-18:
Similar issue with shotcut/mlt7(melt7)

#### sys info

- kernel 6.1-rc8 drm-tip
- intel-media-sdk / libmfx 2.6.4
- intel-media-driver 2.6.4
- intel-gmmlib 22.3.2
- ffmpeg 4.4.3
- mlt7 7.8.0 / 7.12.0
- shotcut beta (doesn't crash, melt7 does)

#### files

[segfault.txt](https://github.com/intel/media-driver/files/10254172/segfault.txt) (command `melt-7 -verbose shotcut-cHoyVM.mlt`)

[shotcut-cHOyVM.mlt.txt](https://github.com/intel/media-driver/files/10254175/shotcut-cHOyVM.mlt.txt)

XinfengZhang, 2023-05-08:
could you help to share more info, such as reproduce method, or call stack in media driver? compile media driver and run again, you will get it with gdb

zlice, 2023-05-08:
fwiw - Believe my issue was fixed by https://github.com/intel/media-driver/issues/1580 https://github.com/intel/media-driver/commit/e68feab7d