intel/media-driver #1297: Missing "releases" for "tags"

Karlson2k, 2021-11-26:
This repository has many [tags](https://github.com/intel/media-driver/tags) without matching [releases](https://github.com/intel/media-driver/releases).
21.4.1, 21.4.2, 21.3.0, 21.3.1, 21.3.2, 21.3.4, 21.3.4 are all tags without releases.
In https://github.com/intel/media-driver/issues/1290 it was mentioned, that code is tagged only after proper testing.

Does it mean that releases are made only after deeper testing? 
Or difference between "tag" and "release" is only additional files, created specially for release, and "tags" and "releases" have similar code quality and level of testing?

mattst88, 2021-12-14:
Ping?

Sherry-Lin, 2021-12-17:
Media-driver supports two release model
- Bi-weekly rolling release, to support some faster integration. That's the `tags` you see in GitHub.
- Quarterly release, OSV is major pick up this one. That's the `releases` you see in GitHub.

Basically these two release qualification cycles are similar but release scope may be different.

mattst88, 2021-12-30:
Thanks, this makes sense. Feel free to close.

closed 2021-12-31 by Karlson2k