intel/media-driver #1298: libva error: /usr/lib/dri/iHD_drv_video.so init failed

D0nnieD4rk0, 2021-11-26:
running an intel chip Intel Celeron N3150 post broadwell (brasswell released in 2015) and i get this with the intel-media-driver on arch
any suggestions on how to make it work ?
thanks in advance

XinfengZhang, 2021-12-05:
please use https://github.com/intel/intel-vaapi-driver, media driver just support BDW and gen9+

eero-t, 2022-02-08:
Which is also clearly documented in the media-driver front page (README.md).  "Supported platforms" section might note that for older HW the old intel-vaapi-driver should be used instead though.

david-morris, 2022-04-05:
I'm actually getting this same error when I try to load the driver on a Jasper Lake chip:
`00:02.0 VGA compatible controller: Intel Corporation JasperLake [UHD Graphics] (rev 01)`
Any ideas on what might not be compatible? I'm using `5.17.1-arch1-1` kernel, `xf86-video-intel 1:2.99`, and `xorg-server 21.1.3-6` X11. My driver version is `intel-media-driver 22.2.2-1`.

mauricioquintela, 2022-04-06:
Exact same error started occuring to me after recent update of `intel-media-driver 22.2.2-1`. On a `00:02.0 VGA compatible controller: Intel Corporation Skylake GT2 [HD Graphics 520] (rev 07)` (i3-6006U), running manjaro with kernel 5.17.1

XinfengZhang, 2022-04-07:
> I'm actually getting this same error when I try to load the driver on a Jasper Lake chip: `00:02.0 VGA compatible controller: Intel Corporation JasperLake [UHD Graphics] (rev 01)` Any ideas on what [...]

could you help to share the vainfo log and the chipID details

XinfengZhang, 2022-04-07:
> i3-6006U

I could not find the spec of i3-6006U in ark.intel.com 
but suppose it is older than BDW,  media-driver start support from BDW, older platform please use i965 https://github.com/intel/intel-vaapi-driver

mauricioquintela, 2022-04-07:
> could you help to share the vainfo log and the chipID details

Running vainfo with `LIBVA_DRIVER_NAME=iHD` results simply in   
```
~ > LIBVA_DRIVER_NAME=iHD vainfo                                                 
libva error: /usr/lib/dri/iHD_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
```
With i965, manually set systemwide via environment variables, 
```
 ~ > vainfo
vainfo: VA-API version: 1.13 (libva 2.14.0)
vainfo: Driver version: Intel i965 driver for Intel(R) Skylake - 2.4.1
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointStats
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointStats
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointStats
      VAProfileH264MultiviewHigh      :	VAEntrypointVLD
      VAProfileH264MultiviewHigh      :	VAEntrypointEncSlice
      VAProfileH264StereoHigh         :	VAEntrypointVLD
      VAProfileH264StereoHigh         :	VAEntrypointEncSlice
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
```

Regarding chipID, do you mean the output of `lscpu`? 
```
 ~ > lscpu
Architecture:            x86_64
  CPU op-mode(s):        32-bit, 64-bit
  Address sizes:         39 bits physical, 48 bits virtual
  Byte Order:            Little Endian
CPU(s):                  4
  On-line CPU(s) list:   0-3
Vendor ID:               GenuineIntel
  Model name:            Intel(R) Core(TM) i3-6006U CPU @ 2.00GHz
    CPU family:          6
    Model:               78
    Thread(s) per core:  2
    Core(s) per socket:  2
    Socket(s):           1
    Stepping:            3
    CPU(s) scaling MHz:  44%
    CPU max MHz:         2000,0000
    CPU min MHz:         400,0000
    BogoMIPS:            4001.60
    Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscal
                         l nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulq
                         dq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer a
                         es xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single ssbd ibrs ibpb stibp tpr_shadow vnmi flexpri
                         ority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xs
                         avec xgetbv1 xsaves dtherm arat pln pts hwp hwp_notify hwp_act_window hwp_epp md_clear flush_l1d arch_capabilities
Virtualization features: 
  Virtualization:        VT-x
Caches (sum of all):     
  L1d:                   64 KiB (2 instances)
  L1i:                   64 KiB (2 instances)
  L2:                    512 KiB (2 instances)
  L3:                    3 MiB (1 instance)
NUMA:                    
  NUMA node(s):          1
  NUMA node0 CPU(s):     0-3
Vulnerabilities:         
  Itlb multihit:         KVM: Mitigation: VMX disabled
  L1tf:                  Mitigation; PTE Inversion; VMX vulnerable
  Mds:                   Vulnerable; SMT vulnerable
  Meltdown:              Vulnerable
  Spec store bypass:     Vulnerable
  Spectre v1:            Vulnerable: __user pointer sanitization and usercopy barriers only; no swapgs barriers
  Spectre v2:            Vulnerable, IBPB: disabled, STIBP: disabled
  Srbds:                 Vulnerable
  Tsx async abort:       Not affected
```

mauricioquintela, 2022-04-07:
> > i3-6006U
> 
> I could not find the spec of i3-6006U in ark.intel.com but suppose it is older than BDW, media-driver start support from BDW, older platform please use i965 https://github.com/intel/intel-vaapi-driver

I was using iHD because, as mentioned in the documentation, it supports skylake. The ark.intel page for the processor: https://ark.intel.com/content/www/us/en/ark/products/91157/intel-core-i36006u-processor-3m-cache-2-00-ghz.html

XinfengZhang, 2022-04-07:
> I was using iHD because, as mentioned in the documentation, it supports skylake. The ark.intel page for the processor: [...]

thanks, yes, it is SKL, and was in the list
https://github.com/intel/media-driver/blob/master/media_driver/linux/gen9/ddi/media_sysinfo_g9.cpp#L705
will check it , thanks

radupotop, 2022-04-07:
Getting the same error on Kaby Lake i5-8250U

```
> LIBVA_DRIVER_NAME=iHD vainfo 
libva error: /usr/lib/dri/iHD_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit

> LIBVA_DRIVER_NAME=i965 vainfo
vainfo: VA-API version: 1.13 (libva 2.14.0)
vainfo: Driver version: Intel i965 driver for Intel(R) Kaby Lake - 2.4.1
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
...
```

* Linux 5.17.1-arch1-1 SMP PREEMPT Mon, 28 Mar 2022 20:55:33 +0000 x86_64 GNU/Linux
* intel-media-driver 22.2.2-1
* libva-intel-driver 2.4.1-2
* xorg-server 21.1.3-6

Happy to provide more info if needed.

david-morris, 2022-04-08:
> could you help to share the vainfo log and the chipID details

@XinfengZhang I wasn't able to find any log file for vainfo, but here's a terminal session showing a few different invocations:
```
[dmorris@tinkerbell:~]$ LIBVA_DRIVER_NAME=i965 vainfo                  (04-08 21:06)
vaInitialize failed with error code -1 (unknown libva error),exit
[dmorris@tinkerbell:~][3]$ LIBVA_DRIVER_NAME=iHD vainfo                (04-08 21:06)
libva error: /usr/lib/dri/iHD_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
[dmorris@tinkerbell:~][3]$  vainfo                                     (04-08 21:06)
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
vaInitialize failed with error code -1 (unknown libva error),exit
```
I don't actually know what chipID is. Here's all the data I can find about my integrated graphics and processor:
```
sudo lspci -kvs 00:02.0
[sudo] password for dmorris: 
00:02.0 VGA compatible controller: Intel Corporation JasperLake [UHD Graphics] (rev 01) (prog-if 00 [VGA controller])
        DeviceName: Onboard - Video
        Subsystem: ASUSTeK Computer Inc. Device 8813
        Flags: bus master, fast devsel, latency 0, IRQ 136
        Memory at 6000000000 (64-bit, non-prefetchable) [size=16M]
        Memory at 4000000000 (64-bit, prefetchable) [size=256M]
        I/O ports at 4000 [size=64]
        Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
        Capabilities: [40] Vendor Specific Information: Len=0c <?>
        Capabilities: [70] Express Root Complex Integrated Endpoint, MSI 00
        Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable+ 64bit-
        Capabilities: [d0] Power Management version 2
        Capabilities: [100] Process Address Space ID (PASID)
        Capabilities: [200] Address Translation Service (ATS)
        Capabilities: [300] Page Request Interface (PRI)
        Kernel driver in use: i915
        Kernel modules: i915
[dmorris@tinkerbell:~]$ lscpu                                                                                                                               (04-08[0/4578]
Architecture:            x86_64                                                                                                                                           
  CPU op-mode(s):        32-bit, 64-bit                                                                                                                                   
  Address sizes:         39 bits physical, 48 bits virtual                                                                                                                
  Byte Order:            Little Endian                                                                                                                                    
CPU(s):                  4                                                                                                                                                
  On-line CPU(s) list:   0-3                                                                                                                                              
Vendor ID:               GenuineIntel                                                                                                                                     
  Model name:            Intel(R) Celeron(R) N5100 @ 1.10GHz                                                                                                              
    CPU family:          6                                                                                                                                                
    Model:               156                                                                                                                                              
    Thread(s) per core:  1                                                                                                                                                
    Core(s) per socket:  4                                                                                                                                                
    Socket(s):           1                                                                                                                                                
    Stepping:            0                                                                                                                                                
    CPU(s) scaling MHz:  28%                                                                                                                                              
    CPU max MHz:         2800.0000                                                                                                                                        
    CPU min MHz:         800.0000                                                                                                                                         
    BogoMIPS:            2228.00                                                                                                                                          
    Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm cons
[... cut ...]
  Spectre v1:            Mitigation; usercopy/swapgs barriers and __user pointer sanitization                                                                             
  Spectre v2:            Mitigation; Enhanced IBRS, IBPB conditional, RSB filling                                                                                         
  Srbds:                 Not affected                                                                                                                                     
  Tsx async abort:       Not affected 
```
Thanks for reaching out.  Let me know if there's other commands or log files I should post.

david-morris, 2022-04-08:
> Exact same error started occuring to me after recent update of `intel-media-driver 22.2.2-1`. On a `00:02.0 VGA compatible controller: Intel Corporation Skylake GT2 [HD Graphics 520] (rev 07)` (i3-6006U), running manjaro with kernel 5.17.1, `xorg-server 21.1.3-4`, `xf86-video-intel 1:2.99`.
> 
> Forcing LIBVA_DRIVER_NAME=i965 allows hwaccel to work normally.

Forcing LIBVA_DRIVER_NAME doesn't do anything as a `vainfo` argument to me.  Downgrading to intel-media-driver 21.4.3-1 or 21.4.1-1 does not change the output of `vainfo` or stop the visible stuttering. `intel_gpu_top` shows that I am pegging the gpu at 100%, but as a Render/3D load, while no usage is shown for Video or VideoEnhance.

david-morris, 2022-04-09:
Interesting... unfortunately I can't provide much further testing, because my video acceleration seems to be working now!

It looks like it may have been an issue with the Arch package which the maintainer fixed today.

after checking the output of `vainfo` while forcing the iHD driver after my downgrade, there was an error indicating a version mismatch of `libigdgmm.so`. It wanted version 10, and I only had version 12, so I figured, I better upgrade again to see what fails when it has the right version.  So I did a full system upgrade, with these packages:
```
Package (19)                     Old Version  New Version  Net Change  Download Size

extra/adwaita-icon-theme         41.0-1       42.0-1        -6.08 MiB       4.35 MiB
extra/gsettings-desktop-schemas  41.0-1       42.0-1         0.03 MiB       0.65 MiB
core/gzip                        1.11-1       1.12-1        -0.01 MiB       0.08 MiB
core/hwdata                      0.357-1      0.358-1        0.00 MiB       0.46 MiB
community/intel-media-driver     21.4.1-1     22.3.1-1      -3.78 MiB       5.22 MiB
core/krb5                        1.19.2-2     1.19.3-1      -0.11 MiB       1.25 MiB
multilib/lib32-libffi            3.4.2-3      3.4.2-4        0.00 MiB       0.02 MiB
multilib/lib32-libgpg-error      1.44-1       1.45-1         0.00 MiB       0.06 MiB
multilib/lib32-libx11            1.7.3.1-1    1.7.4-1        0.01 MiB       0.62 MiB
multilib/lib32-mesa              22.0.0-1     22.0.1-3       0.01 MiB      10.80 MiB
multilib/lib32-vulkan-intel      22.0.0-1     22.0.1-3       0.01 MiB       2.20 MiB
core/libaio                      0.3.112-2    0.3.112-3      0.02 MiB       0.02 MiB
core/libffi                      3.4.2-4      3.4.2-5        0.00 MiB       0.04 MiB
core/libgpg-error                1.44-1       1.45-1         0.00 MiB       0.24 MiB
extra/libibus                    1.5.25-5     1.5.26-2       0.01 MiB       9.62 MiB
extra/libva                      2.13.0-2     2.14.0-1       0.05 MiB       0.18 MiB
community/python-zipp            3.7.0-1      3.8.0-1        0.00 MiB       0.01 MiB
core/wpa_supplicant              2:2.10-3     2:2.10-4       0.09 MiB       1.56 MiB
core/xz                          5.2.5-2      5.2.5-3        0.00 MiB       0.29 MiB
```

And now here's the output from `vainfo` when I set LIBVA_DRIVER_NAME (or unset $DISPLAY but then it complains about not knowing where X is, of course):

```
vainfo: VA-API version: 1.14 (libva 2.14.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.3.1 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSliceLP
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSliceLP
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile1            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileVP9Profile3            : VAEntrypointVLD
      VAProfileHEVCMain422_10         : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         : VAEntrypointVLD
      VAProfileHEVCMain444_10         : VAEntrypointEncSliceLP
```
Then, playing a 4k video with MPV with LIBVA_DRIVER_NAME set caused small Render/3D and Blitter loads as read by `intel_gpu_top`.

Playing a 4k video on Kodi without touching the LIBVA_DRIVER_NAME causes moderate Render/3D and Video loads and the playback is once again silky smooth, including shifting perspectives, sunlit water, snow, and sparklers.

It looks like Bruno Pagani, the Arch packager, updated the package today.  So that might be what fixed it.

jike212, 2022-04-11:
It's interesting that you saw activity with `Blitter` in intel_gpu_top, I have hardware acceleration working with totem (both Video and VideoEnhance engines are working) but I see zero `Blitter` activity in any task (video, web browsing, etc.)

From the looks of it, it doesn't look like your MPV is fully accelerated if you're not seeing Video or VideoEnhance modes. Mine is the same way (only shows Video load) and I think that's an FFMPEG issue.

I've had better results with gstreamer-backed media players like Totem and the necessary gstreamer packages (including `gstreamer1-plugins-ugly` and `gstreamer1-vaapi1`)

Curious to know how you got Blitter working

jike212, 2022-04-11:
> Anything listed in the decoding / encoding table here as "Es":
https://github.com/intel/media-driver/ means using shader pipeline in media mode for encoding, which in intel_gpu_top shows up as Render/3D. Only low-power (fixed HW) encoding (listed as "E" in the table) uses the GPU video encoding HW block(s).

I'm actually not sure if the OP was having issues encoding or decoding but, in my case, I see both `Video` and `VideoEnhance` activity when using Totem (running Fedora 36 beta) however I do not see any activity on `VideoEnhance` when using MPV or VLC.

Should those engines show activity when decoding a supported format? I just checked htop while playing video in both Totem and MPV and MPV actually shows lower CPU usage than Totem (even though Totem is using `VideoEnhance`).

eero-t, 2022-04-11:
Yes, decoding should happen with fixed HW part of GPU.  Frame scaling / format conversion can also be done with fixed HW, if format is something supported by it, and driver is asked for that, so maybe `VideoEnhance` is due to scaling / format conversion.  Do you know whether Totem asks GPU to provide video in different format/size than MPV/VLC?

In all cases I would assume actual output to screen to be done using 3D pipeline, but scaling could be done either when video is processed (good when downscaling), or when it's drawn to screen (good when upscaling).  This may include round-trip to CPU side, especially if format is something that's supported neither by fixed video HW, nor drivers GPU shaders.

PS. If you want to know more, just profile it:
* start video
* start profiling "sudo perf record -a"
* after a while, stop profiling with ^C
* post-process data "perf report -n"

jike212, 2022-04-11:
Thanks! Some new information there that I hadn't seen before.

In the case of MPV, the following lines appear and the video opens at full resolution:
```
[vo/gpu/wayland] GNOME's wayland compositor lacks support for the idle inhibit protocol. This means the screen can blank during playback.
[ffmpeg] AVHWDeviceContext: Failed to query surface attributes: 20 (the requested function is not implemented).
[vo/gpu/vaapi-egl] failed to retrieve libavutil frame constraints
```
Running Totem on the other hand with `--gst-debug-level=4`, I see the instantiation of `vaapipostproc` which may be what is leveraging the `VideoEnhance` engine: https://gstreamer.freedesktop.org/documentation/vaapi/vaapipostproc.html?gi-language=c

foxmajik, 2022-04-15:
This bug causes attempts to play some video formats to fail on Plex for Synology D-series "+" NAS (the "+" version uses Intel Celeron processor) because it uses hardware transcode by default,

The error reported by OP is output to the console upon trying to play such videos.

The workaround is to uncheck the box "Use hardware-accelerated video encoding" in Settings > Transcoder.

mauricioquintela, 2022-04-16:
> > > i3-6006U
> > 
> > 
> > I could not find the spec of i3-6006U in ark.intel.com but suppose it is older than BDW, media-driver start support from BDW, older platform please use i965 https://github.com/intel/intel-vaapi-driver
> 
> I was using iHD because, as mentioned in the documentation, it supports skylake. The ark.intel page for the processor: https://ark.intel.com/content/www/us/en/ark/products/91157/intel-core-i36006u-processor-3m-cache-2-00-ghz.html
> 
> EDIT (because i replied to the wrong comment): Running vainfo with `LIBVA_DRIVER_NAME=iHD` results simply in
> 
> ```
> ~ > LIBVA_DRIVER_NAME=iHD vainfo                                                 
> libva error: /usr/lib/dri/iHD_drv_video.so init failed
> vaInitialize failed with error code 1 (operation failed),exit
> ```
> 
> With i965, manually set systemwide via environment variables,
> 
> ```
>  ~ > vainfo
> vainfo: VA-API version: 1.13 (libva 2.14.0)
> vainfo: Driver version: Intel i965 driver for Intel(R) Skylake - 2.4.1
> vainfo: Supported profile and entrypoints
> ```

Returning to this issue, today's update to `intel-media-driver 22.2.2-2` (manjaro testing branch) appears to have fixed the issue. 
```
LIBVA_DRIVER_NAME=iHD vainfo
vainfo: VA-API version: 1.13 (libva 2.14.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.2.2 ()
vainfo: Supported profile and entrypoints 
(...)
```

MartinX3, 2022-06-03:
Using arch linux and a intel i7-6700hq I get this error

```bash
0:00:00.419213354    34 0x5643fe7cfe30 WARN                 nvcodec plugin.c:94:plugin_init: Failed to init cuda, ret: 0x3e7
0:00:00.528437759    34 0x5643fe7cfe30 WARN                  ladspa gstladspa.c:508:plugin_init:<plugin127> no LADSPA plugins found, check LADSPA_PATH
0:00:00.555492010    34 0x5643fe7cfe30 FIXME                    lv2 gstlv2.c:203:lv2_plugin_discover: plugin lv2plug-in-plugins-eg-fifths has no audio pads
DRM_IOCTL_I915_GEM_APERTURE failed: Das Argument ist ung?ltig
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
0:00:01.052902314    24 0x55c24d373e30 WARN                   vaapi gstvaapiutils.c:78:gst_vaapi_warning: /usr/lib/dri/iHD_drv_video.so init failed
```

But
```bash
$ vainfo
vainfo: VA-API version: 1.14 (libva 2.14.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.4.2 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointFEI
```

```bash
$ sudo cat /sys/kernel/debug/dri/128/name
i915 dev=0000:00:02.0 master=pci:0000:00:02.0 unique=0000:00:02.0
```

```bash
$ sudo cat /sys/kernel/debug/dri/129/name
nvidia-drm dev=0000:02:00.0 unique=0000:02:00.0
```

david-morris, 2022-06-15:
I've noticed on my setup, video acceleration stops working after a few days of uptime.  That's not great on a fileserver/media-pc/smart-tv-brain.  I'll try reloading various kernel modules before rebooting next time it happens, but are there any particular debug steps I could do to generate logs that would be useful for developers?

david-morris, 2022-06-16:
Unloading i915 with dependencies and restarting X caused video acceleration to work again, hopefully this was not a coincidence.
```
$ sudo dmesg | grep i915                                                                                                                                               (06-16 08:02)
[    3.395778] fb0: switching to i915 from EFI VGA
[    3.396575] i915 0000:00:02.0: vgaarb: deactivate vga console
[    3.398347] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    3.406706] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/icl_dmc_ver1_09.bin (v1.9)
[    3.424173] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    3.522102] fbcon: i915drmfb (fb0) is primary device
[    3.589597] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[    3.833609] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[    3.963231] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
```
```
$ sudo dmesg | grep video                                                                                                                                           (06-16 08:05)
[    3.469728] ACPI: video: Video Device [GFX0] (multi-head: yes  rom: no  post: no)
```
No other reverse dependencies had any kernel messages.

david-morris, 2022-06-18:
Unfortunately, causing the modules to reload was only associated with the problem going away as a coincidence.  I'd be grateful if someone could give me a better workaround, or if this went away in a future version of the drivers.  Please feel free to ask for any diagnostic information.

sonichy, 2023-11-29:
Devuan daedalus
Edge 119
i3 540

eshirvana, 2024-03-17:
I'm trying to enable transcoding with jellyfin ( insode docker) on N3050 and having the same issue on kernel 5.15 .
anyone can help how I can solve this issue?

david-morris, 2024-03-18:
> I'm trying to enable transcoding with jellyfin ( insode docker) on N3050 and having the same issue on kernel 5.15 . anyone can help how I can solve this issue? vainfo on the host :
> 
> ```
> vainfo
> error: can't connect to X server!
> libva info: VA-API version 1.14.0
> libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
> libva info: Found init function __vaDriverInit_1_14
> libva error: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed
> libva info: va_openDriver() returns 1
> libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
> libva info: Found init function __vaDriverInit_1_10
> libva info: va_openDriver() returns 0
> vainfo: VA-API version: 1.14 (libva 2.12.0)
> vainfo: Driver version: Intel i965 driver for Intel(R) CherryView - 2.4.1
> ```

IIRC that sort of `vainfo` output happens when you don't have environment variables set from x11.  Are you using a wayland compositor or running this logged in as root?

eero-t, 2024-03-18:
> I'm trying to enable transcoding with jellyfin ( insode docker) on N3050 and having the same issue on kernel 5.15 . 
> ```
> ...
> vainfo: Driver version: Intel i965 driver for Intel(R) CherryView - 2.4.1
> ```

@eshirvana This bug tracker is for media-driver, which supports HW up from Broadwell: https://github.com/intel/media-driver/blob/master/README.md#supported-platforms

However, while Braswell (N3050) is of the same (GPU) GEN8 as Broadwell, GEN8 Atoms like yours are supported only by the legacy i965 video driver: https://github.com/intel/intel-vaapi-driver

(as can be seen from your host vainfo output.)

> I'm trying to enable transcoding with jellyfin ( insode docker)

Please install the correct driver package _also_ to your container image:
```
$ dpkg -S /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
i965-va-driver:amd64: /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
```

and make sure to configure your container so that driver within it can write to the GPU device (maps GPU device and has correct group and user, or or run it as privileged, i.e. unsecured).

> IIRC that sort of vainfo output happens when you don't have environment variables set from x11. Are you using a wayland compositor or running this logged in as root?

@david-morris As it's running inside container, it's unlikely to have access to (host?) display server?

eshirvana, 2024-03-19:
@eero-t  thank you for your reply. how do I install that driver  https://github.com/intel/intel-vaapi-driver ? sorry newbie to linux drivers
by the way on the host i see i965_drv_video.so is installed but not inside the container

eero-t, 2024-03-19:
@eshirvana install the same driver package from your distro repositories, as you have on the host, also to the container.

In my case, i.e. Ubuntu (and Debian), the package containing the driver is:
```
$ dpkg -S /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
i965-va-driver:amd64: /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
```

(In case you do not use Debian derived distro, package managers for most other distros should also have a similar option for listing which package provides given file.  Google can be your friend in that case.)

eshirvana, 2024-03-19:
@eero-t  thank you for your help again , i just noticed even though I had the driver `i956-va-driver` installed on the host . I'm still getting that error message on the host as well. so I must be missing something else too. no need to say I installed the driver inside the container an no success

eero-t, 2024-03-20:
> @eero-t thank you for your help again , i just noticed even though I had the driver `i956-va-driver` installed on the host . 

As stated, this is intel-media bug tracker for the latest HW, and you're using legacy HW, supported by the legacy driver.  You should file your issue at https://github.com/intel/intel-vaapi-driver, not here.

> I'm still getting that error message on the host as well. so I must be missing something else too. no need to say I installed the driver inside the container an no success

What error?  VA-API frontend tries first iHD backend driver (as it's preferred one), it tells that the HW is not supported by it, frontend tries then the next backend (i965 one) and that supports your HW...?

@XinfengZhang IMHO this bug should be closed as the original issue was filed against wrong project, and then people started commenting here things for completely different HW (some of them relevant for this project, some not). It's not useful any more.

eshirvana, 2024-03-20:
@eero-t  thank you

clion007, 2024-06-18:
play H265 video, I get /usr/lib/dri/iHD_drv_video.so init failed error every time, in jellyfin-web on Intel® HD Graphics 2000.

eero-t, 2024-06-18:
> play H265 video, I get /usr/lib/dri/iHD_drv_video.so init failed error every time, in jellyfin-web on Intel® HD Graphics 2000.

That's expected as Sandy Bridge iGPU (released in 2011) is not supported by this driver (as documented in main README).  You need the legacy driver mentioned earlier in the ticket.

Note: that old GPUs do not support H.265:
https://en.wikipedia.org/wiki/Intel_Graphics_Technology#Hardware-accelerated_algorithms

Because the H.265 standard came only after them:
https://en.wikipedia.org/wiki/High_Efficiency_Video_Coding

tbertels, 2025-02-07:
intel-vaapi-driver development has been discontinued by Intel.
https://github.com/intel/intel-vaapi-driver/commit/fd727a4e9cb8b2878a1e93d4dddc8dd1c1a4e0ea

eero-t, 2025-02-07:
> intel-vaapi-driver development has been discontinued by Intel. [intel/intel-vaapi-driver@fd727a4](https://github.com/intel/intel-vaapi-driver/commit/fd727a4e9cb8b2878a1e93d4dddc8dd1c1a4e0ea)

Although Intel does not any more maintain the _user-space_ drivers for more-than-decade-old GPU architectures, those drivers are still included in latest Linux distributions, for example:
* https://packages.debian.org/sid/i965-va-driver
* https://packages.ubuntu.com/plucky/i965-va-driver

=> I would suggest filing issues with them against corresponding distribution driver package.

I think their package maintainers expect that people using legacy HW have done some minimal investigation before filing bugs though (e.g. checked docs and verified that correct driver package is installed for given HW, `strace`d the program to see that it actually tries and uses that driver, and has access rights to corresponding device).

PS. AFAIK Intel still maintains _kernel_ side driver (`i915`) for these older GPU architectures (with automated testing to catch regressions).

PPS. I'm not driver developer, just another user of the drivers (with some soon-to-be-legacy HW).