intel/media-driver #1299: can't

lordofcinderr, 2021-11-28:
## System information
- CPU information
` Intel(R) Core(TM) i5-9600K CPU @ 3.70GHz`
- GPU information:
`00:02.0 VGA compatible controller [0300]: Device [1234:1111] (rev 02) `
`00:10.0 VGA compatible controller [0300]: Intel Corporation Device [8086:3e98]`

## Issue behavior
case when I use `export LIBVA_DRIVER_NAME=i965`
 expect supported profile for decode/encode as the document said
I also tried to use`export  LIBVA_DRIVER_NAME=iHD'
but I get the error messae

## Debug information
- What's libva/libva-utils/gmmlib/media-driver version?
```
libva-drm2: 2.4.0-1
libva-wayland2: 2.4.0-1
libva-x11-2: 2.4.0-1
libva2: 2.4.0-1

i965-va-driver: 2.3.0+dfsg1-1
```
- Could you confirm whether GPU hardware exist or not by `ls /dev/dri`?
1. by-path  
2. card0	
3. card1   
4. renderD128 ⬅️
- Could you provide vainfo log if possible by `vainfo >vainfo.log 2>&1`? 
When set `LIBVA_DRIVER_NAME=iHD`, I get
```error: can't connect to X server!
libva info: VA-API version 1.4.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_4
libva error: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 1
vaInitialize failed with error code 1 (operation failed),exit
```
When set `LIBVA_DRIVER_NAME=i965`, I get
```
libva info: Open new log file /tmp/libva_trace.log.112044.thd-0x000047cb for the thread 0x000047cb
libva info: LIBVA_TRACE is on, save log into /tmp/libva_trace.log.112044.thd-0x000047cb
libva info: VA-API version 1.4.0
libva info: va_getDriverName() returns 0
libva info: User requested driver 'i965'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_4
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.4 (libva 2.4.0)
vainfo: Driver version: Intel i965 driver for Intel(R) Coffee Lake - 2.3.0
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
```
There's no more supported profile printed.And when I use ffmpeg it went wrong with 'No support for codec hevc profile 2'
- Do you want to contribute a PR? (yes/no): no
- Could you attach dmesg log if it's GPU hang by `dmesg >dmesg.log 2>&1`?
[    0.000000] Linux version 5.10.0-0.bpo.9-amd64 (debian-kernel@lists.debian.org) (gcc-8 (Debian 8.3.0-6) 8.3.0, GNU ld (GNU Binutils for Debian) 2.31.1) #1 SMP Debian 5.10.70-1~bpo10+1 (2021-10-10)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.10.0-0.bpo.9-amd64 root=UUID=7550d17c-de06-4691-8208-c50ab8973409 ro quiet
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000bffd9fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bffda000-0x00000000bfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feffc000-0x00000000feffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x00000002bfffffff] usable
[... 694 lines cut ...]
[ 2372.569571] device vetha8e9d3a entered promiscuous mode
[ 2372.570294] br-99f47321c320: port 1(vetha8e9d3a) entered blocking state
[ 2372.570295] br-99f47321c320: port 1(vetha8e9d3a) entered forwarding state
[ 2372.570340] br-99f47321c320: port 1(vetha8e9d3a) entered disabled state
[ 2372.817961] eth0: renamed from veth102b73e
[ 2372.845071] IPv6: ADDRCONF(NETDEV_CHANGE): vetha8e9d3a: link becomes ready
[ 2372.845091] br-99f47321c320: port 1(vetha8e9d3a) entered blocking state
[ 2372.845092] br-99f47321c320: port 1(vetha8e9d3a) entered forwarding state
[ 2372.845119] IPv6: ADDRCONF(NETDEV_CHANGE): br-99f47321c320: link becomes ready
[ 3044.673943] br-99f47321c320: port 1(vetha8e9d3a) entered disabled state
[ 3044.674496] veth102b73e: renamed from eth0
[ 3044.720958] br-99f47321c320: port 1(vetha8e9d3a) entered disabled state
[ 3044.723913] device vetha8e9d3a left promiscuous mode
[ 3044.723918] br-99f47321c320: port 1(vetha8e9d3a) entered disabled state
[ 3256.794886] br-f0a354559987: port 1(vethf41436a) entered blocking state
[ 3256.794887] br-f0a354559987: port 1(vethf41436a) entered disabled state
[ 3256.795013] device vethf41436a entered promiscuous mode
[ 3256.795394] br-f0a354559987: port 1(vethf41436a) entered blocking state
[ 3256.795395] br-f0a354559987: port 1(vethf41436a) entered forwarding state
[ 3256.796377] br-f0a354559987: port 1(vethf41436a) entered disabled state
[ 3257.119551] eth0: renamed from veth1de06ac
[ 3257.138989] IPv6: ADDRCONF(NETDEV_CHANGE): vethf41436a: link becomes ready
[ 3257.139010] br-f0a354559987: port 1(vethf41436a) entered blocking state
[ 3257.139011] br-f0a354559987: port 1(vethf41436a) entered forwarding state
[ 3257.139035] IPv6: ADDRCONF(NETDEV_CHANGE): br-f0a354559987: link becomes ready

XinfengZhang, 2021-12-05:
could you help to try  vainfo --device card1 --display drm

burkely-00orso69, 2022-03-15:
Hello, has there been any sort of resolution to this issue that I'm not seeing at all?  Figured I'd ask as to try and not contribute to more issues...  

But I'm currently running an Intel  i7-1065G7 CPU @ 1.30GHz w/ Iris Plus Graphics G7 and have spent the bulk of last weekend trying to figure out why it continues to fail..  Although I do have other profiles listed below, and despite my best attempt at trying to resolve this I still get the following when I run vainfo --device card1 --display drm

libva info: VA-API version 1.13.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_13
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.13 (libva 2.6.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 21.4.1 (be92568)
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats

I also the following output when running sudo modinfo i915 | egrep -i "guc|huc|dmc" :
firmware:       i915/bxt_dmc_ver1_07.bin
firmware:       i915/skl_dmc_ver1_27.bin
firmware:       i915/kbl_dmc_ver1_04.bin
firmware:       i915/glk_dmc_ver1_04.bin
firmware:       i915/cnl_dmc_ver1_07.bin
firmware:       i915/icl_dmc_ver1_07.bin
firmware:       i915/tgl_dmc_ver2_04.bin
firmware:       i915/skl_huc_ver01_07_1398.bin
firmware:       i915/skl_guc_33.0.0.bin
firmware:       i915/bxt_huc_ver01_8_2893.bin
firmware:       i915/bxt_guc_33.0.0.bin
firmware:       i915/kbl_huc_ver02_00_1810.bin
firmware:       i915/kbl_guc_33.0.0.bin
firmware:       i915/glk_huc_ver03_01_2893.bin
firmware:       i915/glk_guc_33.0.0.bin
firmware:       i915/kbl_huc_ver02_00_1810.bin
firmware:       i915/kbl_guc_33.0.0.bin
firmware:       i915/icl_huc_ver8_4_3238.bin
firmware:       i915/icl_guc_33.0.0.bin
parm:           enable_guc:Enable GuC load for GuC submission and/or HuC load. Required functionality can be selected using bitmask values. (-1=auto, 0=disable [default], 1=GuC submission, 2=HuC load) (int)
parm:           guc_log_level:GuC firmware logging level. Requires GuC to be loaded. (-1=auto [default], 0=disable, 1..4=enable with verbosity min..max) (int)
parm:           guc_firmware_path:GuC firmware path to use instead of the default one (charp)
parm:           huc_firmware_path:HuC firmware path to use instead of the default one (charp)
parm:           dmc_firmware_path:DMC firmware path to use instead of the default one (charp)

I apoogize in advance if I'm hijacking someone's issue, however, this most similar proplem to what I seem to be experiencing and I no longer seem to be able to figure out what it is I need to look into as info out there seems to contradict other sites.

XinfengZhang, 2022-03-16:
@burkely-00orso69 , let me have a summary:
your platform is ICL, i7-1065G7, with gpu device id 0x8A52. vainfo only report
VAProfileNone : VAEntrypointVideoProc
VAProfileNone : VAEntrypointStats

right?

have you compiled media_driver, if so, could you set a breakpoint in 
MediaLibvaCapsG11::LoadProfileEntrypoints
it will load all codec profile entrypoints, you could check what happened there
.

XinfengZhang, 2022-03-16:
also, if possible , please share libva_trace

XinfengZhang, 2023-05-08:
there are two issues included:
1st issue was reported by @ghost ,  the platform is CFL, graphics deviceID is 3e98
but there are another graphic card in this machine.  so , I guess there are such possible issue:
   open incorrect device with iHD

2nd issue was reported by @burkely-00orso69 , the device is ICL , summary at https://github.com/intel/media-driver/issues/1299#issuecomment-1068657783
possibility:  1. not full feature driver. 2. driver issue

because no more information provided, I close the issue, any concern , feel free to reopen it.
if you reopen it, please try latest media driver.

closed 2023-05-08 by XinfengZhang