intel/media-driver #1300: double free

SiewHoon, 2021-11-29:
## System information
- CPU information(`cat /proc/cpuinfo | grep "model name" | uniq`):
- GPU information(`lspci -nn | grep -E 'VGA|isplay`):
- Display server if rendering to display(X or wayland):

## Issue behavior
### Describe the current behavior

### Describe the expected behavior

## Debug information
- What's libva/libva-utils/gmmlib/media-driver version?
- Could you confirm whether GPU hardware exist or not by `ls /dev/dri`?
- Could you provide vainfo log if possible by `vainfo >vainfo.log 2>&1`?
- Could you provide libva trace log if possible? Run cmd `export LIBVA_TRACE=/tmp/libva_trace.log` first then execute the case.
- Could you attach dmesg log if it's GPU hang by `dmesg >dmesg.log 2>&1`?
- Do you want to contribute a PR? (yes/no):

SiewHoon, 2021-11-29:
This issue does not reproduce by using Q2'2021 Media stack release.

SiewHoon, 2021-11-30:
This issue not able reproduce without this patch : https://github.com/intel/media-driver/commit/99690b6c66eccc710772a1ea82b8d626a7c7089c + Q3'2021 media driver.

Also git checkout this commit 99690b6c66eccc710772a1ea82b8d626a7c7089c  and install it in the system able to reproduce the issue.

Sherry-Lin, 2021-12-01:
it's related to MOS change, so move to media common team debug.

pingliwa, 2021-12-02:
Please help to try this PR https://github.com/intel/media-driver/pull/1305

SiewHoon, 2021-12-02:
Done verified using this PR#1305. The issue no longer able to reproduce.

Sherry-Lin, 2021-12-03:
Thanks @SiewHoon  for quick check. We will try to merge the code as quick as possible.

hye5, 2021-12-15:
Can u update the HSD to fix implemented?

SiewHoon, 2021-12-15:
The issue fixed in PR#1305 fixed and code already merged. Move this issue to closed.

closed 2021-12-15 by SiewHoon