intel/media-driver #1301: Building with -DGEN8=OFF fails

edwarddavidbaker, 2021-11-29:
## System information
- GPU platform enabling for new feature: Build configuration request and not specific to hardware.
- Host machine if it's descreted card: Not a discrete card.
- libva/libva-utils/gmmlib/media-driver version you are using? Top of master branches
- Are you willing to contribute it?(Yes/No) Yes

## Feature Information
### Describe the feature and expected behavior
Building with ```-DGEN8=OFF``` should complete. It currently fails on an include.

```
cmake -DGEN8=OFF ../
make -j$(nproc)
<snip>
[ 21%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/codec/hal/codechal_decode_sfc_hevc.cpp.o
[ 21%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/codec/hal/codechal_decode_jpeg.cpp.o
[ 21%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/codec/hal/codechal_mmc_decode_hevc.cpp.o
In file included from /mnt/data/media-driver/media_driver/agnostic/common/cm/cm_execution_adv.cpp:47:
/mnt/data/media-driver/media_driver/agnostic/gen12/cm/cm_hal_g12.h:32:10: fatal error: mhw_render_hwcmd_g8_X.h: No such file or directory
   32 | #include "mhw_render_hwcmd_g8_X.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:531: media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/cm/cm_execution_adv.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /mnt/data/media-driver/media_driver/agnostic/common/cm/cm_command_buffer.cpp:44:
/mnt/data/media-driver/media_driver/agnostic/gen12/cm/cm_hal_g12.h:32:10: fatal error: mhw_render_hwcmd_g8_X.h: No such file or directory
   32 | #include "mhw_render_hwcmd_g8_X.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:674: media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/cm/cm_command_buffer.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:378: media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
```

### Customer usage and impact for this feature
The final driver shared object is unnecessarily large. For example in Chrome OS we could disable media generations or specific SoC families that are not applicable.
```
ed@ebaker-desk:/mnt/data/media-driver/build$ du -sm media_driver/iHD_drv_video.so 
223     media_driver/iHD_drv_video.so
ed@ebaker-desk:/mnt/data/media-driver/build$ strip media_driver/iHD_drv_video.so 
ed@ebaker-desk:/mnt/data/media-driver/build$ du -sm media_driver/iHD_drv_video.so 
36      media_driver/iHD_drv_video.so
```

mattst88, 2021-11-29:
There's a similar issue filed as https://github.com/intel/media-driver/issues/1065

It makes some sense to me to support disabling particular generations, but especially Gen10 since it never shipped and costs ~3.8 MiB on disk.

edwarddavidbaker, 2021-11-29:
Thanks Matt. Not sure how I missed that when searching for prior issues. I'll close this since it is a duplicate and the other thread has in progress conversations.

closed 2021-11-29 by edwarddavidbaker