intel/media-driver #1306: [JSL] Jasper Lake Gen11 doesn't support non-low-power(PAK+kernel+VME) encoding

nyanmisaka, 2021-12-02:
## System information
- CPU information: Intel(R) Pentium(R) Silver N6005 @ 2.00GHz
- GPU information: Intel Corporation JasperLake [UHD Graphics] [8086:4e71] (rev 01)
- Display server if rendering to display: X11

## Issue behavior
### Describe the current behavior
On Jasper Lake Gen11 platform, the non low-power encoding(PAK+kernel+VME encoding) is not available.
This means only CQP rc_mode is supported in ffmpeg-qsv if the Guc/Huc firmware is not enabled or outdated.
And the CBR or VBR rate control mode won't work in VAAPI or MediaSDK encoder.

For comparison, CBR and VBR works well in Windows driver.

### Describe the expected behavior
Support non low-power encoding(PAK+kernel+VME encoding) on Jasper Lake platform.

## Debug information
- What's libva/libva-utils/gmmlib/media-driver version? lastest release and master version
- Could you confirm whether GPU hardware exist or not by `ls /dev/dri`? /dev/dri/renderD128
- Could you provide vainfo log if possible by `vainfo >vainfo.log 2>&1`?
```
libva info: VA-API version 1.12.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_12
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.12 (libva 2.10.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 21.3.3 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSliceLP
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSliceLP
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointEncSliceLP
      VAProfileVP9Profile1            : VAEntrypointVLD
      VAProfileVP9Profile1            : VAEntrypointEncSliceLP
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointEncSliceLP
      VAProfileVP9Profile3            : VAEntrypointVLD
      VAProfileVP9Profile3            : VAEntrypointEncSliceLP
      VAProfileHEVCMain422_10         : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         : VAEntrypointVLD
      VAProfileHEVCMain444_10         : VAEntrypointEncSliceLP
```

XinfengZhang, 2021-12-05:
@nyanmisaka , suppose it is not expectation? JSL has no  VME, so

nyanmisaka, 2021-12-05:
Can you confirm that JSL doesn't support VME and VEBox?
I see the VEBox was also disabled in the EHL/JSL sku code. The 3D-LUT(HDR10 TM) is disabled as a result.

https://github.com/intel/media-driver/blob/791f0c11359c5bcb8bbd58facb3e696f93d83398/media_driver/linux/gen11/ddi/media_sku_wa_g11.cpp#L364

XinfengZhang, 2021-12-06:
it support VEBox with pass-though mode only, in another words, it just used to program the command for SFC .  so, you could still run CSC , Scaling etc.

nyanmisaka, 2021-12-06:
Thanks for your comments. Close it for now.

closed 2021-12-06 by nyanmisaka