intel/media-driver #1307: Crash in firefox playing media on Intel+nVidia Lenovo 20TD002MCK

jankratochvil, 2021-12-04:
## System information
- CPU information(`cat /proc/cpuinfo | grep "model name" | uniq`):
model name	: 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz
- GPU information(`lspci -nn | grep -E 'VGA|isplay`):
00:02.0 VGA compatible controller [0300]: Intel Corporation TigerLake-LP GT2 [Iris Xe Graphics] [8086:9a49] (rev 01)
05:00.0 3D controller: NVIDIA Corporation TU117M [GeForce MX450] (rev a1)
- Display server if rendering to display(X or wayland):
X modeset

## Issue behavior
### Describe the current behavior
In firefox-94.0-2.fc35.x86_64 (it was not happening on firefox-93) with my Firefox config (maybe not reproducible by default, not sure) I am getting firefox tab crashes often (almost unable to read multimedia pages but sometimes it may take some minutes to reproduce it).

### Describe the expected behavior
No crashes.

## Debug information
- What's libva/libva-utils/gmmlib/media-driver version?
libva-2.13.0-2.fc35.x86_64
intel-gmmlib-21.3.2-1.fc35.x86_64
intel-media-driver-21.3.4-1.fc35.x86_64
libva-vdpau-driver-0.7.4-110.fc35.x86_64
libva-intel-driver-2.4.1-6.fc35.x86_64
- Could you confirm whether GPU hardware exist or not by `ls /dev/dri`?
crw-rw----+ 1 root video  226,   0 Dec  4 21:37 card0
crw-rw----+ 1 root video  226,   1 Nov 30 19:41 card1
crw-rw-rw-  1 root render 226, 128 Nov 30 19:41 renderD128
crw-rw-rw-  1 root render 226, 129 Nov 30 19:41 renderD129
- Could you provide vainfo log if possible by `vainfo >vainfo.log 2>&1`?
[vainfo.log](https://github.com/intel/media-driver/files/7654801/vainfo.log)
- Could you provide libva trace log if possible? Run cmd `export LIBVA_TRACE=/tmp/libva_trace.log` first then execute the case.
I could but I hope it is not needed.
- Could you attach dmesg log if it's GPU hang by `dmesg >dmesg.log 2>&1`?
no hang, not attached
- Do you want to contribute a PR? (yes/no):
What is PR (Problem Report?)? I am doing that now.

[firefox.core.txt](https://github.com/intel/media-driver/files/7654804/firefox.core.txt)

#0  0x0000000000000000 in ?? ()
(gdb) up
#1  0x00007f4c21ecaf0d in MmdDevice::CreateFactory (osDriverContext=<optimized out>) at /usr/src/debug/intel-media-driver-21.3.4-1.fc35.x86_64/media_driver/linux/common/media_interfaces/media_interfaces.cpp:372
372	        MMD_FAILURE();
(gdb) p mhwInterfaces
$5 = (MhwInterfaces *) 0x0
(gdb) p osInterface
$6 = (PMOS_INTERFACE) 0x7f4c24711000
(gdb) p osInterface->pfnDestroy
$7 = (void (*)(PMOS_INTERFACE, int32_t)) 0x0

#define MMD_FAILURE()                                       \
{                                                           \
    if (mhwInterfaces != nullptr)                           \
    {                                                       \
        mhwInterfaces->Destroy();                           \
    }                                                       \
    MOS_Delete(mhwInterfaces);                              \
    if (osInterface != nullptr)                             \
    {                                                       \
        osInterface->pfnDestroy(osInterface, false);        \
        MOS_FreeMemory(osInterface);                        \
    }                                                       \

XinfengZhang, 2021-12-05:
actually, there are some issue with sandbox of firefox , could you help to try latest master with commit 4e70cd699 , this commit remove IPC call for TGL

jankratochvil, 2021-12-05:
Yes, it is working now. Thanks for finding the issue. I have [requested update for rpmfusion](https://bugzilla.rpmfusion.org/show_bug.cgi?id=6162).
rpmfusion is probably following the official intel-media-driver releases, maybe you could make a new release as this issue is serious.

closed 2021-12-05 by jankratochvil

XinfengZhang, 2021-12-05:
actually, we have another PR for older platform, #1293 , so, suppose we will have a release after it merged , then TGL and older platform will  both work.