intel/media-driver #1309: no vp9 encode for TGL

xsmart, 2021-12-05:
## System information
- CPU information(`cat /proc/cpuinfo | grep "model name" | uniq`): 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz
- GPU information(`lspci -nn | grep -E 'VGA|isplay`): 00:02.0 VGA compatible controller [0300]: Intel Corporation Device [8086:9a49] (rev 01)
- Display server if rendering to display(X or wayland):X
- 
Driver: https://github.com/intel/media-driver/releases/tag/intel-media-21.3.4
Gmmlib: https://github.com/intel/gmmlib/releases/tag/intel-gmmlib-21.3.1
libva: https://github.com/intel/libva/releases/tag/2.13.0
libva-utils: https://github.com/intel/libva-utils/releases/tag/2.12.0

libva info: VA-API version 1.13.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /home/user/h5s/lib/dri//iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_13
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.13 (libva 2.13.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 21.3.4 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointFEI
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointFEI
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointFEI
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointFEI
      VAProfileHEVCMain               : VAEntrypointEncSliceLP
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSlice
      VAProfileHEVCMain10             : VAEntrypointEncSliceLP
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile1            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileVP9Profile3            : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointEncSlice
      VAProfileHEVCMain422_10         : VAEntrypointVLD
      VAProfileHEVCMain422_10         : VAEntrypointEncSlice
      VAProfileHEVCMain422_12         : VAEntrypointVLD
      VAProfileHEVCMain422_12         : VAEntrypointEncSlice
      VAProfileHEVCMain444            : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         : VAEntrypointVLD
      VAProfileHEVCMain444_10         : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_12         : VAEntrypointVLD
      VAProfileHEVCSccMain            : VAEntrypointVLD
      VAProfileHEVCSccMain            : VAEntrypointEncSliceLP
      VAProfileHEVCSccMain10          : VAEntrypointVLD
      VAProfileHEVCSccMain10          : VAEntrypointEncSliceLP
      VAProfileHEVCSccMain444         : VAEntrypointVLD
      VAProfileHEVCSccMain444         : VAEntrypointEncSliceLP
      VAProfileAV1Profile0            : VAEntrypointVLD
      VAProfileHEVCSccMain444_10      : VAEntrypointVLD
      VAProfileHEVCSccMain444_10      : VAEntrypointEncSliceLP

XinfengZhang, 2021-12-05:
@xsmart , could you share your build option ? by default VP9 enc is enabled, but if you build with ENABLE_NONFREE_KERNELS = OFF, it will be disabled.

xsmart, 2021-12-06:
thanks 
 cmake -DMEDIA_RUN_TEST_SUITE=OFF -DCMAKE_CXX_FLAGS=-fPIC -DCMAKE_C_FLAGS=-fPIC -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/home/user/h5sv/output/linux-x86_64-64bit/  ../../;

when check build command, it seem VP ENC is enable

-DENABLE_KERNELS -DHEVC_FEI_ENABLE_CMRT -DIGFX_GEN10_CNL_SUPPORTED -DIGFX_GEN10_SUPPORTED -DIGFX_GEN11_ICLLP_SUPPORTED -DIGFX_GEN11_JSL_SUPPORTED -DIGFX_GEN11_SUPPORTED -DIGFX_GEN12_ADLP_SUPPORTED -DIGFX_GEN12_ADLS_SUPPORTED -DIGFX_GEN12_RKL_SUPPORTED -DIGFX_GEN12_SUPPORTED -DIGFX_GEN12_TGLLP_CMFCPATCH_SUPPORTED -DIGFX_GEN12_TGLLP_CMFC_SUPPORTED -DIGFX_GEN12_TGLLP_SUPPORTED -DIGFX_GEN12_TGLLP_SWSB_SUPPORTED -DIGFX_GEN8_BDW_SUPPORTED -DIGFX_GEN8_SUPPORTED -DIGFX_GEN9_BXT_SUPPORTED -DIGFX_GEN9_CFL_SUPPORTED -DIGFX_GEN9_CML_SUPPORTED -DIGFX_GEN9_CMPV_SUPPORTED -DIGFX_GEN9_GLK_SUPPORTED -DIGFX_GEN9_KBL_SUPPORTED -DIGFX_GEN9_SKL_SUPPORTED -DIGFX_GEN9_SUPPORTED -DMEDIA_VERSION=\"21.3.4\" -DMEDIA_VERSION_DETAILS=\"\" -DVEBOX_AUTO_DENOISE_SUPPORTED=1 -DX11_FOUND -D_AV1_DECODE_SUPPORTED -D_AVC_DECODE_SUPPORTED -D_AVC_ENCODE_VDENC_SUPPORTED -D_AVC_ENCODE_VME_SUPPORTED -D_COMMON_ENCODE_SUPPORTED -D_DECODE_PROCESSING_SUPPORTED -D_HEVC_DECODE_SUPPORTED -D_HEVC_ENCODE_VDENC_SUPPORTED -D_HEVC_ENCODE_VME_SUPPORTED -D_JPEG_DECODE_SUPPORTED -D_JPEG_ENCODE_SUPPORTED -D_MMC_SUPPORTED -D_MPEG2_DECODE_SUPPORTED -D_MPEG2_ENCODE_VME_SUPPORTED -D_VC1_DECODE_SUPPORTED -D_VP8_DECODE_SUPPORTED -D_VP8_ENCODE_SUPPORTED -D_VP9_DECODE_SUPPORTED -D_VP9_ENCODE_VDENC_SUPPORTED -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -D__VPHAL_SFC_SUPPORTED=1

XinfengZhang, 2021-12-07:
@ansokolo , could you help to take a look?

dyang23, 2021-12-07:
As I know by default TGL disable HuC firmware, without HuC firmware media driver will take as no VP9 encode support, please enable HuC firmware first.

xsmart, 2021-12-07:
The OS is Description:    Ubuntu 20.04.3 LTS

echo "options i915 enable_guc=2" > /etc/modprobe.d/i915.conf (as root)

and i find there has error, 
# dmesg  | grep guc
[    2.983617] Setting dangerous option enable_guc - tainting kernel
[    3.039451] i915 0000:00:02.0: Direct firmware load for i915/tgl_guc_49.0.1.bin failed with error -2
[    3.039457] i915 0000:00:02.0: [drm] GuC firmware i915/tgl_guc_49.0.1.bin: fetch failed with error -2

dyang23, 2021-12-07:
You can check HuC/GuC firmware load status by "cat /sys/kernel/debug/dri/0/i915_huc_load_status"
If no please make sure you installed the firmware in your Ubuntu system, then try to load it.

xsmart, 2021-12-07:
manual download the file to from https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/ to /lib/firmware/i915/ will be OK.
it works. thanks
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointEncSliceLP
      VAProfileVP9Profile1            : VAEntrypointVLD
      VAProfileVP9Profile1            : VAEntrypointEncSliceLP
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointEncSliceLP
      VAProfileVP9Profile3            : VAEntrypointVLD
      VAProfileVP9Profile3            : VAEntrypointEncSliceLP

closed 2021-12-07 by xsmart