intel/media-driver #1311: The latest release fails to build with gmmlib 21.3.4

primeos, 2021-12-11:
See https://github.com/intel/gmmlib/issues/93. I haven't tested with `master` but 42ffd7983a32b75cf8403c235d8d0f727191390c might fix this. Maybe it's time for a new release?

ffontaine, 2021-12-21:
Indeed, a new release would be great as https://github.com/intel/media-driver/commit/42ffd7983a32b75cf8403c235d8d0f727191390c does not apply cleanly on 21.4.3.

primeos, 2021-12-30:
https://github.com/intel/gmmlib/issues/92#issuecomment-996067414 fixed this by switching to a better versioning scheme (not sure why I even opened this issue for media-driver as well) and there's a new media-driver release now: https://github.com/intel/media-driver/releases/tag/intel-media-22.1.0.

closed 2021-12-30 by primeos