intel/media-driver #1313: [Encode] GLK VP8

edwarddavidbaker, 2021-12-13:
## System information
- GPU platform enabling for new feature: `N5000` device `0x3184`.
- Host machine if it's descreted card: Not applicable.
- libva/libva-utils/gmmlib/media-driver version you are using? Top of tree.
- Are you willing to contribute it? Yes.

## Feature Information
### Describe the feature and expected behavior
If possible, GLK platforms should support `VAProfileVP8Version0_3          : VAEntrypointEncSlice`.

As an example the following is a diff between `vainfo` with i965 and iHD.
```diff
-vainfo: Driver version: Intel i965 driver for Intel(R) Gemini Lake - 2.4.1
+vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 21.2.3 ()
       VAProfileH264ConstrainedBaseline:        VAEntrypointEncSlice
       VAProfileH264ConstrainedBaseline:        VAEntrypointEncSliceLP
+      VAProfileH264ConstrainedBaseline:        VAEntrypointFEI
       VAProfileH264ConstrainedBaseline:        VAEntrypointVLD
       VAProfileH264High               :        VAEntrypointEncSlice
       VAProfileH264High               :        VAEntrypointEncSliceLP
+      VAProfileH264High               :        VAEntrypointFEI
       VAProfileH264High               :        VAEntrypointVLD
       VAProfileH264Main               :        VAEntrypointEncSlice
       VAProfileH264Main               :        VAEntrypointEncSliceLP
+      VAProfileH264Main               :        VAEntrypointFEI
       VAProfileH264Main               :        VAEntrypointVLD
-      VAProfileH264MultiviewHigh      :        VAEntrypointEncSlice
-      VAProfileH264MultiviewHigh      :        VAEntrypointVLD
-      VAProfileH264StereoHigh         :        VAEntrypointEncSlice
-      VAProfileH264StereoHigh         :        VAEntrypointVLD
       VAProfileHEVCMain10             :        VAEntrypointEncSlice
       VAProfileHEVCMain10             :        VAEntrypointVLD
       VAProfileHEVCMain               :        VAEntrypointEncSlice
+      VAProfileHEVCMain               :        VAEntrypointFEI
       VAProfileHEVCMain               :        VAEntrypointVLD
       VAProfileJPEGBaseline           :        VAEntrypointEncPicture
       VAProfileJPEGBaseline           :        VAEntrypointVLD
       VAProfileMPEG2Main              :        VAEntrypointVLD
       VAProfileMPEG2Simple            :        VAEntrypointVLD
+      VAProfileNone                   :        VAEntrypointStats
       VAProfileNone                   :        VAEntrypointVideoProc
       VAProfileVC1Advanced            :        VAEntrypointVLD
       VAProfileVC1Main                :        VAEntrypointVLD
       VAProfileVC1Simple              :        VAEntrypointVLD
-      VAProfileVP8Version0_3          :        VAEntrypointEncSlice
       VAProfileVP8Version0_3          :        VAEntrypointVLD
-      VAProfileVP9Profile0            :        VAEntrypointEncSlice
       VAProfileVP9Profile0            :        VAEntrypointVLD
       VAProfileVP9Profile2            :        VAEntrypointVLD
```

### Customer usage and impact for this feature
VP8 encode is an important capability for real time conferencing scenarios. Lacking VP8 hardware support shifts encode activity to the CPU. Libvpx does have numerous SIMD optimizations, and intrinsics up through SSE4.2 are applicable to GLK SoCs. However, the additional CPU footprint is still likely problematic for multitasking scenarios (video call + browsing or document editing).

Sherry-Lin, 2021-12-17:
Thank you to provide the patch. Could you submit a pull request for it?

edwarddavidbaker, 2021-12-17:
Thanks Sherry. I opened a pull request.

PR #1315 (2021-12-31): [Encode] GLK enable VP8
Enable VP8 encode for Gemini Lake. Interface code was inherited from
KBL paths. Resolves #1313.
commit 4485349d (2021-12-31): [Encode] GLK enable VP8

Enable VP8 encode for Gemini Lake. Interface code was inherited from
KBL paths. Resolves #1313.
closed 2021-12-31 by intel-mediadev