intel/media-driver #1314: Core dumped when running vp9 tile encoding

xhaihao, 2021-12-16:
### CPU info
model name      : Genuine Intel(R) CPU 0000 @ 3.60GHz

### GPU info
00:02.0 VGA compatible controller [0300]: Intel Corporation Device [8086:3e92]
03:00.0 VGA compatible controller [0300]: Intel Corporation Device [8086:4907]

### Describe the current behavior

Core dumped when running the command beblow:
```
sample_encode vp9 -device /dev/dri/renderD129 -i in.yuv -w 3840 -h 2160 -trows 4 -tcols 4 -o out.ivf
libva info: VA-API version 1.13.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_13
libva info: va_openDriver() returns 0
Encoding Sample Version 8.4.27.0

Input file format       YUV420
Output video            VP9
Source picture:
        Resolution      3840x2160
        Crop X,Y,W,H    0,0,3840,2160
Destination picture:
        Resolution      3840x2160
        Crop X,Y,W,H    0,0,3840,2160
Frame rate      30.00
Bit rate(Kbps)  48928
Gop size        65535
Ref dist        1
Ref number      1
Idr Interval    0
Target usage    balanced
Memory type     vaapi
Media SDK impl          hw
Media SDK version       1.35

Processing started
Frame number: 2
Encoding fps: 32
free(): invalid next size (normal)
Aborted (core dumped)
```

### Describe the expected behavior

Run the above command without error. 

## Debug information
- What's libva/libva-utils/gmmlib/media-driver version?
  libva: bc5c4b0e30ff426d7b9d6611098d02fc5c7d45d1
  libva-utils: 83728b910dc72cad575025133cd7fca5b4eaf208
  gmmlib: 70d68b9f7146ef18a2522ecf1ece8075e8d8c207
  media-driver: 66e753116e11b6e54114c06af34d946ec7f8dc29
  MediaSDK: 58231ae06216d6d0eed38a66b9b8ec291cbdebe9

- Could you confirm whether GPU hardware exist or not by `ls /dev/dri`?
  Yes

- Could you provide vainfo log if possible by `vainfo >vainfo.log 2>&1`?
  
- Could you provide libva trace log if possible? Run cmd `export LIBVA_TRACE=/tmp/libva_trace.log` first then execute the case.
- Could you attach dmesg log if it's GPU hang by `dmesg >dmesg.log 2>&1`?
- Do you want to contribute a patch to fix the issue? (yes/no):

eshiryae, 2022-01-19:
Hi @xhaihao, currently this tile configuration (4x4) is not supported, we'll add correct feature support reporting.

Xiaogangli-intel, 2022-05-19:
Hi @xhaihao, could you try #1411 to check if can fix your issue?

PR #1411 (2022-05-25): Correct the calculation method of maxTileNumber for Gen12 VP9
Previously, the calculation of maxTileNumber is wrong on Gen12 VP9, this will cause crash when input 4x4 tiles. Fixes #1314
commit 7700ab9d (2022-05-25): Correct the calculation method of maxTileNumber for Gen12 VP9

Previously, the calculation of maxTileNumber is wrong on Gen12 VP9, this will cause crash when input 4x4 tiles. Fixes #1314
closed 2022-05-25 by intel-mediadev