intel/media-driver #1316: [TGL] VPP sharpen not working for YV12 input

mengker33, 2021-12-22:
## System information
- CPU information: 
  model name	: 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz
- GPU information:
   0000:00:02.0 VGA compatible controller [0300]: Intel Corporation Device [8086:9a49] (rev 01)

### Describe the current behavior
Run vavpp cmd below:
<pre><code>vavpp process_sharpness.cfg
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /home/mengke/opt1/gst-1.0/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_14
libva info: va_openDriver() returns 0
Failed to find integer field: BLENDING_ENABLED
Start to process, processing type is VAProcFilterSharpening ...
Sharpening intensity: 44.000000
Finish processing, performance: 
1 frames processed in: 0.019885 s, ave time = 0.019885s  </pre></code>

 process_sharpness.cfg:
<pre><code> #1.Source YUV(RGB) file information
SRC_FILE_NAME: ./input.yv12
SRC_FRAME_WIDTH: 1920 
SRC_FRAME_HEIGHT: 1080
SRC_FRAME_FORMAT: YV12

#Note .nv12 files are in NV12 format
SRC_FILE_FORMAT: YV12

#2.Destination YUV(RGB) file information
DST_FILE_NAME:    ./vavpp_sharpen_44_yv12.yuv
DST_FRAME_WIDTH:  1920
DST_FRAME_HEIGHT: 1080
DST_FRAME_FORMAT: YV12

DST_FILE_FORMAT: YV12

#3.How many frames to be processed
FRAME_SUM: 1

#4.VPP filter specific parameters. If they are not specified here,
#default value will be applied then.
FILTER_TYPE: VAProcFilterSharpening

#4.1 Sharpening parameters
# (0 ~ 64, default 44)
SHARPENING_INTENSITY: 44 </pre></code>

Different sharpening intensities (tried with 20, 32, 44) generate the yuv which has the same MD5 as original yuv.

### Describe the expected behavior
Ideally, YV12 sharpness can functionally work as on ICL/SKL. If it is not supported on later platforms, better clarify it in document.

mengker33, 2022-01-04:
@FurongZhang, any update on this? 
-Thanks

FurongZhang, 2022-01-04:
We are sure .nv12 file can support sharpness via SFC. Is it possible for you to check NV12. 

Currently, as we know, YV12 sharpness can not be supported via GFX render engine, we are confirming with GFX fixed function.

mengker33, 2022-01-04:
> We are sure .nv12 file can support sharpness via SFC. Is it possible for you to check NV12.
> 
> Currently, as we know, YV12 sharpness can not be supported via GFX render engine, we are confirming with GFX fixed function.

There is no issue with .nv12, only yv12 does not work on TGL and platforms after.

FurongZhang, 2022-01-05:
The conclusion is that YV12 sharpness is not be supported on TGL+ platforms. 

Fixed function(VEBOX+SFC) can not support YV12 format. Render can support YV12 without sharpness due to HW change.

FurongZhang, 2022-01-05:
Will clarify that in document.

FurongZhang, 2022-01-05:
BTW, may I know your usage of YV12?

mengker33, 2022-01-06:
> Will clarify that in document.

Ok, got this. Thanks for your quick help :)

mengker33, 2022-01-06:
> BTW, may I know your usage of YV12?

Actually, we just follow the document and do some tests accordingly. Since no support for YV12 sharpness will exclude this feature in the test.

FurongZhang, 2022-01-06:
> > BTW, may I know your usage of YV12?
> 
> Actually, we just follow the document and do some tests accordingly. Since no support for YV12 sharpness will exclude this feature in the test.

Thanks for your quick reply. OK. We have worked on document update.

FurongZhang, 2022-01-06:
> > BTW, may I know your usage of YV12?
> 
> Actually, we just follow the document and do some tests accordingly. Since no support for YV12 sharpness will exclude this feature in the test.

BTW, which document you are following? What is the link? Will update the link you are referring and update that into this issue link.

mengker33, 2022-01-06:
> The conclusion is that YV12 sharpness is not be supported on TGL+ platforms.
> 
> Fixed function(VEBOX+SFC) can not support YV12 format. Render can support YV12 without sharpness due to HW change.

There is no format-specific description here https://github.com/intel/media-driver#video-processing-features for video processing, the test could be made for all the formats (i.e., NV12 and YV12 for 8bit) if no explicit unsupported format is shown on the page. I have seen a detailed doc on wiki, will leave the comment there.

uartie, 2022-01-06:
There is also SFC table here: https://github.com/intel/media-driver/blob/master/docs/media_features.md#supported-video-processing-cscscaling-format

FurongZhang, 2022-01-06:
We have already changed the document on Wiki.

FurongZhang, 2022-01-06:
Conclusion:  YV12+sharpness is not supported on TGL in media driver.

closed 2022-01-06 by FurongZhang