intel/media-driver #1317: [regression]ffmpeg-qsv encode/10bit/hevc failure on kernel 5.4

fanli-hub, 2021-12-28:
Platform: CFL
Kernel: 5.4.0-91-generic
Firmware 1.173.19
libva : 7be0c8afa5b4
gmmlib : 879c22e4e76e
onevpl : b4094560c82c
msdk-runtime (msdk) : 0db73aa1b2de
onevpl-gpu : 958d8bfa31ce
vaapi-fits-smoke : aaa7b91c5b85
vaapi-fits : df2539a5f754
ffmpeg : db0059ef86df
Good media-driver: 5d333070696482e70a5b5933e0c017f90e080db2
Bad media-driver: 4c689f7e73e9f07a05e8d5f7643f5f96d34710a0

Defect Description:
This issue was caused by the bad commit on kernel 5.4 which works on kernel 5.8 in media-driver.

Reproduce cmd:
ffmpeg -hwaccel qsv -init_hw_device qsv=hw:/dev/dri/renderD128 -hwaccel_output_format qsv -v verbose -f rawvideo -pix_fmt p010le -s:v 1920x1080 -r:v 30 -i /opt/media/src/assets/otc-media/sanity/yuv/1080P_1920x1080.P010 -vf 'format=p010le,hwupload=extra_hw_frames=120' -an -c:v hevc_qsv -profile:v main10 -preset 4 -g 30 -slices 1 -bf 2 -b:v 5000k -maxrate 10000k -refs 2 -low_power 0 -vframes 150 -y /opt/media/src/vaapi-fits-smoke/results/1080P_1920_1080_platform-vbr-main10-30-30-1-4-2-5000k-10000k-2-0.h265

Error message:
[2021-12-28 06:51:03.038819] DEBUG: slash: [hevc_qsv @ 0x564365f70780] Unknown FrameType, set pict_type to AV_PICTURE_TYPE_NONE.
[2021-12-28 06:56:00.821214] INFO: slash: DETAIL: time(ffmpeg:1) = 306.0113s
[2021-12-28 06:56:00.827462] TRACE: slash.core.error: Error added: <Error: AssertionError: CALL TIMEOUT: timeout after 300 seconds (pid: 28708).

uartie, 2021-12-28:
I upgraded kernel from 5.4 to **5.11.0-43-generic** on one CFL that is having this issue.  But the issue did not go away with **5.11** kernel.

```
Intel(R) Core(TM) i7-8559U CPU @ 2.70GHz
00:02.0 VGA compatible controller [0300]: Intel Corporation Iris Plus Graphics 655 [8086:3ea5] (rev 01)
```

uartie, 2021-12-28:
> I upgraded kernel from 5.4 to **5.11.0-43-generic** on one CFL that is having this issue. But the issue did not go away with **5.11** kernel.
> 
> ```
> Intel(R) Core(TM) i7-8559U CPU @ 2.70GHz
> 00:02.0 VGA compatible controller [0300]: Intel Corporation Iris Plus Graphics 655 [8086:3ea5] (rev 01)
> ```

Also on another system with exact same HW, we have 5.8.0-63-generic kernel and the problem still exists.

uartie, 2021-12-28:
This bug also breaks several hevc and vp9 8/10bit decode cases and all codecs for encode.  Also see this issue on other gen9 like SKL and KBL, too.

uartie, 2021-12-28:
Issue also seen using gstreamer-vaapi, gstreamer-msdk and ffmpeg-vaapi.

xuguangxin, 2021-12-29:
@XinfengZhang , could you check this?
thanks

XinfengZhang, 2021-12-29:
@pingliwa are checking this issue. will update it as soon as possible

XinfengZhang, 2022-01-14:
the commit is reverted. but still not clear with the root cause.

fanli-hub, 2022-02-07:
Add avc and mpeg2 cmd and error messages.
Encode/avc cmd:
ffmpeg -hwaccel qsv -init_hw_device qsv=hw:/dev/dri/renderD128 -hwaccel_output_format qsv -v verbose -f rawvideo -pix_fmt yuv420p -s:v 3840x2160 -r:v 30 -i /opt/media/ci/assets/otc-media/sanity/yuv/4k_2013_3840x2160_I420_75frames.yuv -vf 'format=nv12,hwupload=extra_hw_frames=120' -an -c:v h264_qsv -profile:v main -preset 4 -g 30 -slices 1 -bf 3 -b:v 10000k -maxrate 20000k -refs 1 -low_power 0 -vframes 75 -y /home/media-ci/vaapi-env/vaapi-fits-smoke/results/smoke/0b191b88-87e5-11ec-ac0d-1c697a03af23_0/_0.test.ffmpeg-qsv.encode.avc/vbr/4k2013-vbr-main-30-30-1-4-3-10000k-20000k-1-0.h264

Error massage:
[h264_qsv @ 0x7f7e94003180] Error during encoding: unknown error (-21) speed=N/A

Encode/mpeg2 cmd:
ffmpeg -hwaccel qsv -init_hw_device qsv=hw:/dev/dri/renderD128 -hwaccel_output_format qsv -v verbose -f rawvideo -pix_fmt yuv420p -s:v 720x480 -i /opt/media/ci/assets/otc-media/sanity/yuv/tcela-7-slices_720x480_I420.yuv -vf 'format=nv12,hwupload=extra_hw_frames=120' -an -c:v mpeg2_qsv -q 6 -preset 4 -g 30 -bf 2 -vframes 61 -y /home/media-ci/vaapi-env/vaapi-fits-smoke/results/smoke/1be15efa-87e8-11ec-ac0d-1c697a03af23_0/_0.test.ffmpeg-qsv.encode.mpeg2/cqp/tcela-7-slices-cqp-30-10-4-2.m2v

Error massage:
[mpeg2_qsv @ 0x7f3cf001fe00] Error during encoding: device failed (-17)speed=N/A

fanli-hub, 2022-02-16:
This issue doesn't exist now.

closed 2022-02-16 by fanli-hub