intel/media-driver #1318: iHD driver load with core dump on VF

TerrenceXu, 2021-12-28:
## System information
- CPU information: 12th Gen Intel(R) Core(TM) i9-12900E
- GPU information: 8086:4680

## Issue behavior
### Describe the current behavior 
OS is Ubuntu 18.04 or Ubuntu20.04
It is works on host, but not works on on VF (it is VM, create with GPU SRIOV)
The intel_context created failed. (InitStreamParameters:404: Invalid (nullptr) Pointer.)

### Describe the expected behavior
VF also works well

## Debug information
Use the latest libva/libva-utils/gmmlib/media-driver:

GPU hardware is existed in VF, as below:
# tree
.
├── by-path
│   ├── pci-0000:00:02.0-card -> ../card0
│   └── pci-0000:00:02.0-render -> ../renderD128
├── card0
└── renderD128

vainfo log:
error: can't connect to X server!
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /mnt/media-sdk-linux-gt/perf-tool-based-msdk/output/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_14

dmesg:
[  983.523749] vainfo[728]: segfault at 0 ip 0000000000000000 sp 00007ffdb7e47518 error 14 in vainfo[55b64d400000+6000]
[  983.523780] Code: Unable to access opcode bytes at RIP 0xffffffffffffffd6.
[strace.log](https://github.com/intel/media-driver/files/7784702/strace.log)

strace log as the attachment:

TerrenceXu, 2021-12-28:
Add "video=vesafb:off video=efifb:off" on host grub, still met this issue.

pingliwa, 2021-12-29:
Failed at mos_gem_context_create_ext, the driver parameter is correct. Need help form uapi.

TerrenceXu, 2021-12-29:
It run DRM IOCTL "I915_GEM_CONTEXT_CREATE_EXT" failed, drm.debug log as below:
[ 54.174971] [drm:drm_ioctl] comm="vainfo" pid=701, dev=0xe280, auth=1, I915_GEM_CREATE
[ 54.176435] [drm:drm_ioctl] comm="vainfo" pid=701, dev=0xe280, auth=1, I915_GEM_CONTEXT_CREATE_EXT
[ 54.176445] [drm:drm_ioctl] comm="vainfo", pid=701, ret=-5

XinfengZhang, 2022-01-14:
@zxye  to be aware of this issue. still not clear about the root cause, it is an IOCTL failure issue, maybe related with VM.

TerrenceXu, 2022-01-20:
Thanks all, I already root cause this issue is related to VF & GuC, not related to media driver.

closed 2022-01-20 by TerrenceXu