intel/media-driver #1319: 3DLUT on Icelake?

fhvwy, 2022-01-01:
## System information
`Intel(R) Core(TM) i7-1065G7 CPU @ 1.30GHz`
`00:02.0 VGA compatible controller: Intel Corporation Iris Plus Graphics G7 (rev 07)`

## Issue behavior
### Describe the current behavior
Filters show that 3DLUT is present, but it then returns no capabilities.

vadumpcaps output:
```
                        {
                            "filter": 9,
                            "name": "3DLUT",
                            "types": [
                            ],
                        },
```

### Describe the expected behavior
Either 3DLUT is present and it should return capabilities, or it isn't and the filter should not be visible at all.

The comment at https://github.com/intel/media-driver/blob/master/media_driver/linux/common/vp/ddi/media_libva_vp.c#L4517-L4518 suggests that it should be as Gen11+, but the following line then tests for Gen12+.  I don't know which is right.

## Debug information
Current master iHD, git 4485349d8f056ccc3ebde978082c056c4666f014.

FurongZhang, 2022-01-06:
As for graphics hardware capability of 3DLUT, there is no change between Gen11 and Gen12. Gen12 is the first platform we enabled from SW side, that is why we have the code https://github.com/intel/media-driver/blob/master/media_driver/linux/common/vp/ddi/media_libva_vp.c#L4517-L4518. 

I will take a look at the vadumpcaps output you mentioned.

fhvwy, 2022-01-09:
I found it confusing that the 3DLUT filter was advertised but then no configuration offered to actually use it.  Trying other cases, I see the same output on Gen9 as well - it shows the filter as present even though I'm pretty sure that doesn't have the hardware at all there.  If Gen11 isn't actually implemented (yet) then I guess this is really a capabilities reporting issue?

FurongZhang, 2022-01-17:
Will check this and get back to you.