intel/media-driver #1322: J6413 gpu enable_guc=2 works on linux 5.14, 5.15 does not

nzlov, 2022-01-05:
## System information
- CPU information(`cat /proc/cpuinfo | grep "model name" | uniq`): model name	: Intel(R) Celeron(R) J6413 @ 1.80GHz
- GPU information(`lspci -nn | grep -E 'VGA|isplay`): 00:02.0 VGA compatible controller [0300]: Intel Corporation Device [8086:4555] (rev 01)
- Display server if rendering to display(X or wayland): no

## info
```
# cat /etc/modprobe.d/i915.conf
options i915 enable_guc=2 enable_dc=0
```

## Issue behavior
### Describe the current behavior
```
# vainfo
error: can't connect to X server!
libva error: /usr/lib/dri/iHD_drv_video.so init failed
vaInitialize failed with error code -1 (unknown libva error),exit
```

### Describe the expected behavior
```
enable guc
```

## Debug information
- What's libva/libva-utils/gmmlib/media-driver version? 
   `intel-media-driver 21.4.3-1`
- Could you confirm whether GPU hardware exist or not by `ls /dev/dri`?
```
# ls -alh /dev/dri
drwxr-xr-x 2 root root         80  1月  5 10:33 by-path
crw-rw---- 1 root video  226,   0  1月  5 10:33 card0
crw-rw-rw- 1 root render 226, 128  1月  5 10:33 renderD128
```
- Could you provide vainfo log if possible by `vainfo >vainfo.log 2>&1`?
```
error: can't connect to X server!
libva error: /usr/lib/dri/iHD_drv_video.so init failed
vaInitialize failed with error code -1 (unknown libva error),exit
```
- Could you provide libva trace log if possible? Run cmd `export LIBVA_TRACE=/tmp/libva_trace.log` first then execute the case.
```
# vainfo
error: can't connect to X server!
libva info: Open new log file /tmp/libva_trace.log.023726.thd-0x00001875 for the thread 0x00001875
libva info: LIBVA_TRACE is on, save log into /tmp/libva_trace.log.023726.thd-0x00001875
libva error: /usr/lib/dri/iHD_drv_video.so init failed
vaInitialize failed with error code -1 (unknown libva error),exit

# cat /tmp/libva_trace.log.023726.thd-0x00001875
[1126.538323][ctx       none]==========va_TraceInitialize
[1126.538332][ctx       none]=========vaInitialize ret = VA_STATUS_ERROR_UNKNOWN, unknown libva error
[1126.538335][ctx       none]==========va_TraceTerminate
[1126.538345][ctx       none]=========vaTerminate ret = VA_STATUS_SUCCESS, success (no error)
```
- Could you attach dmesg log if it's GPU hang by `dmesg >dmesg.log 2>&1`?
```
[    0.000000] Linux version 5.15.12-arch1-1 (linux@archlinux) (gcc (GCC) 11.1.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP PREEMPT Wed, 29 Dec 2021 12:04:56 +0000
[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-linux root=UUID=2cdd106c-0ef1-430e-9b58-8a9e56b662be rw loglevel=3 quiet
[    0.000000] x86/split lock detection: #AC: crashing the kernel on kernel split_locks and warning on user-space split_locks
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Enabled xstate features 0x3, context size is 576 bytes, using 'compacted' format.
[    0.000000] signal: max sigframe size: 1520
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007428bfff] usable
[    0.000000] BIOS-e820: [mem 0x000000007428c000-0x00000000764f5fff] reserved
[... 946 lines cut ...]
[   13.935670] docker0: port 1(veth64edc2f) entered blocking state
[   13.935676] docker0: port 1(veth64edc2f) entered forwarding state
[   13.935761] IPv6: ADDRCONF(NETDEV_CHANGE): docker0: link becomes ready
[   13.935849] docker0: port 1(veth64edc2f) entered disabled state
[   13.966617] br-4c8bf6c32188: port 1(vethf8b744f) entered blocking state
[   13.966959] br-4c8bf6c32188: port 1(vethf8b744f) entered disabled state
[   13.967087] device vethf8b744f entered promiscuous mode
[   13.967961] br-4c8bf6c32188: port 1(vethf8b744f) entered blocking state
[   13.967966] br-4c8bf6c32188: port 1(vethf8b744f) entered forwarding state
[   13.968492] IPv6: ADDRCONF(NETDEV_CHANGE): br-4c8bf6c32188: link becomes ready
[   13.968583] br-4c8bf6c32188: port 1(vethf8b744f) entered disabled state
[   13.974945] br-7d5086f92ecf: port 1(veth8eb8c36) entered blocking state
[   13.974961] br-7d5086f92ecf: port 1(veth8eb8c36) entered disabled state
[   13.975030] device veth8eb8c36 entered promiscuous mode
[   13.978983] br-7d5086f92ecf: port 1(veth8eb8c36) entered blocking state
[   13.978990] br-7d5086f92ecf: port 1(veth8eb8c36) entered forwarding state
[   13.979020] IPv6: ADDRCONF(NETDEV_CHANGE): br-7d5086f92ecf: link becomes ready
[   13.980798] br-7d5086f92ecf: port 1(veth8eb8c36) entered disabled state
[   13.983463] br-aa911237b111: port 1(veth3239376) entered blocking state
[   13.983469] br-aa911237b111: port 1(veth3239376) entered disabled state
[   13.983544] device veth3239376 entered promiscuous mode
[   13.985585] br-aa911237b111: port 1(veth3239376) entered blocking state
[   13.985591] br-aa911237b111: port 1(veth3239376) entered forwarding state
[   13.985660] IPv6: ADDRCONF(NETDEV_CHANGE): br-aa911237b111: link becomes ready
```

XinfengZhang, 2022-01-07:
it looks a guc enabling issue, @zxye , could you help to take a look?

fhvwy, 2022-01-09:
See https://gitlab.freedesktop.org/drm/intel/-/issues/4067 - HuC on gen11 is entirely broken on the Linux 5.15 stable branch because of this.

XinfengZhang, 2022-01-10:
thanks @fhvwy to figure out the issue

XinfengZhang, 2022-01-13:
@nzlov , could we close this issue?

closed 2022-01-13 by nzlov