intel/media-driver #1325: Generated vpkrnheader.h is missing some definitions needed to compile the driver

frantisekz, 2022-01-14:
## System information
media-driver-22.1.1
cm-compiler-1.0.119
intel-graphics-compiler-1.0.9636
compute-runtime-21.50.21939
gmmlib-22.0.1
opencl-clang-13.0.0
gcc-11.2.1
cmake-3.22.1
libva-2.13.0

## Issue behavior
### Describe the current behavior

I am working on getting the stripped, free versions of the intel-media-driver into Fedora ( https://pagure.io/fedora-workstation/issue/209#comment-771853 , also, if anybody is proficient with Fedora/rpm build process, there are references to 1:1 replicate my build environment). I am currently at the point when I am able to get a working driver that might be included in future Fedora releases.

The cmake command I am using is:
`/usr/bin/cmake -S . -B redhat-linux-build -DCMAKE_C_FLAGS_RELEASE:STRING=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE:STRING=-DNDEBUG -DCMAKE_Fortran_FLAGS_RELEASE:STRING=-DNDEBUG -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON -DCMAKE_INSTALL_DO_STRIP:BOOL=OFF -DCMAKE_INSTALL_PREFIX:PATH=/usr -DINCLUDE_INSTALL_DIR:PATH=/usr/include -DLIB_INSTALL_DIR:PATH=/usr/lib64 -DSYSCONF_INSTALL_DIR:PATH=/etc -DSHARE_INSTALL_PREFIX:PATH=/usr/share -DLIB_SUFFIX=64 -DBUILD_SHARED_LIBS:BOOL=ON -DENABLE_NONFREE_KERNELS:BOOL=OFF -DBUILD_KERNELS:BOOL=ON -DMEDIA_BUILD_FATAL_WARNINGS=OFF -Wno-dev -G Ninja`

I can upload full compilation logs if that's needed. Also worht of mentioning, I am using -O0, because GenKrn is crashing otherwise (I'll create an issue for that too later on).

However, I've hit a few bumps in the driver build.  I am not sure yet if they're connected somehow. To state upfront, the goal is to get the driver built from free-only sources, however, these issues happened even on non-stripped tarball, that is https://github.com/intel/media-driver/archive/intel-media-22.1.1.tar.gz

The generated media_driver/agnostic/common/vp/kernel/vpkrnheader.h is missing some definitions that are required to build the driver, these are (resulted in "not defined in scope compiler errors"):
- IDR_VP_HDR_preprocess
- IDR_VP_Call_XOR_Mono_Composite
- IDR_VP_Save_444Scale16_P016
- IDR_VP_Save_444Scale16_SrcY416

For the time-being, I've duct-tape patched them into the .c files in the build process which resulted in a working driver. The affected files are:
- media_driver/agnostic/gen9/vp/hal/vphal_render_hdr_g9_base.cpp
- media_driver/agnostic/gen11/vp/kdll/hal_kernelrules_g11.c
- media_driver/agnostic/gen12_tgllp/vp/kdll/hal_kernelrules_g12lpcmfc.c
- media_driver/agnostic/gen12_tgllp/vp/kdll/hal_kernelrules_g12lp.c

So, the related to this issue may or may not be the fact I had to patch around a few things in the build process, otherwise, I was getting errors about needed files not existing (for which I can create a separate issue if that's desired):

```
---
 media_driver/agnostic/media_srcs.cmake | 5 ++++-
 1 file changed, 4 insertions(+), 1 deletion(-)

diff --git a/media_driver/agnostic/media_srcs.cmake b/media_driver/agnostic/media_srcs.cmake
index 141eeab..274a1d7 100755
--- a/media_driver/agnostic/media_srcs.cmake
+++ b/media_driver/agnostic/media_srcs.cmake
@@ -82,12 +82,14 @@ if (BUILD_KERNELS)
             DEPENDS GenKrnBin ${hexs} ${link_file}
             COMMAND ${CMAKE_COMMAND} -E copy ${link_file} ./
             COMMAND GenKrnBin . ${name} ${genx} ${kind}
+            COMMAND ${CMAKE_COMMAND} -E copy igvpkrn_11_icllp.h ${krn}.h
             COMMAND ${CMAKE_COMMAND} -E copy ${krn}.h ${krn_header})
 
         # Generating kernel from the .bin file
         add_custom_command(
             OUTPUT ${krn_dir}/${krn}.c ${krn_dir}/${krn}.h
             DEPENDS KernelBinToSource ${krn}.bin
+            COMMAND ${CMAKE_COMMAND} -E copy igvpkrn_11_icllp.bin ${krn}.bin
             COMMAND KernelBinToSource -i ${krn}.bin -o ${krn_dir}/)
     endfunction()
 
@@ -169,7 +171,8 @@ if (BUILD_KERNELS)
             OUTPUT ${krn_dir}/${krn}.c ${krn_dir}/${krn}.h
             DEPENDS KernelBinToSource ${CMAKE_CURRENT_LIST_DIR}/common/codec/kernel/merge.py ${dats}
             WORKING_DIRECTORY ${out_dir}
-            COMMAND ${PYTHON} ${CMAKE_CURRENT_LIST_DIR}/common/codec/kernel/merge.py -o ${krn}.krn ${dats}
+            #COMMAND ${PYTHON} ${CMAKE_CURRENT_LIST_DIR}/common/codec/kernel/merge.py -o ${krn}.krn ${dats}
+            COMMAND ${PYTHON} ${CMAKE_CURRENT_LIST_DIR}/common/codec/kernel/merge.py -o ${krn}.krn downscale_kernel_genx.isa downscale_convert_kernel_genx.isa hme_kernel_genx.isa
             # ${index} is needed to match a description in the following file:
             #   media_driver/agnostic/common/codec/kernel/codeckrnheader.h
             COMMAND KernelBinToSource -i ${krn}.krn -o ${krn_dir}/ -v ${krn} -index ${index} -t ${IDR_CODEC_TOTAL_NUM_KERNELS}
-- 
```

### Describe the expected behavior
Missing definitions should be either present in the generated media_driver/agnostic/common/vp/kernel/vpkrnheader.h or shouldn't be required for a successful compilation.

## Debug information
Versions included above.

Thanks!

frantisekz, 2022-01-17:
> @frantisekz : You can consider breaking the task into 2 steps. First you can build the driver stripping all the shaders (EU kernels) away. See https://github.com/intel/media-driver/wiki/Media-Driver-Shaders-(EU-Kernels)#disable-shaders-at-compile-time. 

Hmm, I was aiming for free kernels/shaders rebuilt path, didn't know it could work without any of them. Anyhow, I didn't check the wiki and the script to delete the non-free shaders is very useful (if it's getting updated when there are new non-free shaders added to the source tree?). I was glueing together some custom nonfree file removal tool for package inclusion. Thanks for the link!

> By the way, rebuilding shaders requires CM and IGC compilers. Do you plan to package these as well or just use them for the build of media-driver?

Yes, I've prepared review requests for all these packages and dependencies. IGC is a prerequisite for OpenCL (compute-runtime, once https://github.com/intel/intel-graphics-compiler/issues/204 gets resolved) so that's a nice side-goal getting accomplished too. The dependency chain is rather on the complicated side, but I won't be maintaining that alone, hopefully.

dvrogozh, 2022-01-17:
We are trying to keep this wiki updated, but something might squeeze in, so if you will spot some issue - let us know. Right now DG2/ATSM upstreaming is happening, so it really might.

MicroYY, 2022-01-18:
I already have an internal fix but not promoted. Will notify you when it's ready.

closed 2022-01-28 by intel-mediadev

frantisekz, 2022-02-18:
@MicroYY Thanks, it seems, however, with #1328 included (22.2.1)  that I am not getting any .fcpatch files in media_driver/agnostic/gen12_tgllp/vp/kernel_free/cache_kernel/ , shall I create a new issue for this?

It's then crashing on (bunch of similar errors just for different files):

```
7 warnings generated.
	Kernel PL2_444AVS16_Buf_0_Rot_90 : 127 registers
upperFile: /builddir/build/BUILD/media-driver-intel-media-22.2.1/media_driver/agnostic/gen12_tgllp/vp/kernel_free/cache_kernel/PL2_444AVS16_Buf_0_Rot_90_0.dat
Upper file/builddir/build/BUILD/media-driver-intel-media-22.2.1/media_driver/agnostic/gen12_tgllp/vp/kernel_free/cache_kernel/PL2_444AVS16_Buf_0_Rot_90_0.dat exists! No need to rename!
upperFile: /builddir/build/BUILD/media-driver-intel-media-22.2.1/media_driver/agnostic/gen12_tgllp/vp/kernel_free/cache_kernel/PL2_444AVS16_Buf_0_Rot_90.fcpatch
Traceback (most recent call last):
  File "/builddir/build/BUILD/media-driver-intel-media-22.2.1/media_driver/agnostic/gen12_tgllp/vp/kernel_free/Source/rename.py", line 18, in <module>
    change_mistake(sys.argv[1], sys.argv[3])
  File "/builddir/build/BUILD/media-driver-intel-media-22.2.1/media_driver/agnostic/gen12_tgllp/vp/kernel_free/Source/rename.py", line 13, in change_mistake
    os.rename(lowerFile, upperFile)
FileNotFoundError: [Errno 2] No such file or directory: '/builddir/build/BUILD/media-driver-intel-media-22.2.1/media_driver/agnostic/gen12_tgllp/vp/kernel_free/cache_kernel/PL2_444AVS16_Buf_0_Rot_90.fcpatch' -> '/builddir/build/BUILD/media-driver-intel-media-22.2.1/media_driver/agnostic/gen12_tgllp/vp/kernel_free/cache_kernel/PL2_444AVS16_Buf_0_Rot_90.fcpatch'

```

If I hack around it with try/except, it then fails on cmake copy failing to find the .fcpatch file.

MicroYY, 2022-03-01:
> @MicroYY Thanks, it seems, however, with #1328 included (22.2.1) that I am not getting any .fcpatch files in media_driver/agnostic/gen12_tgllp/vp/kernel_free/cache_kernel/ , shall I create a new [...]

Hi @frantisekz  Could you pls file another issue for it?

frantisekz, 2022-03-03:
Created https://github.com/intel/media-driver/issues/1356 .