intel/media-driver #1330: Cannot get libva working with vainfo on Ubuntu20.04 : "libva info: va_openDriver() returns 18"

WARUKUCHI, 2022-01-21:
Hi,

I am trying to use the MSDK on a ubuntu20.04 but failed.Then I found that when I typed vainfo, the expected result was not output.
I follow the official steps to build the latest version of MSDK or use the script of one-click deployment , and export iHD_drv_video.so related LIBVA environment variables, yet both failed.

`root@ubuntu:~# cd /usr/lib/x86_64-linux-gnu/dri/
root@ubuntu:/usr/lib/x86_64-linux-gnu/dri# ls
d3d12_dri.so  i915_dri.so  iHD_drv_video.so  kms_swrast_dri.so  nouveau_vieux_dri.so  r300_dri.so  radeon_dri.so    swrast_dri.so      vmwgfx_dri.so
i830_dri.so   i965_dri.so  iris_dri.so       nouveau_dri.so     r200_dri.so           r600_dri.so  radeonsi_dri.so  virtio_gpu_dri.so  zink_dri.so`

Here is my configuration :

`root@ubuntu:~# lspci -nn|grep VGA
03:00.0 VGA compatible controller [0300]: ASPEED Technology, Inc. ASPEED Graphics Family [1a03:2000] (rev 41)
root@ubuntu:~# ls /dev/dri/
by-path  card0
root@ubuntu:~# uname -r
5.4.0-26-generic
root@ubuntu:~# cat /proc/cpuinfo | grep "model name" | uniq
model name      : Intel(R) Xeon(R) Gold 5218 CPU @ 2.30GHz
root@ubuntu:~# dmesg
[    0.000000] Linux version 5.4.0-26-generic (buildd@lcy01-amd64-029) (gcc version 9.3.0 (Ubuntu 9.3.0-10ubuntu2)) #30-Ubuntu SMP Mon Apr 20 16:58:30 UTC 2020 (Ubuntu 5.4.0-26.30-generic 5.4.30)[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.4.0-26-generic root=UUID=e2099faf-0637-4c3c-8b8d-17ea59daa7e2 ro
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai`

Do you have any solutions?Thx.

WARUKUCHI, 2022-01-24:
Here's the vainfo result.

```
root@ubuntu:~# vainfo
error: can't connect to X server!
libva info: VA-API version 1.12.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_12
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
libva error: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 18
vaInitialize failed with error code 18 (invalid parameter),exit
```

XinfengZhang, 2022-01-27:
03:00.0 VGA compatible controller [0300]: ASPEED Technology, Inc. ASPEED Graphics Family [1a03:2000] (rev 41)
you are using ASPEED card?  
intel media driver just support intel graphics.

WARUKUCHI, 2022-01-28:
Thank you. Is it all Intel graphics? Or only a part of it can be supported.

XinfengZhang, 2022-01-28:
https://github.com/intel/media-driver#supported-platforms
if you want older platform support, please use https://github.com/intel/intel-vaapi-driver

XinfengZhang, 2022-01-28:
so, suppose this issue could be closed?

WARUKUCHI, 2022-02-08:
Thank you for your reply, but I have other questions. Can Intel media-driver be used in KVM?

eero-t, 2022-02-08:
> Thank you for your reply, but I have other questions. Can Intel media-driver be used in KVM?

Yes, if you correctly pass-through a PCI device supported by the media-driver.  However, each separate issue should have a separate ticket.  How to configure KVM does not belong to media-driver bug tracker though, but to a KVM one (if your KVM is not documenting it well enough).

WARUKUCHI, 2022-02-10:
I see. I'll close this and open an new issue. Thank you for your reply.

closed 2022-02-10 by WARUKUCHI