intel/media-driver #1337: [regression]Hardware accelerator failed to decode picture

fanli-hub, 2022-01-27:
Platform: ADLP/DG2/ATSM
Kernel: 5.14.0-rc1-CI-CI_DRM_10344+ (ADLP) 
Firmware: 1.187.17
Libva: 453002ce69779c713f6f8315bedce140d34ba805
Gmmlib: 70d68b9f7146ef18a2522ecf1ece8075e8d8c207
libva-utils: 7bad184b2cf2ffaf4fb3cc71d4df63d7b142d592
onevpl dispatcher: b4094560c82cdf3c5f5cd04dcda01ab9edfec14b
onevpl runtime: 6676f7652249bb20ad51d229012f8a6ef26f67b0
mediaSDK: 0db73aa1b2dec5b61900ee54db3fa574de6d1752
cartwheel-ffmpeg: 0d7f26ade04bad5df3ee152e85cd241ba3200523
Good media-driver: 4debb74628c1dd6e05370804e6e0dfcaf9e9adb5 
Bad media-driver: 697e7102ca41574899d8633417629c2d4ee1ea7a 

**Reproduce cmd:**
ffmpeg -hwaccel vaapi -init_hw_device vaapi=hw:/dev/dri/renderD128 -hwaccel_output_format p012 -hwaccel_flags allow_profile_mismatch -v verbose -i /home/media-ci/assets/otc-media/sanity/hevc/12bit/hevc_420_12_lcu32_Kungfu_4096x8192.bin -pix_fmt p012 -f rawvideo -vsync passthrough -autoscale 0 -vframes 2 -y /home/media/hevc_420_12_lcu32_Kungfu_4096x8192_4096x8192_P012.yuv

**Error message:**
[hevc @ 0x5622e24671c0] Failed to end picture decode issue: 23 (internal decoding error).
[hevc @ 0x5622e24671c0] hardware accelerator failed to decode picture
[hevc @ 0x5622e2462b80] Could not find ref with POC 0
[hevc @ 0x5622e2462b80] Failed to end picture decode issue: 23 (internal decoding error).
[hevc @ 0x5622e2462b80] hardware accelerator failed to decode picture
Error while decoding stream #0:0: Input/output error
    Last message repeated 1 times
[graph 0 input from stream 0:0 @ 0x5622e267dc40] w:4096 h:8192 pixfmt:yuv420p12le tb:1/1200000 fr:25/1 sar:0/1
[auto_scale_0 @ 0x5622e2681040] w:iw h:ih flags:'' interl:0
[format @ 0x5622e267ed40] auto-inserting filter 'auto_scale_0' between the filter 'Parsed_null_0' and the filter 'format'
Impossible to convert between the formats supported by the filter 'Parsed_null_0' and the filter 'auto_scale_0'
Error reinitializing filters!
Error while filtering: Function not implemented
Finishing stream 0:0 without any data written to it.
[graph 0 input from stream 0:0 @ 0x5622e2680b80] w:4096 h:8192 pixfmt:yuv420p12le tb:1/1200000 fr:25/1 sar:0/1
[auto_scale_0 @ 0x5622e2683980] w:iw h:ih flags:'' interl:0
[format @ 0x5622e267e4c0] auto-inserting filter 'auto_scale_0' between the filter 'Parsed_null_0' and the filter 'format'
Impossible to convert between the formats supported by the filter 'Parsed_null_0' and the filter 'auto_scale_0'
Error configuring filter graph

fanli-hub, 2022-02-14:
@huangli2018,  could you check this?
thanks

Jexu, 2022-02-24:
Could you try the latest version media driver and CI_DRM build kernel since some fix are merged into both from the Bad version.

fanli-hub, 2022-03-01:
This issue can't be reproduced. I am going to close this issue.