intel/media-driver #1338: new versions of clang warn for -Wbitwise-instead-of-logical

mikebenfield, 2022-01-28:
```
media_driver/agnostic/common/codec/hal/codechal_decode_vc1.cpp:3225:10: error: use of bitwise '|' with boolean operands [-Werror,-Wbitwise-instead-of-logical]
        (m_mfxInterface->IsVc1BPicture(currPic, m_vc1PicParams->picture_fields.is_first_field, picType) |
        ~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

This happens for instance with LLVM commit 18308e171b5b1dd99627a4d88c7d6c5ff21b8c96 and newer.

dvrogozh, 2022-01-28:
That's with clang build from llvm master? not yet released version?

mikebenfield, 2022-01-28:
Yes, I think this warning does not yet happen in the most recent clang release.

commit 4834d8dd (2022-02-07): Change | to || for booleans

This will silence the warning -Wbitwise-instead-of-logical for newer version of clang.

fixes #1338
closed 2022-02-07 by intel-mediadev