intel/media-driver #1339: Compiling errors?

harwck, 2022-01-30:
```
[ 68%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/gen12/ddi/media_libva_caps_g12.cpp.o
[ 68%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/gen12/ddi/media_sysinfo_g12.cpp.o
[ 69%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/gen12/ddi/media_sku_wa_g12.cpp.o
/home/$USERNAME/workspace/media-driver/media_driver/linux/gen12/ddi/media_sysinfo_g12.cpp: In function ‘bool InitTglShadowWa(GfxDeviceInfo*, SHADOW_MEDIA_WA_TABLE*, LinuxDriverInfo*)’:
/home/$USERNAME/workspace/media-driver/media_driver/linux/gen12/ddi/media_sysinfo_g12.cpp:164:14: error: ‘using SHADOW_MEDIA_WA_TABLE = WA_TABLE’ {aka ‘struct _WA_TABLE’} has no member named ‘WaAuxTable64KGranular’; did you mean ‘WaAuxTable16KGranular’?
  164 |     waTable->WaAuxTable64KGranular = 1;
      |              ^~~~~~~~~~~~~~~~~~~~~
      |              WaAuxTable16KGranular
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:8383: media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/gen12/ddi/media_sysinfo_g12.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:253: media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/all] Error 2
make: *** [Makefile:152: all] Error 2

```

## System information
- CPU information(`cat /proc/cpuinfo | grep "model name" | uniq`): **Intel(R) Core(TM) i7-8565U CPU @ 1.80GHz**
- GPU information(`lspci -nn | grep -E 'VGA|isplay'`): **Intel Corporation UHD Graphics 620 (Whiskey Lake)**
- Display server if rendering to display(X or wayland): **X11**

## Issue behavior

- Compiling errors.

### Describe the current behavior
- Failing at 68%

### Describe the expected behavior
- Successful compilation

## Debug information
- What's libva/libva-utils/gmmlib/media-driver version?
*VA-API version: 1.14 (libva 2.13.0.pre1)*
media driver: commit ae09cc33e
intel-gmmlib-22.0.2 commit 9290546
- Could you confirm whether GPU hardware exist or not by `ls /dev/dri`?
by-path  card0  renderD128
- Could you provide vainfo log if possible by `vainfo >vainfo.log 2>&1`?
```
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_14
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.14 (libva 2.13.0.pre1)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.2.0 (daeea8255)
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointFEI
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSlice
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
```
- Could you provide libva trace log if possible? Run cmd `export LIBVA_TRACE=/tmp/libva_trace.log` first then execute the case.
I would like to but compiling takes quite a bit of time...
- Could you attach dmesg log if it's GPU hang by `dmesg >dmesg.log 2>&1`?
N/A
- Do you want to contribute a patch to fix the issue? (yes/no):
Can't

ablanco-stratio, 2022-02-01:
Same error in my case.

DotIN13, 2022-02-01:
Commenting out `waTable->WaAuxTable64KGranular = 1;` in `media_driver/linux/gen12/ddi/media_sysinfo_g12.cpp` enabled me to pass the compilation.

dvrogozh, 2022-02-01:
What cmake configuration cmdline you were using?

DotIN13, 2022-02-01:
Plain old `mkdir build && cd build` and `cmake ..`.

dvrogozh, 2022-02-01:
Driver builds fine on my side. This variable which was not found (`WaAuxTable64KGranular`) is declared in https://github.com/intel/gmmlib which is media-driver build dependency. Please, make sure that you've built and installed gmmlib on the system and pointed to gmmlib installation via PKG_CONFIG_PATH if you did not install it system wide (which would be the case if you've used bare `cmake ..`, you might need to use `cmake -DCMAKE_INSTALL_PREFIX=... -DCMAKE_INSTALL_LIBDIR=...`).
```
$ fgrep -rsn WaAuxTable64KGranular /usr/include/
/usr/include/igdgmm/inc/common/sku_wa.h:483:        WaAuxTable64KGranular,

```

https://github.com/intel/gmmlib/blob/9290546f493656d80a4aa05b63dfeaf9f8527b0a/Source/inc/common/sku_wa.h#L483

DotIN13, 2022-02-01:
Ohh, that should be the problem, I’m using gmmlib shipped with manjaro.

harwck, 2022-02-01:
@dvrogozh Thanks for the help, worked as expected!

closed 2022-02-01 by harwck