intel/media-driver #1341: ADLS E2E Compression causing problems

oCanna, 2022-01-30:
Alder Lake S 12th gen iGPU suffers from the same problem when e2e compression is enabled.
Mobile processors are probably affected as well.

https://github.com/intel/media-driver/commit/29d5f8fc118476dee84f3f784443badc819a3241
https://github.com/intel/media-driver/issues/1210

```
--- intel-media-driver-22.1.1.orig/media_driver/linux/gen12/ddi/media_sku_wa_g12.cpp
+++ intel-media-driver-22.1.1/media_driver/linux/gen12/ddi/media_sku_wa_g12.cpp
@@ -509,6 +509,9 @@ static bool InitAdlsMediaSku(struct GfxD
     //Disable VP8 for ADLS
     MEDIA_WR_SKU(skuTable, FtrIntelVP8VLDDecoding, 0);

+    // turn off Compression on ADLS as HW known issue
+    MEDIA_WR_SKU(skuTable, FtrE2ECompression, 0);
+
     return true;
 }

```

XinfengZhang, 2022-02-07:
it should already been fixed by https://github.com/intel/media-driver/commit/ed0876c2af7741c32facf15245391c581ae64b5a , right?

XinfengZhang, 2022-02-10:
@oCanna , it should already been fixed. driver still enable compression , but will disable it with default surface creation in vaCreateSurfaces. could we close it?

oCanna, 2022-02-10:
Sorry for the late response. At the moment I have no time to test the master branch. The 22.1.1 rls was the latest I have tested, but if it is already fixed like you said without completely disabling the compression, then sure, this can be closed.

Thank you!