intel/media-driver #1343: Encoding with hevc_vaapi FFmpeg on RKL GT1 results in garbage

lat9nq, 2022-02-08:
## System information
Output of `inxi -GSC -xx`:
```
System:
  Host: Qykopi Kernel: 5.15.18-200.fc35.x86_64 x86_64 bits: 64 compiler: gcc
    v: 2.37-10.fc35 Desktop: GNOME 41.3 tk: GTK 3.24.31 wm: gnome-shell dm: GDM
    Distro: Fedora release 35 (Thirty Five)
CPU:
  Info: 6-core model: 11th Gen Intel Core i5-11600K bits: 64 type: MT MCP
    arch: Rocket Lake rev: 1 cache: L1: 480 KiB L2: 3 MiB L3: 12 MiB
  Speed (MHz): avg: 4478 high: 4615 min/max: 800/4900 cores: 1: 4165
    2: 4600 3: 4615 4: 4600 5: 4600 6: 4541 7: 4552 8: 4600 9: 3957 10: 4308
    11: 4600 12: 4600 bogomips: 93888
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
Graphics:
  Device-1: Intel RocketLake-S GT1 [UHD Graphics 750] vendor: ASUSTeK
    driver: i915 v: kernel bus-ID: 00:02.0 chip-ID: 8086:4c8a
  Device-2: NVIDIA GM200 [GeForce GTX 980 Ti] vendor: ZOTAC driver: nvidia
    v: 470.94 bus-ID: 01:00.0 chip-ID: 10de:17c8
  Device-3: AMD Tonga PRO [Radeon R9 285/380] vendor: XFX Pine
    driver: amdgpu v: kernel bus-ID: 04:00.0 chip-ID: 1002:6939
  Device-4: NVIDIA GK104GL [Quadro K4200] driver: nvidia v: 470.94
    bus-ID: 06:00.0 chip-ID: 10de:11b4
  Display: x11 server: X.Org 1.20.14 compositor: gnome-shell driver:
    loaded: amdgpu,ati,modesetting,nvidia unloaded: fbdev,nouveau,vesa
    alternate: nv resolution: 1: 1920x1200~60Hz 2: 2560x1440 s-dpi: 96
  OpenGL: renderer: Mesa Intel Graphics (RKL GT1) v: 4.6 Mesa 21.3.5
    direct render: Yes
```

## Issue behavior
### Describe the current behavior

Attempting to use the `hevc_vaapi` encoder with FFmpeg results in garbage. `hevc_qsv` works fine, by contrast. 

Output of `ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -hwaccel_output_format vaapi -ss 23 -i 2022020716313100-CBA841B50A92A904E313AE06DF4EF71A.mp4 -r 30 -vf scale_vaapi=-1:450 -c:v hevc_vaapi -qp 35 -b:v 0 -c:a libopus -b:a 64k 2022020716313100-CBA841B50A92A904E313AE06DF4EF71A.hevc.mp4`:

- GitHub attempts to but can't embed HEVC, so I formatted the link in text: `https://user-images.githubusercontent.com/22451773/152902414-081ce5f8-ed6d-4af6-933f-9a32915539c3.mp4`

Inserting `-low_power 1` after `-c:v hevc_vaapi` causes the following:
```
[hevc_vaapi @ 0x5641092c93c0] Failed to end picture encode issue: 24 (internal encoding error). 
[hevc_vaapi @ 0x5641092c93c0] Encode failed: -5.
Video encoding failed
```

### Describe the expected behavior
To show VAAPI works, here's the output of `ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -hwaccel_output_format vaapi -ss 23 -i 2022020716313100-CBA841B50A92A904E313AE06DF4EF71A.mp4 -r 30 -vf scale_vaapi=-1:450 -c:v h264_vaapi -qp 35 -b:v 0 -c:a libopus -b:a 64k 2022020716313100-CBA841B50A92A904E313AE06DF4EF71A.h264.mp4`: 

https://user-images.githubusercontent.com/22451773/152901460-563f7b98-d5ff-4eb3-b2b2-b735d34bf101.mp4

To show the HEVC encoding hardware is working, output of `ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -hwaccel_output_format vaapi -ss 23 -i 2022020716313100-CBA841B50A92A904E313AE06DF4EF71A.mp4 -r 30 -vf scale_vaapi=-1:450,hwdownload,format=nv12 -c:v hevc_qsv -b:v 2M -c:a libopus -b:a 64k 2022020716313100-CBA841B50A92A904E313AE06DF4EF71A.hevc.qsv.mp4`:

- GitHub attempts to but can't embed HEVC, so I formatted the link in text: `https://user-images.githubusercontent.com/22451773/152901654-a20f2dc5-db7b-46a5-a91e-875b3bb4915d.mp4`

## Debug information
- What's libva/libva-utils/gmmlib/media-driver version?
```
$ dnf list installed | grep libva
libva.i686                                             2.13.0-3.fc35                          @updates                                              
libva.x86_64                                           2.13.0-3.fc35                          @updates                                              
libva-devel.x86_64                                     2.13.0-3.fc35                          @updates                                              
libva-intel-driver.x86_64                              2.4.1-6.fc35                           @rpmfusion-free                                       
libva-utils.x86_64                                     2.13.0-1.fc35                          @fedora                                               
libva-vdpau-driver.x86_64                              0.7.4-110.fc35                         @fedora
$ dnf list installed | grep gmmlib
intel-gmmlib.x86_64                                    21.3.5-1.fc35                          @updates
$ dnf list installed | grep media-driver
intel-media-driver.x86_64                              21.4.3-1.fc35                          @rpmfusion-nonfree-updates
```
- Could you confirm whether GPU hardware exist or not by `ls /dev/dri`?
`by-path  card0  card1  card2  card3  renderD128  renderD129  renderD130  renderD131`

- Could you provide vainfo log if possible by `vainfo >vainfo.log 2>&1`?
[vainfo.log](https://github.com/intel/media-driver/files/8020015/vainfo.log)

- Could you provide libva trace log if possible? Run cmd `export LIBVA_TRACE=/tmp/libva_trace.log` first then execute the case.
First case: [libva_trace.log.015828.thd-0x00017.tar.gz](https://github.com/intel/media-driver/files/8020034/libva_trace.log.015828.thd-0x00017.tar.gz)
With `-low_power 1`: [libva_trace.log.015728.thd-0x00017a.tar.gz](https://github.com/intel/media-driver/files/8020029/libva_trace.log.015728.thd-0x00017a.tar.gz)

- ~~Could you attach dmesg log if it's GPU hang by `dmesg >dmesg.log 2>&1`?~~ N/A
- Do you want to contribute a patch to fix the issue? (yes/no): No, I don't have any understanding of the media driver or the underlying issue.

babam86, 2022-03-06:
I also got that glitch on Intel UHD G4 and Iris Xe (hevc_vaapi), but on Windows 10 it works fine (hevc_qsv).

Thanks.

babam86, 2022-03-28:
Build FFmpeg with this patch https://github.com/intel-media-ci/cartwheel-ffmpeg/blob/master/patches/0025-lavc-vaapi_encode_h265-fix-max_transform_hierarchy_d.patch

XeCycle, 2022-04-10:
Above patch fixes the same problem for me, but on Alder Lake (i5 12400).  Applied that patch on Arch Linux PKGBUILD 2:5.0-6.

Xiaogangli-intel, 2022-04-18:
Thank you, @lat9nq and @babam86, we are working on this issue. Will let you know once we have founding.

aidan2020sh, 2022-05-11:
@lat9nq 
     i cannot reproduce the issue with latest FFmpeg build from repo: https://github.com/FFmpeg/FFmpeg.git. Could you please confirm the latest status on your side?

aidan2020sh, 2022-05-11:
the patch [0025-lavc-vaapi_encode_h265-fix-max_transform_hierarchy_d.patch](https://github.com/intel-media-ci/cartwheel-ffmpeg/blob/master/patches/0025-lavc-vaapi_encode_h265-fix-max_transform_hierarchy_d.patch) is not in the pool any longer, it should already be merged.

feiwan1, 2022-05-11:
> the patch [0025-lavc-vaapi_encode_h265-fix-max_transform_hierarchy_d.patch](https://github.com/intel-media-ci/cartwheel-ffmpeg/blob/master/patches/0025-lavc-vaapi_encode_h265-fix-max_transform_hierarchy_d.patch) is not in the pool any longer, it should already be merged.

Yes, this patch already upstreamed. @lat9nq , maybe you can confirm by using latest ffmpeg/libva/meida-driver to check if the issue still there.

Xiaogangli-intel, 2022-05-12:
Hi @lat9nq, Could you try the latest FFmpeg? if cannot reproduce this issue, please help close it. Thanks a lot!

lat9nq, 2022-05-12:
I am still able to reproduce the original issue using `LD_LIBRARY_PATH=/home/lat9nq/repos/FFmpeg/build/out/usr/local/lib /home/lat9nq/repos/FFmpeg/build/out/usr/local/bin/ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -hwaccel_output_format vaapi -ss 23 -i 2022020716313100-CBA841B50A92A904E313AE06DF4EF71A.mp4 -r 30 -vf scale_vaapi=-1:450 -c:v hevc_vaapi -qp 35 -b:v 0 -c:a libopus -b:a 64k 2022020716313100-CBA841B50A92A904E313AE06DF4EF71A.hevc.mp4`.

FFmpeg is checked out to `cbc1b8adad (HEAD -> master, origin/master, origin/HEAD) avfilter/af_biquads: add zdf transform type`

Configuration is: `--enable-libvorbis --enable-libopus --enable-libaom --enable-libx264 --enable-libx265 --enable-pthreads --disable-static --enable-shared --enable-gpl --disable-debug --enable-opengl --cpu=rocketlake --enable-vulkan --enable-libglslang --enable-libsoxr --enable-nonfree --enable-libmfx`

Let me know if I configured something wrong here. The FFmpeg configuration is just something I had for another project, and it happens to support VAAPI I believe.

Xiaogangli-intel, 2022-05-12:
@aidan2020sh Please help to verify the configuration of @lat9nq again, and check is there any difference between yours and @lat9nq's, thank you!

aidan2020sh, 2022-05-12:
@lat9nq  my ffmpeg test command is same as yours, the ffmpeg version is "[fate/matroska: Add tests for muxing PGS into Matroska](https://github.com/FFmpeg/FFmpeg/commit/f98b6cf543cc41037b18dbf4d9fa72e426383f16)" .

my ffmpeg configuration is:  ./configure --enable-libmfx --enable-libopus --enable-gpl --enable-libx265 
media driver & msdk are both quite new, you can try updating the code base and use the latest build;
i915 kernel version is: y21_ww47.2_kernel_5.15.2

BTW, there are two kernel boot up configuration which also need be checked on your system: "i915.enable_guc=2 i915.force_probe=*", without it, guc&huc will be in off status and cannot try lowpower option. You can try below command to check the guc&huc status on your system:
cat /sys/kernel/debug/dri/0/gt/uc/huc_info
cat /sys/kernel/debug/dri/0/gt/uc/guc_info

lat9nq, 2022-05-14:
It seemed that only `i915.enable_guc=2` was necessary? `cat /sys/kernel/debug/dri/0/gt/uc/huc_info` and `cat /sys/kernel/debug/dri/0/gt/uc/guc_info` reported that they were running since that setting was already present, but I went and added `i915.force_probe=*` anyway, and it still seems fine.

Media Driver currently passes CMake configuration stages, but building fails due to [missing symbols](https://gist.github.com/lat9nq/d539749a3d944002ac7f5cfe521befed) in Fedora 35's libva version -- 1.13.0 installed at the moment even though 1.8.0 is required. Pulling from https://github.com/intel/libva ~~and hacking CMakeLists.txt so specifying CMAKE_PREFIX_PATH will work with an alternate libva install location~~ fixes it. These appear to be a separate issues. Let me know if I should open those (I haven't checked open Issues for this, yet).

The FFmpeg command now works with the configuration @aidan2020sh provided. Specfically:
`LD_LIBRARY_PATH=$HOME/.test/usr/local/lib:$HOME/.test/usr/local/lib64 $HOME/.test/usr/local/bin/ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -hwaccel_output_format vaapi -ss 23 -i 2022020716313100-CBA841B50A92A904E313AE06DF4EF71A.mp4 -r 30 -vf scale_vaapi=-1:450 -c:v hevc_vaapi -qp 35 -b:v 0 -c:a libopus -b:a 64k 2022020716313100-CBA841B50A92A904E313AE06DF4EF71A.hevc.mp4`
Where `~/.test/usr/local` has FFmpeg, libva, gmmlib, media-driver, and Media-SDK installed in. Not sure who to thank but glad this works now!

closed 2022-05-14 by lat9nq

Xiaogangli-intel, 2022-05-15:
@lat9nq,  that's great it works on your side. Thanks @aidan2020sh help debug this issue.