intel/media-driver #1344: GCC v11 warnings for CM component

eero-t, 2022-02-08:
## System information
- Ubuntu 21.10 container with GCC v11.2.0

## Issue behavior
### Describe the current behavior

```
cmake -LH -Wno-dev -G Ninja -DCMAKE_INSTALL_PREFIX=${INSTALL_DIR} .  &&  ninja
...
// Turn compiler warnings into fatal errors
MEDIA_BUILD_FATAL_WARNINGS:BOOL=ON
...
../media_driver/agnostic/common/cm/cm_mem_sse2_impl.cpp: In function 'void FastMemCopy_SSE2_movntdq_movdqa(void*, void*, size_t)':
../media_driver/agnostic/common/cm/cm_mem_sse2_impl.cpp:42:64: warning: expression does not compute the number of elements in this array; element type is 'uint32_t' {aka 'unsigned int'}, not 'DQWORD' {aka '__m128'} [-Wsizeof-array-div]
   42 |     const size_t doubleQuadWordsPerPrefetch = sizeof(PREFETCH) / sizeof(DQWORD);
      |                                               ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
../media_driver/agnostic/common/cm/cm_mem_sse2_impl.cpp:42:66: note: add parentheses around 'sizeof (DQWORD)' to silence this warning
   42 |     const size_t doubleQuadWordsPerPrefetch = sizeof(PREFETCH) / sizeof(DQWORD);
      |                                                                  ^~~~~~~~~~~~~~
      |                                                                  (             )
../media_driver/agnostic/common/cm/cm_mem_sse2_impl.cpp: In function 'void FastMemCopy_SSE2_movdqu_movdqa(void*, void*, size_t)':
../media_driver/agnostic/common/cm/cm_mem_sse2_impl.cpp:84:64: warning: expression does not compute the number of elements in this array; element type is 'uint32_t' {aka 'unsigned int'}, not 'DQWORD' {aka '__m128'} [-Wsizeof-array-div]
   84 |     const size_t doubleQuadWordsPerPrefetch = sizeof(PREFETCH) / sizeof(DQWORD);
      |                                               ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
../media_driver/agnostic/common/cm/cm_mem_sse2_impl.cpp:84:66: note: add parentheses around 'sizeof (DQWORD)' to silence this warning
   84 |     const size_t doubleQuadWordsPerPrefetch = sizeof(PREFETCH) / sizeof(DQWORD);
      |                                                                  ^~~~~~~~~~~~~~
      |                                                                  (             )
../media_driver/agnostic/common/cm/cm_mem_sse2_impl.cpp: In function 'void FastMemCopy_SSE2_movntdq_movdqu(void*, const void*, size_t)':
../media_driver/agnostic/common/cm/cm_mem_sse2_impl.cpp:126:64: warning: expression does not compute the number of elements in this array; element type is 'uint32_t' {aka 'unsigned int'}, not 'DQWORD' {aka '__m128'} [-Wsizeof-array-div]
  126 |     const size_t doubleQuadWordsPerPrefetch = sizeof(PREFETCH) / sizeof(DQWORD);
      |                                               ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
../media_driver/agnostic/common/cm/cm_mem_sse2_impl.cpp:126:66: note: add parentheses around 'sizeof (DQWORD)' to silence this warning
  126 |     const size_t doubleQuadWordsPerPrefetch = sizeof(PREFETCH) / sizeof(DQWORD);
      |                                                                  ^~~~~~~~~~~~~~
      |                                                                  (             )
../media_driver/agnostic/common/cm/cm_mem_sse2_impl.cpp: In function 'void FastMemCopy_SSE2_movdqu_movdqu(void*, const void*, size_t)':
../media_driver/agnostic/common/cm/cm_mem_sse2_impl.cpp:166:64: warning: expression does not compute the number of elements in this array; element type is 'uint32_t' {aka 'unsigned int'}, not 'DQWORD' {aka '__m128'} [-Wsizeof-array-div]
  166 |     const size_t doubleQuadWordsPerPrefetch = sizeof(PREFETCH) / sizeof(DQWORD);
      |                                               ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
../media_driver/agnostic/common/cm/cm_mem_sse2_impl.cpp:166:66: note: add parentheses around 'sizeof (DQWORD)' to silence this warning
  166 |     const size_t doubleQuadWordsPerPrefetch = sizeof(PREFETCH) / sizeof(DQWORD);
      |                                                                  ^~~~~~~~~~~~~~
      |                                                                  (             )
```

### Describe the expected behavior

* Build to fail when there are warnings because you are enabling "MEDIA_BUILD_FATAL_WARNINGS" by default
* As the logic itself seems fine (based on variable name), do what the compiler suggests: `add parentheses around 'sizeof (DQWORD)' to silence this warning`

dvrogozh, 2022-02-10:
So, I provided a fix in https://github.com/intel/media-driver/pull/1350, but I actually don't like it. We can check with code owners if we can redeclare array as DQWORD but it might not make sense actually since we might need to have access to each byte to populate it. If that's so, then that's where I don't like current fix. It's difficult to explain to others why code was written in this way, and even I will forget the reason in a month and will need to step into compiling warning again to remember. Warning per-se makes sense, but I think language don't have a reasonable instrument to resolve it. Having some std:: function to calculate the size in explicit way might help here.

eero-t, 2022-02-11:
> It's difficult to explain to others why code was written in this way, and even I will forget the reason in a month and will need to step into compiling warning again to remember.

Or just add a comment.  Something like: "GCC gives warning when dividing sizeof(array) with sizeof(another type), in case division was intended to calculate array size. Extra parenthesis for the divisor suppresses the warning".

Another option is just to disable that particular GCC warning for that part of code:
https://stackoverflow.com/questions/3378560/how-to-disable-gcc-warnings-for-a-few-lines-of-code

But that's a bit ugly too.

Last option is just to ignore this bug. :-)

dvrogozh, 2022-02-11:
> "media-driver" enables option called "MEDIA_BUILD_FATAL_WARNINGS"

Yes, that's what I was talking about saying "stop" forcing errors - that media driver scripts should stop forcing them.

closed 2022-02-14 by intel-mediadev