intel/media-driver #1352: GuC for Jasper Lake (Celeron N5095)

ti3gib, 2022-02-20:
**## System information**
- **CPU information(`cat /proc/cpuinfo | grep "model name" | uniq`):**
`model name      : Intel(R) Celeron(R) N5095 @ 2.00GHz`
Kernel
`5.16.10-051610-generic`

- **GPU information(`lspci -nn | grep -E 'VGA|isplay`):**
`00:02.0 VGA compatible controller [0300]: Intel Corporation Device [8086:4e55] (rev 01)`

- Libva, gmmlib, Media Deriver, Media SDK built from Intel Github.
- options i915 force_probe=4e55 enable_guc=2 in modeprobe.d
-  Running Ubuntu 20.04 LTS

**## Issue behavior**
No hardware acceleration through Plex. GuC loading errors in dmesg i915, drm and vga is as follows:
`[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.16.10-051610-generic root=UUID=9dd85497-83cf-4379-b360-cc16a4b83a65 ro i915.force_probe=4e55
[    0.044443] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-5.16.10-051610-generic root=UUID=9dd85497-83cf-4379-b360-cc16a4b83a65 ro i915.force_probe=4e55
[    0.364097] pci 0000:00:02.0: vgaarb: setting as boot VGA device
[    0.364097] pci 0000:00:02.0: vgaarb: VGA device added: decodes=io+mem,owns=io+mem,locks=none
[    0.364097] pci 0000:00:02.0: vgaarb: bridge control possible
[    0.364097] vgaarb: loaded
[    0.465237] fb0: EFI VGA frame buffer device
[    1.018289] ACPI: bus type drm_connector registered
[    1.292597] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    1.295759] fb0: switching to i915 from EFI VGA
[    1.299377] i915 0000:00:02.0: vgaarb: deactivate vga console
[    1.299434] i915 0000:00:02.0: [drm] Transparent Hugepage mode 'huge=within_size'
[    1.304093] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    1.304665] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/icl_dmc_ver1_09.bin (v1.9)
[    1.318606] i915 0000:00:02.0: Direct firmware load for i915/ehl_guc_62.0.0.bin failed with error -2
[    1.318621] i915 0000:00:02.0: [drm] GuC firmware i915/ehl_guc_62.0.0.bin: fetch failed with error -2
[    1.318625] i915 0000:00:02.0: [drm] GuC firmware(s) can be downloaded from https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915
[    1.477330] i915 0000:00:02.0: [drm] GuC is uninitialized
[    1.492691] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    1.534349] fbcon: i915drmfb (fb0) is primary device
[    1.599545] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[    3.661966] systemd[1]: Condition check resulted in Load Kernel Module drm being skipped.
[    4.495132] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    4.607762] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[ 1219.960599] i915 0000:00:02.0: [drm] GuC is uninitialized
[ 1220.748976] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[ 2420.802910] i915 0000:00:02.0: [drm] GuC is uninitialized
[ 2421.782782] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])`

On requesting a file for transcode, the CPU runs at full utilization on all content (H.264, H.265, SDR, HDR)

**### Describe the expected behavior**
Plex to display (hw) tag when running a transcode.

**## Debug information**
- What's libva/libva-utils/gmmlib/media-driver version?
`error: can't connect to X server!
libva info: VA-API version 1.15.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_13
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.15 (libva 2.13.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 21.4.1 (be92568)
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSliceLP
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSliceLP
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile1            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileVP9Profile3            : VAEntrypointVLD
      VAProfileHEVCMain422_10         : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         : VAEntrypointVLD
      VAProfileHEVCMain444_10         : VAEntrypointEncSliceLP`

- Could you confirm whether GPU hardware exist or not by `ls /dev/dri`?
Confirmed.

Happy to run any other commands that could be helpful or provide any logs.

dvrogozh, 2022-02-20:
As suggested in dmesg, check that required version of GUC firmware is actually available on your system. Try to search:
```
find /lib/firmware -name ehl_guc_62.0.0.bin
```

If it's not found, first try to upgrade FW package in your OS and check again. If your OS don't yet package this FW version, download manually from https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git (see in i915 folder) and copy them to `/lib/firmware/i915/` where the rest of i915 FW is located.

ti3gib, 2022-02-20:
I've updated the kernel to see if it would come with the necessary drivers, which seemed to do nothing, but I also installed the latest linux-firmware package, which doesn't seem to effect the error.

The file it's looking is there:

`XXX@XXX:/media/intel$ find /lib/firmware -name ehl_guc_62.0.0.bin
/lib/firmware/i915/ehl_guc_62.0.0.bin
XXX@XXX:/media/intel$ ls -la /lib/firmware/i915/ehl_guc_62.0.0.bin
-rw-r--r-- 1 root root 327488 Feb 20 14:04 /lib/firmware/i915/ehl_guc_62.0.0.bin`

Updating the kernel doesn't seem to help. I used this command.

`update-initramfs -u -k all`

ti3gib, 2022-02-20:
I just booted with the other kernel and it seems to work! I feel so silly. Sorry for wasting your time. Closing this. 

For future reference, doing this worked for me if anyone ends up here after extensive googling

1) Make sure you're on the latest kernels
2) Install drivers (build from intel repos, or try one of the available packages)
3) Set options i915 force_probe=4e55 enable_guc=2 in /etc/modprobe.d/i915.conf
4) Install latest linux-firmware 
5) then update-initramfs -u -k all
6) Check /dev/dri for contents.

closed 2022-02-20 by ti3gib