intel/media-driver #1353: Compile error media_ddi_decode_vp8.cpp when using -Werror and gcc 12.0.1

heitbaum, 2022-02-27:
## System information
Cross compiling using gcc 12-20220220

## Issue behavior
### Describe the current behavior
Fails compile with `will never be NULL [-Werror=address]`

### Describe the expected behavior
Compiles correctly with gcc 11.2.0

## build log

```
[562/942] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_decode_vp8.cpp.o
FAILED: media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/codec/ddi/media_ddi_decode_vp8.cpp.o 
/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/bin/x86_64-libreelec-linux-gnu-g++ -DCLASS_TRACE=0 -DENABLE_KERNELS -DHEVC_FEI_ENABLE_CMRT -DIGFX_GEN10_CNL_SUPPORTED -DIGFX_GEN10_SUPPORTED -DIGFX_GEN11_ICLLP_SUPPORTED -DIGFX_GEN11_JSL_SUPPORTED -DIGFX_GEN11_SUPPORTED -DIGFX_GEN12_ADLP_SUPPORTED -DIGFX_GEN12_ADLS_SUPPORTED -DIGFX_GEN12_DG1_SUPPORTED -DIGFX_GEN12_RKL_SUPPORTED -DIGFX_GEN12_SUPPORTED -DIGFX_GEN12_TGLLP_CMFCPATCH_SUPPORTED -DIGFX_GEN12_TGLLP_CMFC_SUPPORTED -DIGFX_GEN12_TGLLP_SUPPORTED -DIGFX_GEN12_TGLLP_SWSB_SUPPORTED -DIGFX_GEN8_BDW_SUPPORTED -DIGFX_GEN8_SUPPORTED -DIGFX_GEN9_BXT_SUPPORTED -DIGFX_GEN9_CFL_SUPPORTED -DIGFX_GEN9_CML_SUPPORTED -DIGFX_GEN9_CMPV_SUPPORTED -DIGFX_GEN9_GLK_SUPPORTED -DIGFX_GEN9_KBL_SUPPORTED -DIGFX_GEN9_SKL_SUPPORTED -DIGFX_GEN9_SUPPORTED -DIGFX_MHW_INTERFACES_NEXT_SUPPORT -DMEDIA_VERSION=\"22.2.1\" -DMEDIA_VERSION_DETAILS=\"46ecaa9e0b\" -DVEBOX_AUTO_DENOISE_SUPPORTED=1 -D_AV1_DECODE_SUPPORTED -D_AVC_DECODE_SUPPORTED -D_AVC_ENCODE_VDENC_SUPPORTED -D_AVC_ENCODE_VME_SUPPORTED -D_COMMON_ENCODE_SUPPORTED -D_DECODE_PROCESSING_SUPPORTED -D_HEVC_DECODE_SUPPORTED -D_HEVC_ENCODE_VDENC_SUPPORTED -D_HEVC_ENCODE_VME_SUPPORTED -D_JPEG_DECODE_SUPPORTED -D_JPEG_ENCODE_SUPPORTED -D_MMC_SUPPORTED -D_MPEG2_DECODE_SUPPORTED -D_MPEG2_ENCODE_VME_SUPPORTED -D_RELEASE -D_VC1_DECODE_SUPPORTED -D_VP8_DECODE_SUPPORTED -D_VP8_ENCODE_SUPPORTED -D_VP9_DECODE_SUPPORTED -D_VP9_ENCODE_VDENC_SUPPORTED -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -D__VPHAL_SFC_SUPPORTED=1 -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/linux/common/os/i915/include -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/linux/common/os/i915/include/uapi -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/include/igdgmm/util -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/include/igdgmm/inc/common -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/include/igdgmm/inc -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/include/igdgmm/GmmLib/inc -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/include/igdgmm/GmmLib -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/include/igdgmm -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_common/agnostic/common/os -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_common/agnostic/common/codec/shared -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_common/agnostic/common/vp/hal -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_common/agnostic/common/hw/vdbox -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_common/agnostic/common/hw -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_common/agnostic/common/shared/user_setting -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_common/agnostic/common/shared -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_common/agnostic/common/media_interfaces -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_common/linux/common/os -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/cm -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/codec/hal -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/codec/kernel -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/codec/shared -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/cp -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/heap_manager -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/hw/vdbox -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/hw -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/media_interfaces -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/os -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/renderhal -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/shared/user_setting -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/shared -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/vp/hal -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/vp/kdll -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/vp/kernel -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/vp/cm_fc_ld -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/gen8/cm -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/gen8/codec/hal -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/gen8/codec/kernel -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/gen8/hw/vdbox -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/gen8/hw -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/gen8/renderhal -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/gen8/vp/hal -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/gen8/vp/kernel -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/gen8_bdw/codec/hal -I/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver [... cut ...] r=address]
  245 |     if (codecPicParams->ucMvUpdateProb[1] && picParam->mv_probs[1])
      |                                              ~~~~~~~~~~~~~~~~~~~~^
/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/include/va/va_dec_vp8.h:154:13: note: '_VAPictureParameterBufferVP8::mv_probs' declared here
  154 |     uint8_t mv_probs[2][19];
      |             ^~~~~~~~
cc1plus: all warnings being treated as errors
```

FCLC, 2022-04-13:
hitting a similar issue on my end,  gcc-12 (GCC) 12.0.1 20220412 (experimental) is non-functional. 

```
[  7%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/codec/hal/codechal_utilities.cpp.o
In file included from /home/felix/intel/media-driver/media_common/agnostic/common/os/mos_utilities.h:146,
                 from /home/felix/intel/media-driver/media_common/agnostic/common/os/mos_os.h:32,
                 from /home/felix/intel/media-driver/media_driver/agnostic/common/os/mos_context.h:30,
                 from /home/felix/intel/media-driver/media_driver/linux/common/ddi/media_libva_common.h:37,
                 from /home/felix/intel/media-driver/media_driver/linux/common/cm/ddi/media_libva_cm.h:30,
                 from /home/felix/intel/media-driver/media_driver/linux/common/cm/hal/cm_def_os.h:41,
                 from /home/felix/intel/media-driver/media_driver/agnostic/common/cm/cm_def.h:30,
                 from /home/felix/intel/media-driver/media_driver/agnostic/common/cm/cm_kernel.h:30,
                 from /home/felix/intel/media-driver/media_driver/agnostic/common/cm/cm_kernel_rt.h:30,
                 from /home/felix/intel/media-driver/media_driver/agnostic/common/cm/cm_kernel_ex.h:28,
                 from /home/felix/intel/media-driver/media_driver/agnostic/common/cm/cm_kernel_ex.cpp:27:
In static member function ‘static _Ty* MosUtilities::MosNewArrayUtil(int32_t) [with _Ty = unsigned char; _Types = {}]’,
    inlined from ‘_Ty* MOS_NewArrayUtil(int32_t) [with _Ty = unsigned char; _Types = {}]’ at /home/felix/intel/media-driver/media_common/agnostic/common/os/mos_utilities.h:193:46,
    inlined from ‘virtual int32_t CmKernelEx::Initialize(const char*, const char*)’ at /home/felix/intel/media-driver/media_driver/agnostic/common/cm/cm_kernel_ex.cpp:192:22,
    inlined from ‘virtual int32_t CmKernelEx::Initialize(const char*, const char*)’ at /home/felix/intel/media-driver/media_driver/agnostic/common/cm/cm_kernel_ex.cpp:70:9:
/home/felix/intel/media-driver/media_softlet/agnostic/common/os/mos_utilities_next.h:106:20: error: argument 1 range [18446744071562067968, 18446744073709551615] exceeds maximum object size 9223372036854775807 [-Werror=alloc-size-larger-than=]
  106 |         _Ty* ptr = new (std::nothrow) _Ty[numElements]();
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/felix/tmp/gcc-12_build/build/include/c++/12.0.1/bits/exception_ptr.h:40,
                 from /home/felix/tmp/gcc-12_build/build/include/c++/12.0.1/exception:168,
                 from /home/felix/tmp/gcc-12_build/build/include/c++/12.0.1/ios:39,
                 from /home/felix/tmp/gcc-12_build/build/include/c++/12.0.1/ostream:38,
                 from /home/felix/tmp/gcc-12_build/build/include/c++/12.0.1/iostream:39,
                 from /home/felix/intel/media-driver/media_driver/linux/common/cm/hal/cm_def_os.h:34:
/home/felix/tmp/gcc-12_build/build/include/c++/12.0.1/new: In member function ‘virtual int32_t CmKernelEx::Initialize(const char*, const char*)’:
/home/felix/tmp/gcc-12_build/build/include/c++/12.0.1/new:142:26: note: in a call to allocation function ‘void* operator new [](std::size_t, const std::nothrow_t&)’ declared here
  142 | _GLIBCXX_NODISCARD void* operator new[](std::size_t, const std::nothrow_t&) _GLIBCXX_USE_NOEXCEPT
      |                          ^~~~~~~~
[  7%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/codec/hal/codechal_mmc.cpp.o
[  7%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/codec/hal/codechal_allocator.cpp.o
[  7%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/codec/hal/codechal_memdecomp.cpp.o
[  7%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/codec/hal/codechal_decode_nv12top010.cpp.o
cc1plus: all warnings being treated as errors
make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:576: media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/cm/cm_kernel_ex.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:272: media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/all] Error 2
make: *** [Makefile:171: all] Error 2

```

eero-t, 2022-04-14:
"exceeds maximum object size 9223372036854775807" = 2^63-1, which looks suspiciously like wrongly treated negative value...

Which is completely different from the originally reported "the address of '_VAPictureParameterBufferVP8::mv_probs' will never be NULL" error.

FCLC, 2022-04-14:
> "exceeds maximum object size 9223372036854775807" = 2^63-1, which looks suspiciously like wrongly treated negative value...
> 
> Which is completely different from the originally reported "the address of '_VAPictureParameterBufferVP8::mv_probs' will never be NULL" error.

Agreed. More so I wanted to highlight that there are other issues related to building on gcc-12.0.1

dvrogozh, 2022-04-14:
gcc-12 is not yet a released compiler. And unfortunately we are not aware of the way to get it other than build manually (vs. clang which provides apt channel for the compiler which is in active development). As a result build against gcc-12 is completely off the ci radar and new issues might squeeze in literally on any commit. With the current resources we have it will appear on radar as soon as it will be reasonably easy available. We appreciate if you will help suggesting a way to get gcc-12 other than manually build. This might be:
* Official gcc APT repo channel (which we might have missed)
* One of the official docker OS images containing gcc-12

We also appreciate a PRs to fix this issue, we will gladly review them. Mind that setting up environment for gcc-12 takes away resources on our side, thus expect delay with fixing these issues.

FCLC, 2022-04-14:
> gcc-12 is not yet a released compiler. And unfortunately we are not aware of the way to get it other than build manually (vs. clang which provides apt channel for the compiler which is in active [...]

I'm currently working on a PR addressing issues with #1342. For now it's fairly minor and only appears in specific edge cases. 

I'll add a note to the readme in my PR

cqian2, 2022-04-18:
Our PR https://github.com/intel/media-driver/pull/1365 should have fixed this compile issue also; Would you please help try the latest github driver?

closed 2022-09-26 by cqian2