intel/media-driver #1354: Compile error -Werror when compiling with gcc 12.0.1

heitbaum, 2022-02-27:
## System information
Cross compiling using gcc 12-20220220

## Issue behavior
### Describe the current behavior
Fails compile with:
- `warning: dangling pointer 'address2Counter' to 'ctr' may be used [-Wdangling-pointer=]`
- `warning: argument 1 range [18446744071562067968, 18446744073709551615] exceeds maximum object size 9223372036854775807 [-Walloc-size-larger-than=]`

### Describe the expected behavior
Compiles correctly with gcc 11.2.0
Using the following compile options allows these errors to pass as warnings:
- `export CXXFLAGS="${CXXFLAGS} -Wno-error=alloc-size-larger-than= -Wno-error=dangling-pointer"`

## build log

```
[43/942] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/cm/cm_kernel_ex.cpp.o
In file included from /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_common/agnostic/common/os/mos_utilities.h:138,
                 from /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_common/agnostic/common/os/mos_os.h:32,
                 from /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/os/mos_context.h:30,
                 from /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/linux/common/ddi/media_libva_common.h:37,
                 from /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/linux/common/cm/ddi/media_libva_cm.h:30,
                 from /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/linux/common/cm/hal/cm_def_os.h:41,
                 from /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/cm/cm_def.h:30,
                 from /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/cm/cm_kernel.h:30,
                 from /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/cm/cm_kernel_rt.h:30,
                 from /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/cm/cm_kernel_ex.h:28,
                 from /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/cm/cm_kernel_ex.cpp:27:
In static member function 'static _Ty* MosUtilities::MosNewArrayUtil(int32_t) [with _Ty = unsigned char; _Types = {}]',
    inlined from '_Ty* MOS_NewArrayUtil(int32_t) [with _Ty = unsigned char; _Types = {}]' at /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_common/agnostic/common/os/mos_utilities.h:185:46,
    inlined from 'virtual int32_t CmKernelEx::Initialize(const char*, const char*)' at /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/cm/cm_kernel_ex.cpp:192:22,
    inlined from 'virtual int32_t CmKernelEx::Initialize(const char*, const char*)' at /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/cm/cm_kernel_ex.cpp:70:9:
/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_softlet/agnostic/common/os/mos_utilities_next.h:106:20: warning: argument 1 range [18446744071562067968, 18446744073709551615] exceeds maximum object size 9223372036854775807 [-Walloc-size-larger-than=]
  106 |         _Ty* ptr = new (std::nothrow) _Ty[numElements]();
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/x86_64-libreelec-linux-gnu/include/c++/12.0.1/bits/exception_ptr.h:40,
                 from /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/x86_64-libreelec-linux-gnu/include/c++/12.0.1/exception:168,
                 from /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/x86_64-libreelec-linux-gnu/include/c++/12.0.1/ios:39,
                 from /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/x86_64-libreelec-linux-gnu/include/c++/12.0.1/ostream:38,
                 from /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/x86_64-libreelec-linux-gnu/include/c++/12.0.1/iostream:39,
                 from /var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/linux/common/cm/hal/cm_def_os.h:34:
/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/x86_64-libreelec-linux-gnu/include/c++/12.0.1/new: In member function 'virtual int32_t CmKernelEx::Initialize(const char*, const char*)':
/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/x86_64-libreelec-linux-gnu/include/c++/12.0.1/new:142:26: note: in a call to allocation function 'void* operator new [](std::size_t, const std::nothrow_t&)' declared here
  142 | _GLIBCXX_NODISCARD void* operator new[](std::size_t, const std::nothrow_t&) _GLIBCXX_USE_NOEXCEPT
      |                          ^~~~~~~~
[95/942] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/codec/hal/codechal_encoder_base.cpp.o
/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/codec/hal/codechal_encoder_base.cpp: In member function 'MOS_STATUS CodechalEncoderState::ReadCounterValue(uint16_t, EncodeStatusReport*)':
/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/codec/hal/codechal_encoder_base.cpp:3941:43: warning: dangling pointer 'address2Counter' to 'ctr' may be used [-Wdangling-pointer=]
 3941 |     encodeStatusReport->HWCounterValue.IV = *(++address2Counter);
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/codec/hal/codechal_encoder_base.cpp:3926:18: note: 'ctr' declared here
 3926 |         uint32_t ctr[4] = { 0 };
      |                  ^~~
/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/codec/hal/codechal_encoder_base.cpp:3937:48: warning: dangling pointer 'address2Counter' to 'ctr' may be used [-Wdangling-pointer=]
 3937 |     encodeStatusReport->HWCounterValue.Count = *address2Counter;
      |                                                ^~~~~~~~~~~~~~~~
/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/media-driver-22.2.1/media_driver/agnostic/common/codec/hal/codechal_encoder_base.cpp:3926:18: note: 'ctr' declared here
 3926 |         uint32_t ctr[4] = { 0 };
      |                  ^~~
```

pingliwa, 2022-03-24:
Created a PR to fix it.
https://github.com/intel/media-driver/pull/1365

heitbaum, 2022-03-24:
> Created a PR to fix it.
> #1365

Tested - working 👍

PR #1365 (2022-03-31): Fixes #1354
Fixes #1354. It is to fix the compiling issue for gcc12.0.1.
[-Werror=address] [-Werror=dangling-pointer=][-Werror=alloc-size-larger-than=]
commit c8457540 (2022-03-31): Fixes #1354

Fixes #1354. It is to fix the compiling issue for gcc12.0.1.
[-Werror=address] [-Werror=dangling-pointer=][-Werror=alloc-size-larger-than=]
closed 2022-03-31 by intel-mediadev