intel/media-driver #1355: Possible Inconsistency in readme and docs

DavidReinberger, 2022-03-02:
## System information
- CPU information(`cat /proc/cpuinfo | grep "model name" | uniq`): Intel(R) Celeron(R) N4000 CPU @ 1.10GHz
- GPU information(`lspci -nn | grep -E 'VGA|isplay`): Intel Corporation GeminiLake [UHD Graphics 600] [8086:3185] (rev 03)
- Display server if rendering to display(X or wayland):

## Issue behavior
### Describe the current behavior
GPU Hardware Accelerates `p010` surface encoding or at least it seems to via ffmpeg even though the [readme](https://github.com/intel/media-driver#decodingencoding-features)/[docs](https://github.com/intel/media-driver/blob/master/docs/media_features.md#supported-video-processing-cscscaling-format) states otherwise (or I understand it that way). 

### Describe the expected behavior
Either docs should reflect this ability to **encode** `p010` surfaces back to the video when transcoding or throw an error that this is not possible (my guess this is not ffmpeg's job but the driver's).

## Debug information
- What's libva/libva-utils/gmmlib/media-driver version? 22.2.0
- Could you confirm whether GPU hardware exist or not by `ls /dev/dri`? Yes
- Could you provide vainfo log if possible by `vainfo >vainfo.log 2>&1`?

```
error: can't connect to X server!
libva info: Open new log file /tmp/libva_trace.log.103645.thd-0x003e6417 for the thread 0x003e6417
libva info: LIBVA_TRACE is on, save log into /tmp/libva_trace.log.103645.thd-0x003e6417
libva info: VA-API version 1.13.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_13
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.13 (libva 2.10.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.2.0 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointFEI
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSlice
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
```

- Could you provide libva trace log if possible? Run cmd `export LIBVA_TRACE=/tmp/libva_trace.log` first then execute the case.
- Could you attach dmesg log if it's GPU hang by `dmesg >dmesg.log 2>&1`?
- Do you want to contribute a patch to fix the issue? (yes/no): no
- Executed FFMPEG command
```
ffmpeg -y -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -hwaccel_output_format vaapi -I /input.mkv -sn -max_muxing_queue_size 9999 -map 0:V:0 -c:V:0 hevc_vaapi -rc_mode QVBR -compression_level 7 -bufsize 50M -vf scale_vaapi=format=p010 -b:V:0 8M -maxrate:V:0 10M -sei hdr -color_trc smpte2084 -chroma_sample_location topleft -color_primaries bt2020 -colorspace bt2020nc -color_range tv -map_metadata 0 -map_chapters 0 /test.mkv
```
- FFPROBE on input 
```
Stream #0:0: Video: hevc (Main 10), yuv420p10le(tv, bt2020nc/bt2020/smpte2084), 3840x2160 [SAR 1:1 DAR 16:9], 24 fps, 24 tbr, 1k tbn, 24 tbc (default)
```
- FFPROBE on output 
```
Stream #0:0: Video: hevc (Main 10), yuv420p10le(tv, bt2020nc/bt2020/smpte2084, progressive), 3840x2160 [SAR 1:1 DAR 16:9], 24 fps, 24 tbr, 1k tbn, 24 tbc (default)
```

OttoYang, 2023-07-28:
Hi @DavidReinberger,
Let me clarify the problem you have encountered, you want to set both input and output format to p010 on GLK, but it's not working?

OttoYang, 2023-10-30:
Hi @DavidReinberger , could you help clarify the above?