intel/media-driver #1356: fcpatch files are completely missing in Free Build

frantisekz, 2022-03-03:
## System information
intel-igc-1.0.10395
intel-opencl-clang-13.0.0
intel-gmmlib-22.0.2
intel-compute-runtime-22.08.22549
intel-cm-compiler-1.0.119

## Issue behavior
### Describe the current behavior
Since #1328 , the free build path is failing on multiple fcpatch files missing (or more precisely, there aren't any). 

I am using sources stripped by:
(according to https://github.com/intel/media-driver/wiki/Media-Driver-Shaders-(EU-Kernels)#build-with-open-source-shaders )
```
find . -name kernel | grep gen | xargs rm -r
find . -name cm_gpucopy_kernel* | xargs rm
find . -name cmrt_kernel | xargs rm -r
```

I've used a small patch to try to get further, however, it failed soon after due to missing files for cmake copy commands:

```

From da1a66291d58f1e5e1c7485318ca089ef299f89c Mon Sep 17 00:00:00 2001
From: =?UTF-8?q?Franti=C5=A1ek=20Zatloukal?= <fzatlouk@redhat.com>
Date: Fri, 18 Feb 2022 15:13:10 +0100
Subject: [PATCH] Hacky try

---
 media_driver/agnostic/Xe_M/Xe_HPM/vp/kernel_free/Source/rename.py   | 7 +++++--
 media_driver/agnostic/gen12_tgllp/vp/kernel_free/Source/rename.py   | 7 +++++--
 2 files changed, 10 insertions(+), 4 deletions(-)

diff --git a/media_driver/agnostic/Xe_M/Xe_HPM/vp/kernel_free/Source/rename.py b/media_driver/agnostic/Xe_M/Xe_HPM/vp/kernel_free/Source/rename.py
index c925647..9097da3 100644
--- a/media_driver/agnostic/Xe_M/Xe_HPM/vp/kernel_free/Source/rename.py
+++ b/media_driver/agnostic/Xe_M/Xe_HPM/vp/kernel_free/Source/rename.py
@@ -10,8 +10,11 @@ def change_mistake(path, upperFileName):
         print("Upper file" + upperFile + " exists! No need to rename!")
     else:
         lowerFile = upperFile.replace('DScale', 'Dscale')
-        os.rename(lowerFile, upperFile)
-        print(lowerFile + "replaced!")
+        try:
+            os.rename(lowerFile, upperFile)
+            print(lowerFile + "replaced!")
+        except:
+            print("Rename failed!")

 change_mistake(sys.argv[1], sys.argv[2])
 if(len(sys.argv) == 4):
diff --git a/media_driver/agnostic/gen12_tgllp/vp/kernel_free/Source/rename.py b/media_driver/agnostic/gen12_tgllp/vp/kernel_free/Source/rename.py
index 58861e7..5d6ec80 100644
--- a/media_driver/agnostic/gen12_tgllp/vp/kernel_free/Source/rename.py
+++ b/media_driver/agnostic/gen12_tgllp/vp/kernel_free/Source/rename.py
@@ -10,8 +10,11 @@ def change_mistake(path, upperFileName):
         print("Upper file" + upperFile + " exists! No need to rename!")
     else:
         lowerFile = upperFile.replace('DScale', 'Dscale')
-        os.rename(lowerFile, upperFile)
-        print(lowerFile + "replaced!")
+        try:
+            os.rename(lowerFile, upperFile)
+            print(lowerFile + "replaced!")
+        except:
+            print("Rename failed!")

 change_mistake(sys.argv[1], sys.argv[2])
 if(len(sys.argv) == 4):
--
2.34.1

```
I've attached the entire build.log below, feel free to ping with anything that I shall try or for more information about the build env.

Thanks!

[build.log](https://github.com/intel/media-driver/files/8180572/build.log)

### Describe the expected behavior

## Debug information

frantisekz, 2022-03-03:
@MicroYY

MicroYY, 2022-03-15:
This may be issues on the compiler. I was testing on internal compiler. Let me check with provided environment.

FurongZhang, 2022-03-16:
@MicroYY , I suppose you fix the issue. If yes, can we close this issue.

frantisekz, 2022-05-29:
@MicroYY this is still an issue, with:
cm-compiler-1.0.144
intel-igc-1.0.11279
intel-compute-runtime-22.21.23269

Full compilation log attached:
[build.log](https://github.com/intel/media-driver/files/8793621/build.log)

frantisekz, 2023-01-18:
@bai-isaac Sorry for pinging directly, just trying if you wouldn't have any ideas about this maybe?

Thanks!

DemiMarie, 2024-09-09:
@frantisekz Can you try building without stripping?