intel/media-driver #1358: GPU HANG with libva-intel-media-driver 22.x but not with 21.x  (NUC8i7BEH:  i7-8559U & Iris Plus Graphics 655)

damocles-dev, 2022-03-14:
## System information
- [x] Display server if rendering to display(X or wayland): X.Org X Server 1.21.1.3 (Wayland disabled)
- [x] Impacted GPU platform: Intel Corporation CoffeeLake-U GT3e [Iris Plus Graphics 655] [8086:3ea5] (rev 01)
- [x] Host machine if it's descreted card: Intel(R) Core(TM) i7-8559U CPU @ 2.70GHz 
- [x] Linux kernel version: 5.16.8
- [x] Good libva/libva-utils/gmmlib/media-driver version? 
    gmmlib-21.3.5
    libva-2.13.0
    libva-utils-2.13.0
    libva-intel-media-driver-21.4.3
- [x] Bad libva/libva-utils/gmmlib/media-driver version?
    gmmlib-21.3.5
    libva-2.13.0
    libva-utils-2.13.0
    libva-intel-media-driver-22.1.0

## Issue behavior
[64304.295273] i915 0000:00:02.0: [drm] Resetting vcs1 for preemption time out
[64304.295294] i915 0000:00:02.0: [drm] mpv[89574] context reset due to GPU hang
[64304.298461] i915 0000:00:02.0: [drm] GPU HANG: ecode 9:4:c87dff99, in mpv [89574]
[64321.318934] i915 0000:00:02.0: [drm] Resetting vcs1 for preemption time out
[64321.318971] i915 0000:00:02.0: [drm] mpv[89632] context reset due to GPU hang
[64321.322055] i915 0000:00:02.0: [drm] GPU HANG: ecode 9:4:c87dff99, in mpv [89632]
[64361.254130] i915 0000:00:02.0: [drm] Resetting vcs1 for preemption time out
[64361.254166] i915 0000:00:02.0: [drm] mpv[89737] context reset due to GPU hang
[64361.257372] i915 0000:00:02.0: [drm] GPU HANG: ecode 9:4:c87dff99, in mpv [89737]

### Describe the current behavior
Video playback stops randomly and you need to kill the process to be able to run it again 

### Describe the expected behavior
Video playback without GPU hangs

#### Full debug trace:
<details><summary>
/sys/class/drm/card0/error
</summary>

```

GPU HANG: ecode 9:4:c87dff99, in mpv [56786]
Kernel: 5.16.8-gentoo-x86_64 x86_64
Driver: 20201103
Time: 1647214334 s 255816 us
Boottime: 58296 s 540259 us
Uptime: 58292 s 563901 us
Capture: 4352963712 jiffies; 26834625 ms ago
Active process (on ring vcs1): mpv [56786]
Reset count: 0
Suspend count: 0
Platform: COFFEELAKE
Subplatform: 0x1
PCI ID: 0x3ea5
PCI Revision: 0x01
PCI Subsystem: 8086:2074
IOMMU enabled?: 1
DMC loaded: yes
DMC fw version: 1.4
RPM wakelock: yes
PM suspended: no
GT awake: yes
EIR: 0x00000000
IER: 0x08080000
GTIER[0]: 0x09090909
GTIER[1]: 0x09090909
GTIER[2]: 0x00000000
GTIER[3]: 0x00000909
PGTBL_ER: 0x00000000
FORCEWAKE: 0x00010001
DERRMR: 0x2077efef
[... 159 lines cut ...]
i915.guc_firmware_path=(null)
i915.huc_firmware_path=(null)
i915.dmc_firmware_path=(null)
i915.mmio_debug=1
i915.edp_vswing=0
i915.reset=3
i915.inject_probe_failure=0
i915.fastboot=-1
i915.enable_dpcd_backlight=-1
i915.force_probe=
i915.fake_lmem_start=0
i915.request_timeout_ms=20000
i915.enable_hangcheck=yes
i915.load_detect_test=no
i915.force_reset_modeset_test=no
i915.error_capture=yes
i915.disable_display=no
i915.verbose_state_checks=yes
i915.nuclear_pageflip=no
i915.enable_dp_mst=yes
i915.enable_gvt=no

```

</details>

damocles-dev, 2022-03-14:
updated [gentoo package to gmmlib 22.0.3](https://packages.gentoo.org/packages/media-libs/gmmlib) and I still get GPU hangs 

```
2022-03-14T17:23:16.273087+01:00 nuc kernel: [118956.217815] i915 0000:00:02.0: [drm] Resetting vcs1 for preemption time out
2022-03-14T17:23:16.273112+01:00 nuc kernel: [118956.217853] i915 0000:00:02.0: [drm] mpv[146579] context reset due to GPU hang
2022-03-14T17:23:16.276453+01:00 nuc kernel: [118956.221040] i915 0000:00:02.0: [drm] GPU HANG: ecode 9:4:c87dff99, in mpv [146579]
2022-03-14T17:23:16.273076+01:00 nuc kernel: i915 0000:00:02.0: [drm] Resetting vcs1 for preemption time out
2022-03-14T17:23:16.274055+01:00 nuc kernel: i915 0000:00:02.0: [drm] mpv[146579] context reset due to GPU hang
2022-03-14T17:23:16.276381+01:00 nuc kernel: i915 0000:00:02.0: [drm] GPU HANG: ecode 9:4:c87dff99, in mpv [146579]
```

new debug trace:
<details><summary>
/sys/class/drm/card0/error
</summary>
```

GPU HANG: ecode 9:4:c87dff99, in mpv [56786]
Kernel: 5.16.8-gentoo-x86_64 x86_64
Driver: 20201103
Time: 1647214334 s 255816 us
Boottime: 58296 s 540259 us
Uptime: 58292 s 563901 us
Capture: 4352963712 jiffies; 26834625 ms ago
Active process (on ring vcs1): mpv [56786]
Reset count: 0
Suspend count: 0
Platform: COFFEELAKE
Subplatform: 0x1
PCI ID: 0x3ea5
PCI Revision: 0x01
PCI Subsystem: 8086:2074
IOMMU enabled?: 1
DMC loaded: yes
DMC fw version: 1.4
RPM wakelock: yes
PM suspended: no
GT awake: yes
EIR: 0x00000000
IER: 0x08080000
GTIER[0]: 0x09090909
GTIER[1]: 0x09090909
GTIER[2]: 0x00000000
GTIER[3]: 0x00000909
PGTBL_ER: 0x00000000
FORCEWAKE: 0x00010001
DERRMR: 0x2077efef
  fence[0] = 00000000
  fence[1] = 00000000
  fence[2] = 00000000
  fence[3] = 286803b02080001
  fence[4] = 00000000
  fence[5] = 9c3007008c4003
  fence[6] = 386803b03080001
  fence[7] = 00000000
  fence[8] = 00000000
  fence[9] = 306803b02880001
  fence[10] = 11e803b00a00001
  fence[11] = 00000000
  fence[12] = 00000000
  fence[13] = 00000000
  fence[14] = 00000000
  fence[15] = 00000000
  fence[16] = 00000000
  fence[17] = 00000000
  fence[18] = 00000000
  fence[19] = 00000000
  fence[20] = 00000000
  fence[21] = 00000000
  fence[22] = 00000000
[... 135 lines cut ...]
i915.guc_log_level=-1
i915.guc_firmware_path=(null)
i915.huc_firmware_path=(null)
i915.dmc_firmware_path=(null)
i915.mmio_debug=1
i915.edp_vswing=0
i915.reset=3
i915.inject_probe_failure=0
i915.fastboot=-1
i915.enable_dpcd_backlight=-1
i915.force_probe=
i915.fake_lmem_start=0
i915.request_timeout_ms=20000
i915.enable_hangcheck=yes
i915.load_detect_test=no
i915.force_reset_modeset_test=no
i915.error_capture=yes
i915.disable_display=no
i915.verbose_state_checks=yes
i915.nuclear_pageflip=no
i915.enable_dp_mst=yes
i915.enable_gvt=no
```

</details>

damocles-dev, 2022-03-17:
The random hangs happens only with HEVC decoding. 

If I play the same video several times, sometimes it works and sometimes it doesn't:
```
[ffmpeg/video] hevc: nal_unit_type: 0(TRAIL_N), nuh_layer_id: 0, temporal_id: 0
[vo/gpu] pass 'color conversion + output to screen': last 0us avg 0us peak 0us
[vo/gpu] pass 'drawing osd': last 0us avg 0us peak 0us
[osc] rendering 
[ffmpeg/video] hevc: Output frame with POC 2.
[ffmpeg/video] hevc: Param buffer (type 0, 604 bytes) is 0.
[ffmpeg/video] hevc: Slice 0 param buffer (264 bytes) is 0x1.
[ffmpeg/video] hevc: Slice 0 data buffer (110 bytes) is 0x2.
[ffmpeg/video] hevc: Decode to surface 0x11.
[ffmpeg/video] hevc: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] hevc: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
[ffmpeg/video] hevc: nal_unit_type: 0(TRAIL_N), nuh_layer_id: 0, temporal_id: 0
[ffmpeg/video] hevc: Output frame with POC 3.
[ffmpeg/video] hevc: Param buffer (type 0, 604 bytes) is 0x2.
[ffmpeg/video] hevc: Slice 0 param buffer (264 bytes) is 0x1.
[ffmpeg/video] hevc: Slice 0 data buffer (110 bytes) is 0.
[ffmpeg/video] hevc: Decode to surface 0x10.
[ffmpeg/video] hevc: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] hevc: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
[ffmpeg/video] hevc: nal_unit_type: 0(TRAIL_N), nuh_layer_id: 0, temporal_id: 0
[ffmpeg/video] hevc: Output frame with POC 4.
[ffmpeg/video] hevc: Param buffer (type 0, 604 bytes) is 0.
[ffmpeg/video] hevc: Slice 0 param buffer (264 bytes) is 0x1.
[ffmpeg/video] hevc: Slice 0 data buffer (110 bytes) is 0x2.
[ffmpeg/video] hevc: Decode to surface 0x11.
[ffmpeg/video] hevc: Failed to end picture decode issue: 23 (internal decoding error).
[ffmpeg/video] hevc: hardware accelerator failed to decode picture
Error while decoding frame (hardware decoding)!
[vo/gpu] Updating global uniform 'texture0'
[vo/gpu] Updating global uniform 'texture1'
[vo/gpu] pass 'color conversion + output to screen': last 0us avg 0us peak 0us
[vo/gpu] pass 'drawing osd': last 0us avg 0us peak 0us
Falling back to software decoding.
```

damocles-dev, 2022-03-19:
I tried media-driver 22.4.o and it seems that the hang does not occur. 

<pre>
$ vainfo 
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib64/va/drivers/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_14
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.14 (libva 2.14.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.4.0 (4171a3e4f)
</pre>

babam86, 2022-03-19:
22.4.0?

XinfengZhang, 2022-03-20:
@Jexu , it is CFL, PCI ID 0x3EA5 please help check it

damocles-dev, 2022-03-20:
> 22.4.0? I only get 22.3.0

Gentoo emerge just download git HEAD code and tries to compile it.

<details>
<summary>
Emerge source prepare and configure
</summary>

<pre>
# emerge -v libva-intel-media-driver

These are the packages that would be merged, in order:

Calculating dependencies... done!
[ebuild   R   *] x11-libs/libva-intel-media-driver-9999::gentoo  USE="X redistributable -test" 0 KiB

Total: 1 package (1 reinstall), Size of downloads: 0 KiB

Would you like to merge these packages? [Yes/No] y

>>> Verifying ebuild manifests

>>> Emerging (1 of 1) x11-libs/libva-intel-media-driver-9999::gentoo
>>> Unpacking source...
 * Repository id: intel_media-driver.git
 * To override fetched repository properties, use:
 *   EGIT_OVERRIDE_REPO_INTEL_MEDIA_DRIVER
 *   EGIT_OVERRIDE_BRANCH_INTEL_MEDIA_DRIVER
 *   EGIT_OVERRIDE_COMMIT_INTEL_MEDIA_DRIVER
 *   EGIT_OVERRIDE_COMMIT_DATE_INTEL_MEDIA_DRIVER
 *
* Fetching https://github.com/intel/media-driver ...
git fetch https://github.com/intel/media-driver +HEAD:refs/git-r3/HEAD
git symbolic-ref refs/git-r3/x11-libs/libva-intel-media-driver/0/__main__ refs/git-r3/HEAD
 * Checking out https://github.com/intel/media-driver to /var/tmp/portage/x11-libs/libva-intel-media-driver-9999/work/libva-intel-media-driver-9999 ...
git checkout --quiet refs/git-r3/HEAD
GIT update -->
   repository:               https://github.com/intel/media-driver
   at the commit:            4171a3e4fc8399fc3a372ae846ef6e9e25b2ac8d
>>> Source unpacked in /var/tmp/portage/x11-libs/libva-intel-media-driver-9999/work
>>> Preparing source in /var/tmp/portage/x11-libs/libva-intel-media-driver-9999/work/libva-intel-media-driver-9999 ...
 * Source directory (CMAKE_USE_DIR): "/var/tmp/portage/x11-libs/libva-intel-media-driver-9999/work/libva-intel-media-driver-9999"
 * Build directory  (BUILD_DIR):     "/var/tmp/portage/x11-libs/libva-intel-media-driver-9999/work/libva-intel-media-driver-9999_build"
 * Applying libva-intel-media-driver-20.2.0_x11_optional.patch ...                                                                                               [ ok ]
 * Applying libva-intel-media-driver-21.4.2-Remove-unwanted-CFLAGS.patch ...
patching file media_driver/cmake/linux/media_compile_flags_linux.cmake
Hunk #1 succeeded at 51 with fuzz 1 (offset -1 lines).
Hunk #2 succeeded at 193 (offset 8 lines).                                                                                                                       [ ok ]
 * Applying libva-intel-media-driver-20.4.5_testing_in_src_test.patch ...                                                                                        [ ok ]
 * Hardcoded definition(s) removed in media_driver/linux/ult/CMakeLists.txt:
 *      set(CMAKE_BUILD_TYPE "Debug")
 *      set(CMAKE_BUILD_TYPE "ReleaseInternal")
 *      set(CMAKE_BUILD_TYPE "Release")
 * Hardcoded definition(s) removed in media_driver/CMakeLists.txt:
 *      set(CMAKE_BUILD_TYPE "Release")
 *      set(CMAKE_BUILD_TYPE "ReleaseInternal")
 *      set(CMAKE_BUILD_TYPE "Debug")
 * Hardcoded definition(s) removed in media_softlet/CMakeLists.txt:
 *      set(CMAKE_BUILD_TYPE "Release")
 *      set(CMAKE_BUILD_TYPE "ReleaseInternal")
 *      set(CMAKE_BUILD_TYPE "Debug")
 * Hardcoded definition(s) removed in CMakeLists.txt:
 *      set(CMAKE_INSTALL_PREFIX "/usr/local" CACHE PATH "..." FORCE)
>>> Source prepared.
>>> Configuring source in /var/tmp/portage/x11-libs/libva-intel-media-driver-9999/work/libva-intel-media-driver-9999 ...
 * Source directory (CMAKE_USE_DIR): "/var/tmp/portage/x11-libs/libva-intel-media-driver-9999/work/libva-intel-media-driver-9999"
 * Build directory  (BUILD_DIR):     "/var/tmp/portage/x11-libs/libva-intel-media-driver-9999/work/libva-intel-media-driver-9999_build"
cmake -C /var/tmp/portage/x11-libs/libva-intel-media-driver-9999/work/libva-intel-media-driver-9999_build/gentoo_common_config.cmake -G Ninja -DCMAKE_INSTALL_PREFIX=/usr -DMEDIA_BUILD_FATAL_WARNINGS=OFF -DMEDIA_RUN_TEST_SUITE=no -DBUILD_TYPE=Release -DPLATFORM=linux -DUSE_X11=yes -DENABLE_NONFREE_KERNELS=yes -DLATEST_CPP_NEEDED=ON -DCMAKE_BUILD_TYPE=Release -DCMAKE_TOOLCHAIN_FILE=/var/tmp/portage/x11-libs/libva-intel-media-driver-9999/work/libva-intel-media-driver-9999_build/gentoo_toolchain.cmake /var/tmp/portage/x11-libs/libva-intel-media-driver-9999/work/libva-intel-media-driver-9999
loading initial cache file /var/tmp/portage/x11-libs/libva-intel-media-driver-9999/work/libva-intel-media-driver-9999_build/gentoo_common_config.cmake
[... 36 lines cut ...]
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Checking for module 'igdgmm>=12.0.0'
--   Found igdgmm, version 12.0.0
-- media -- PLATFORM = linux
-- media -- ARCH = 64
-- media -- CMAKE_CURRENT_LIST_DIR = /var/tmp/portage/x11-libs/libva-intel-media-driver-9999/work/libva-intel-media-driver-9999/media_driver
-- media -- INCLUDED_LIBS =
-- media -- LIB_NAME = iHD_drv_video
-- media -- OUTPUT_NAME =
-- media -- BUILD_TYPE/UFO_BUILD_TYPE/CMAKE_BUILD_TYPE = release/release/Release
-- media -- LIBVA_INSTALL_PATH =
-- media -- MEDIA_VERSION = 22.4.0
-- media -- X11 Found
-- Checking for module 'libva-x11'
--   Found libva-x11, version 1.14.0
-- LIBVA_DRIVERS_PATH = /usr/lib64/va/drivers
-- <<< Gentoo configuration >>>
[...]
</details>

Jexu, 2022-03-21:
> 22.4.0? I only get 22.3.0

It is the latest commit version(4171a3e4fc8399fc3a372ae846ef6e9e25b2ac8d), please help try it again. Thanks a lot.

damocles-dev, 2022-03-21:
These are the results:

- 21.4.3: works
- 22.1.0: cpu hang
- 22.2.2: works
- 22.3.0: works

Now I can play videos again, thank you very much for fixing it.

Jexu, 2022-03-21:
Good to know it.  So @damocles-dev please helps close this issue and free to re-open it again if having other problem.

damocles-dev, 2022-03-21:
HEVC decoding is working now.

closed 2022-03-21 by damocles-dev