intel/media-driver #1359: VAAPI hardware acceleration is broken: `vaInitialize failed with error code 2 (resource allocation failed),exit`

AbhyudayaSharma, 2022-03-15:
## System information
- CPU information(`cat /proc/cpuinfo | grep "model name" | uniq`): Intel(R) Core(TM) i7-8705G CPU @ 3.10GHz
- GPU information(`lspci -nn | grep -E 'VGA|isplay`):
```
00:02.0 VGA compatible controller [0300]: Intel Corporation HD Graphics 630 [8086:591b] (rev 04)
00:13.0 Non-VGA unclassified device [0000]: Intel Corporation 100 Series/C230 Series Chipset Family Integrated Sensor Hub [8086:a135] (rev 31)
01:00.0 Display controller [0380]: Advanced Micro Devices, Inc. [AMD/ATI] Polaris 22 XL [Radeon RX Vega M GL] [1002:694e] (rev c0)
```
- Display server if rendering to display(X or wayland): Wayland

## Issue behavior
### Describe the current behavior

Video hardware acceleration is broken on my laptop with both the Intel HD Graphics as well as the Radeon GPU.

```
❯ DRI_PRIME=1 LIBVA_DRIVER_NAME=radeonsi vainfo 
iris: driver missing
libva error: /usr/lib/dri/radeonsi_drv_video.so init failed
vaInitialize failed with error code 2 (resource allocation failed),exit
```

However, when `LIBVA_DRIVER_NAME=iHD`, `vainfo` works perfectly but MPV output is broken as seen in the screenshot below:

[image]

```
[ffmpeg] AVHWDeviceContext: Failed to query surface attributes: 20 (the requested function is not implemented).
[vo/gpu/vaapi-egl] failed to retrieve libavutil frame constraints
```

<details>

```
❯ LIBVA_DRIVER_NAME=iHD vainfo
vainfo: VA-API version: 1.13 (libva 2.13.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.2.2 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointFEI
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointFEI
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointFEI
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointFEI
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSlice
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
```
</details>

With `DRI_PRIME=0` and `LIBVA_DRIVER_NAME=iHD`, video playback does work but keeps dropping frames. Video playback is unwatchable due to frame drops and stuttering. On average, in a 60fps video, 27 frames are dropped every second. 

### Describe the expected behavior

Video playback works without dropping any frames.

## Debug information
- What's libva/libva-utils/gmmlib/media-driver version? `22.2.2-1`
- Could you confirm whether GPU hardware exist or not by `ls /dev/dri`? 
  - Hardware exists
- Could you provide vainfo log if possible by `vainfo >vainfo.log 2>&1`?
  - See above
- Could you provide libva trace log if possible? Run cmd `export LIBVA_TRACE=/tmp/libva_trace.log` first then execute the case.

```
[29928.509702][ctx       none]==========va_TraceInitialize
[29928.509709][ctx       none]=========vaInitialize ret = VA_STATUS_ERROR_ALLOCATION_FAILED, resource allocation failed 
[29928.509714][ctx       none]==========va_TraceTerminate
[29928.509722][ctx       none]=========vaTerminate ret = VA_STATUS_SUCCESS, success (no error)
```

- Could you attach dmesg log if it's GPU hang by `dmesg >dmesg.log 2>&1`?

```
[16126.753267] i915 0000:00:02.0: cannot be used for peer-to-peer DMA as the client and provider (0000:01:00.0) do not share an upstream bridge or whitelisted host bridge
```
- Do you want to contribute a patch to fix the issue? (yes/no): maybe if someone can point me in the right direction

AbhyudayaSharma, 2022-03-15:
Also, the `radeonsi` driver does work when manually pointed to the card:

```
❯ vainfo --device /dev/dri/renderD128 --display drm
vainfo: VA-API version: 1.13 (libva 2.13.0)
vainfo: Driver version: Mesa Gallium driver 21.3.7 for AMD VEGAM (DRM 3.44.0, 5.16.14-arch1-1, LLVM 13.0.1)
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileNone                   : VAEntrypointVideoProc
```

AbhyudayaSharma, 2022-03-15:
@XinfengZhang Libva:

```
Version         : 2.13.0-2
Optional Deps   : intel-media-driver: backend for Intel GPUs (>= Broadwell) [installed]
                  libva-vdpau-driver: backend for Nvidia and AMD GPUs [installed]
                  libva-intel-driver: backend for Intel GPUs (<= Haswell)
```

MPV:

```
LIBVA_DRIVER_NAME=iHD DRI_PRIME=1 mpv --hwdec=auto --vo=gpu 'COSTA RICA IN 4K 60fps HDR (ULTRA HD) [LXb3EKWsInQ].webm' --msg-level=all=trace --log-file=/tmp/mpv.log
```

MPV log with `iHD`: [mpv.log](https://github.com/intel/media-driver/files/8254890/mpv.log)

> why gallium radeonsi does not work on your machine. TBH, it should not be reported here

Well it is an Intel+AMD chip. Moreover, it says `iris` so I thought it would belong here.

Libva logs:

[libva_trace.log.154519.thd-0x00000de5.txt](https://github.com/intel/media-driver/files/8254989/libva_trace.log.154519.thd-0x00000de5.txt)
[libva_trace.log.154519.thd-0x00000dda.txt](https://github.com/intel/media-driver/files/8254991/libva_trace.log.154519.thd-0x00000dda.txt)

AbhyudayaSharma, 2022-03-15:
And this is the mpv log for `radeonsi`:

```
[vo/gpu] Loading hwdec driver 'vaapi-egl'
[vo/gpu/vaapi-egl] using VAAPI EGL interop
[vo/gpu/vaapi-egl] Trying to open a x11 VA display...
[vo/gpu/vaapi-egl] Trying to open a wayland VA display...
[vo/gpu/vaapi-egl/vaapi] libva: VA-API version 1.13.0
[vo/gpu/vaapi-egl/vaapi] libva: User environment variable requested driver 'radeonsi'
[vo/gpu/vaapi-egl/vaapi] libva: Trying to open /usr/lib/dri/radeonsi_drv_video.so
[vo/gpu/vaapi-egl/vaapi] libva: Found init function __vaDriverInit_1_13
iris: driver missing
[vaapi] libva: /usr/lib/dri/radeonsi_drv_video.so init failed
[vo/gpu/vaapi-egl/vaapi] libva: va_openDriver() returns 2
[vo/gpu] Loading failed.
[vo/gpu] Loading hwdec driver 'cuda-nvdec'
Cannot load libcuda.so.1
```

Jexu, 2022-03-21:
Per iHD driver only, the libva-trace log shows invalid surface id sent to media driver to export the surface for display. So it looks like frame drop or corrupted. Perhaps the decode results are correct but getting invalid surface to display.

Could you try other app like sample decoder or ffmpeg to check if the decode yuv is correct?

AbhyudayaSharma, 2022-03-21:
@Jexu Do you mean [this](https://www.intel.com/content/www/us/en/developer/tools/media-sdk/code-samples.html) sample decoder?

Jexu, 2022-03-21:
Please get more help from https://github.com/Intel-Media-SDK/MediaSDK

AbhyudayaSharma, 2022-03-23:
@Jexu Thanks for the information. I first extracted the video stream from the .webm file as described [here](https://stackoverflow.com/questions/48587350/how-to-extract-raw-vp9-stream-from-webm-file). I get this error with sample_decode:

```
❯ DRI_PRIME=1 LIBVA_DRIVER_NAME=iHD ./sample_decode vp9 -d -p vp9d_hw -vaapi -i '/home/abhyudaya/Videos/stream.ivf' -w 3840 -h 2160 -o output.yuv

[ERROR], sts=MFX_ERR_UNSUPPORTED(-3), Init, m_mfxSession.InitEx failed at /home/abhyudaya/Source/MediaSDK/samples/sample_decode/src/pipeline_decode.cpp:437

[ERROR], sts=MFX_ERR_UNSUPPORTED(-3), main, Pipeline.Init failed at /home/abhyudaya/Source/MediaSDK/samples/sample_decode/src/sample_decode.cpp:807
```

Jexu, 2022-03-25:
Could you share the full log of failure. 
How about ffmpeg decode? If possible, you could share your stream.ivf to me to try.

AbhyudayaSharma, 2022-03-26:
@Jexu This is the entire output I get when running sample_decode. LIBVA_TRACE also does not create a log file. 

This is what happens with ffmpeg:

```
❯ LIBVA_TRACE=/tmp/libva_trace.log DRI_PRIME=1 LIBVA_DRIVER_NAME=iHD ffmpeg -loglevel verbose -init_hw_device vaapi
ffmpeg version n5.0 Copyright (c) 2000-2022 the FFmpeg developers
  built with gcc 11.2.0 (GCC)
  configuration: --prefix=/usr --disable-debug --disable-static --disable-stripping --enable-amf --enable-avisynth --enable-cuda-llvm --enable-lto --enable-fontconfig --enable-gmp --enable-gnutls --enable-gpl --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libdav1d --enable-libdrm --enable-libfreetype --enable-libfribidi --enable-libgsm --enable-libiec61883 --enable-libjack --enable-libmfx --enable-libmodplug --enable-libmp3lame --enable-libopencore_amrnb --enable-libopencore_amrwb --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-librav1e --enable-librsvg --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtheora --enable-libv4l2 --enable-libvidstab --enable-libvmaf --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxcb --enable-libxml2 --enable-libxvid --enable-libzimg --enable-nvdec --enable-nvenc --enable-shared --enable-version3
  libavutil      57. 17.100 / 57. 17.100
  libavcodec     59. 18.100 / 59. 18.100
  libavformat    59. 16.100 / 59. 16.100
  libavdevice    59.  4.100 / 59.  4.100
  libavfilter     8. 24.100 /  8. 24.100
  libswscale      6.  4.100 /  6.  4.100
  libswresample   4.  3.100 /  4.  3.100
  libpostproc    56.  3.100 / 56.  3.100
[AVHWDeviceContext @ 0x5560f6cdc440] Trying to use DRM render node for device 0.
[AVHWDeviceContext @ 0x5560f6cdc440] libva: Open new log file /tmp/libva_trace.log.101307.thd-0x0000115c for the thread 0x0000115c
[AVHWDeviceContext @ 0x5560f6cdc440] libva: LIBVA_TRACE is on, save log into /tmp/libva_trace.log.101307.thd-0x0000115c
[AVHWDeviceContext @ 0x5560f6cdc440] libva: VA-API version 1.13.0
[AVHWDeviceContext @ 0x5560f6cdc440] libva: User environment variable requested driver 'iHD'
[AVHWDeviceContext @ 0x5560f6cdc440] libva: Trying to open /usr/lib/dri/iHD_drv_video.so
[AVHWDeviceContext @ 0x5560f6cdc440] libva: Found init function __vaDriverInit_1_13
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [2]
param: 4, val: 0
[AVHWDeviceContext @ 0x5560f6cdc440] libva: /usr/lib/dri/iHD_drv_video.so init failed
[AVHWDeviceContext @ 0x5560f6cdc440] libva: va_openDriver() returns 18
[AVHWDeviceContext @ 0x5560f6cdc440] Failed to initialise VAAPI connection: 18 (invalid parameter).
Device creation failed: -5.
Failed to set value 'vaapi' for option 'init_hw_device': Input/output error
Error parsing global options: Input/output error
```

This is the libva trace:

```
[59187.724968][ctx       none]==========va_TraceInitialize
[59187.724973][ctx       none]=========vaInitialize ret = VA_STATUS_ERROR_INVALID_PARAMETER, invalid parameter 
[59187.724978][ctx       none]==========va_TraceTerminate
[59187.724985][ctx       none]=========vaTerminate ret = VA_STATUS_SUCCESS, success (no error) 
```

This is the video I was trying to play https://www.youtube.com/watch?v=LXb3EKWsInQ. I downloaded it using [yt-dlp](https://github.com/yt-dlp/yt-dlp).

Jexu, 2022-03-28:
From above log, it seems that amd gpu node is opened, but with iHD driver used.
Can you give all gpu node 'll /dev/dri'
and try to open with other gpu node like: LIBVA_TRACE=/tmp/libva_trace.log DRI_PRIME=1 LIBVA_DRIVER_NAME=iHD ffmpeg -loglevel verbose -init_hw_device vaapi=/dev/dri/card0

AbhyudayaSharma, 2022-03-28:
```
❯ ll /dev/dri
total 0
drwxr-xr-x  2 root root        120 Mar 28 07:46 by-path/
crw-rw----+ 1 root video  226,   0 Mar 28 07:46 card0
crw-rw----+ 1 root video  226,   1 Mar 28 07:45 card1
crw-rw-rw-  1 root render 226, 128 Mar 28 07:45 renderD128
crw-rw-rw-  1 root render 226, 129 Mar 28 07:45 renderD129
```

I tried ffmpeg with both `vaapi=/dev/dri/card0` and `vaapi=/dev/dri/card1` and I still get the same error as in my [previous comment](https://github.com/intel/media-driver/issues/1359#issuecomment-1079655886). Setting `DRI_PRIME` to 0 or 1 also does not create any difference.

AbhyudayaSharma, 2022-03-28:
@jexu I also get this in journalctl:

```
Mar 28 10:23:29 arch kernel: i915 0000:00:02.0: cannot be used for peer-to-peer DMA as the client and provider (0000:01:00.0) do not share an upstream bridge or whitelisted host bridge
```

Jexu, 2022-03-28:
What is your kernel version?

iHD driver calls the drm ioctl to get some hw infos from kernel, but your kernel returns invalid result with failure. Perhaps something wrong in kernel.
Please capture the dmesg log with setting ‘echo 0xff > /sys/module/drm/parameters/debug’

_DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [2]
param: 4, val: 0_

Jexu, 2022-03-28:
[query_hw_info.txt](https://github.com/intel/media-driver/files/8360645/query_hw_info.txt)

please try this tool to query the hw info directly with the kernel: ./query_hw_info.txt --device=/dev/dri/renderD128 and ./query_hw_info.txt --device=/dev/dri/renderD129

then share the log on screen.

AbhyudayaSharma, 2022-03-28:
```
device = /dev/dri/renderD128
fd = 3
drm-info: version = 3.44.0
drm-info: name = amdgpu
drm-info: date = 20150101
drm-info: desc = AMD GPU

INFO: DRM_IOCTL_I915_GETPARAM 

INFO: DRM_IOCTL_I915_GEM_GET_APERTURE 
ERROR: DRM_IOCTL_I915_GEM_GET_APERTURE failed: Invalid argument
INFO: aper_size = 0, aper_available_size = 0

INFO: DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM 
ERROR: DRM_IOCTL_I915_QUERY failed
DeviceConfig[0, 0x0, 0, 0x0, 0, 0, 0, 0, 0, 0]
fd closed
```

```
device = /dev/dri/renderD129
fd = 3
drm-info: version = 1.6.0
drm-info: name = i915
drm-info: date = 20201103
drm-info: desc = Intel Graphics

INFO: DRM_IOCTL_I915_GETPARAM 
I915_PARAM_CHIPSET_ID            = 22811
I915_PARAM_CMD_PARSER_VERSION    = 10
I915_PARAM_CS_TIMESTAMP_FREQUENCY  = 12000000
I915_PARAM_EU_TOTAL              = 23
I915_PARAM_HAS_ALIASING_PPGTT    = 2
I915_PARAM_HAS_BLT               = 1
I915_PARAM_HAS_BSD               = 1
I915_PARAM_HAS_BSD2              = 0
I915_PARAM_HAS_COHERENT_PHYS_GTT = 1
I915_PARAM_HAS_COHERENT_RINGS    = 1
I915_PARAM_HAS_CONTEXT_ISOLATION   = 15
I915_PARAM_HAS_EXECBUF2          = 1
I915_PARAM_HAS_EXEC_ASYNC          = 1
I915_PARAM_HAS_EXEC_BATCH_FIRST    = 1
I915_PARAM_HAS_EXEC_CAPTURE        = 1
I915_PARAM_HAS_EXEC_FENCE          = 1
I915_PARAM_HAS_EXEC_FENCE_ARRAY    = 1
I915_PARAM_HAS_EXEC_HANDLE_LUT   = 1
I915_PARAM_HAS_EXEC_NO_RELOC     = 1
I915_PARAM_HAS_EXEC_SOFTPIN      = 1
I915_PARAM_HAS_EXEC_SUBMIT_FENCE   = 1
I915_PARAM_HAS_EXEC_TIMELINE_FENCES = 1
I915_PARAM_HAS_GEM               = 1
I915_PARAM_HAS_GEN7_SOL_RESET    = 1
I915_PARAM_HAS_GPU_RESET         = 2
I915_PARAM_HAS_LLC               = 1
I915_PARAM_HAS_OVERLAY           = 0
I915_PARAM_HAS_PAGEFLIPPING      = 1
I915_PARAM_HAS_PINNED_BATCHES    = 1
I915_PARAM_HAS_POOLED_EU         = 0
I915_PARAM_HAS_PRIME_VMAP_FLUSH  = 1
I915_PARAM_HAS_RELAXED_DELTA     = 1
I915_PARAM_HAS_RELAXED_FENCING   = 1
I915_PARAM_HAS_RESOURCE_STREAMER = 0
I915_PARAM_HAS_SCHEDULER           = 31
I915_PARAM_HAS_SECURE_BATCHES    = 0
I915_PARAM_HAS_SEMAPHORES        = 1
I915_PARAM_HAS_USERPTR_PROBE        = 1
I915_PARAM_HAS_VEBOX             = 1
I915_PARAM_HAS_WAIT_TIMEOUT      = 1
I915_PARAM_HAS_WT                = 0
I915_PARAM_MIN_EU_IN_POOL        = 0
I915_PARAM_MMAP_GTT_COHERENT       = 1
I915_PARAM_MMAP_GTT_VERSION      = 4
I915_PARAM_MMAP_VERSION          = 1
I915_PARAM_NUM_FENCES_AVAIL      = 32
I915_PARAM_PERF_REVISION            = 5
I915_PARAM_REVISION              = 4
I915_PARAM_SLICE_MASK              = 1
I915_PARAM_SUBSLICE_MASK           = 7
I915_PARAM_SUBSLICE_TOTAL        = 3

INFO: DRM_IOCTL_I915_GEM_GET_APERTURE 
INFO: aper_size = 4294967296, aper_available_size = 4193763328

INFO: DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM 
I915_CONTEXT_PARAM_BANNABLE           = 1
I915_CONTEXT_PARAM_GTT_SIZE           = 281474976710656
I915_CONTEXT_PARAM_NO_ERROR_CAPTURE   = 0
I915_CONTEXT_PARAM_PERSISTENCE        = 1
I915_CONTEXT_PARAM_PRIORITY           = 0
I915_CONTEXT_PARAM_PROTECTED_CONTENT  = 0
I915_CONTEXT_PARAM_RECOVERABLE        = 1
I915_CONTEXT_PARAM_SSEU               = 0
I915_CONTEXT_PARAM_VM                 = 1
INFO: engine_class = 0, engine_instance = 0, capabilities = 0, flags = 1
INFO: engine_class = 1, engine_instance = 0, capabilities = 0, flags = 1
INFO: engine_class = 2, engine_instance = 0, capabilities = 3, flags = 1
INFO: engine_class = 3, engine_instance = 0, capabilities = 0, flags = 1
DeviceConfig[4193763328, 0x591b, 4, 0x3ff, 32, 2, 3, 23, 0, 281474976710656]
fd closed
```

Jexu, 2022-03-28:
So it is obvious that renderD128 is amd gpu while renderD129 is intel gpu. But your above failure is caused by iHD driver running on amd drm module.

How about LIBVA_TRACE=/tmp/libva_trace.log DRI_PRIME=1 LIBVA_DRIVER_NAME=iHD ffmpeg -loglevel verbose -init_hw_device vaapi=**/dev/dri/renderD129**

AbhyudayaSharma, 2022-03-28:
@Jexu I get the same error unfortunately:

```
❯ LIBVA_TRACE=/tmp/libva_trace.log DRI_PRIME=1 LIBVA_DRIVER_NAME=iHD ffmpeg -loglevel verbose -init_hw_device vaapi=/dev/dri/renderD129
ffmpeg version n5.0 Copyright (c) 2000-2022 the FFmpeg developers
  built with gcc 11.2.0 (GCC)
  configuration: --prefix=/usr --disable-debug --disable-static --disable-stripping --enable-amf --enable-avisynth --enable-cuda-llvm --enable-lto --enable-fontconfig --enable-gmp --enable-gnutls --enable-gpl --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libdav1d --enable-libdrm --enable-libfreetype --enable-libfribidi --enable-libgsm --enable-libiec61883 --enable-libjack --enable-libmfx --enable-libmodplug --enable-libmp3lame --enable-libopencore_amrnb --enable-libopencore_amrwb --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-librav1e --enable-librsvg --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtheora --enable-libv4l2 --enable-libvidstab --enable-libvmaf --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxcb --enable-libxml2 --enable-libxvid --enable-libzimg --enable-nvdec --enable-nvenc --enable-shared --enable-version3
  libavutil      57. 17.100 / 57. 17.100
  libavcodec     59. 18.100 / 59. 18.100
  libavformat    59. 16.100 / 59. 16.100
  libavdevice    59.  4.100 / 59.  4.100
  libavfilter     8. 24.100 /  8. 24.100
  libswscale      6.  4.100 /  6.  4.100
  libswresample   4.  3.100 /  4.  3.100
  libpostproc    56.  3.100 / 56.  3.100
[AVHWDeviceContext @ 0x55f117a5b3c0] Trying to use DRM render node for device 0.
[AVHWDeviceContext @ 0x55f117a5b3c0] libva: Open new log file /tmp/libva_trace.log.122600.thd-0x00003053 for the thread 0x00003053
[AVHWDeviceContext @ 0x55f117a5b3c0] libva: LIBVA_TRACE is on, save log into /tmp/libva_trace.log.122600.thd-0x00003053
[AVHWDeviceContext @ 0x55f117a5b3c0] libva: VA-API version 1.13.0
[AVHWDeviceContext @ 0x55f117a5b3c0] libva: User environment variable requested driver 'iHD'
[AVHWDeviceContext @ 0x55f117a5b3c0] libva: Trying to open /usr/lib/dri/iHD_drv_video.so
[AVHWDeviceContext @ 0x55f117a5b3c0] libva: Found init function __vaDriverInit_1_13
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [2]
param: 4, val: 0
[AVHWDeviceContext @ 0x55f117a5b3c0] libva: /usr/lib/dri/iHD_drv_video.so init failed
[AVHWDeviceContext @ 0x55f117a5b3c0] libva: va_openDriver() returns 18
[AVHWDeviceContext @ 0x55f117a5b3c0] Failed to initialise VAAPI connection: 18 (invalid parameter).
Device creation failed: -5.
Failed to set value 'vaapi=/dev/dri/renderD129' for option 'init_hw_device': Input/output error
Error parsing global options: Input/output error
```

Jexu, 2022-03-28:
Look like that vaapi=/dev/dri/renderD129 cann't force iHD driver calling into i915 module. Do your try to disable amd gpu?

AbhyudayaSharma, 2022-03-28:
No. I haven't done anything like that. I just have these environment variables setup:

```
DRI_PRIME=1
LIBVA_DRIVER_NAME=radeonsi
VDPAU_DRIVER=radeonsi
```

I don't think these should cause any problems.

Jexu, 2022-03-28:
unset DRI_PRIME and VDPAU_DRIVER since DRI_PRIME =1 will chose dgpu firstly.
then DRI_PRIME=0 LIBVA_DRIVER_NAME=iHD ffmpeg -loglevel verbose -init_hw_device vaapi=/dev/dri/renderD129

AbhyudayaSharma, 2022-03-29:
@Jexu I unset VDPAU_DRIVER and DRI_PRIME and rebooted. I still get the same message:

```
❯ LIBVA_TRACE=/tmp/libva_trace.log DRI_PRIME=0 LIBVA_DRIVER_NAME=iHD ffmpeg -loglevel verbose -init_hw_device vaapi=/dev/dri/renderD129
ffmpeg version n5.0 Copyright (c) 2000-2022 the FFmpeg developers
  built with gcc 11.2.0 (GCC)
  configuration: --prefix=/usr --disable-debug --disable-static --disable-stripping --enable-amf --enable-avisynth --enable-cuda-llvm --enable-lto --enable-fontconfig --enable-gmp --enable-gnutls --enable-gpl --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libdav1d --enable-libdrm --enable-libfreetype --enable-libfribidi --enable-libgsm --enable-libiec61883 --enable-libjack --enable-libmfx --enable-libmodplug --enable-libmp3lame --enable-libopencore_amrnb --enable-libopencore_amrwb --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-librav1e --enable-librsvg --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtheora --enable-libv4l2 --enable-libvidstab --enable-libvmaf --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxcb --enable-libxml2 --enable-libxvid --enable-libzimg --enable-nvdec --enable-nvenc --enable-shared --enable-version3
  libavutil      57. 17.100 / 57. 17.100
  libavcodec     59. 18.100 / 59. 18.100
  libavformat    59. 16.100 / 59. 16.100
  libavdevice    59.  4.100 / 59.  4.100
  libavfilter     8. 24.100 /  8. 24.100
  libswscale      6.  4.100 /  6.  4.100
  libswresample   4.  3.100 /  4.  3.100
  libpostproc    56.  3.100 / 56.  3.100
[AVHWDeviceContext @ 0x5610c33623c0] Trying to use DRM render node for device 0.
[AVHWDeviceContext @ 0x5610c33623c0] libva: Open new log file /tmp/libva_trace.log.114700.thd-0x00000f75 for the thread 0x00000f75
[AVHWDeviceContext @ 0x5610c33623c0] libva: LIBVA_TRACE is on, save log into /tmp/libva_trace.log.114700.thd-0x00000f75
[AVHWDeviceContext @ 0x5610c33623c0] libva: VA-API version 1.13.0
[AVHWDeviceContext @ 0x5610c33623c0] libva: User environment variable requested driver 'iHD'
[AVHWDeviceContext @ 0x5610c33623c0] libva: Trying to open /usr/lib/dri/iHD_drv_video.so
[AVHWDeviceContext @ 0x5610c33623c0] libva: Found init function __vaDriverInit_1_13
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [2]
param: 4, val: 0
[AVHWDeviceContext @ 0x5610c33623c0] libva: /usr/lib/dri/iHD_drv_video.so init failed
[AVHWDeviceContext @ 0x5610c33623c0] libva: va_openDriver() returns 18
[AVHWDeviceContext @ 0x5610c33623c0] Failed to initialise VAAPI connection: 18 (invalid parameter).
Device creation failed: -5.
Failed to set value 'vaapi=/dev/dri/renderD129' for option 'init_hw_device': Input/output error
Error parsing global options: Input/output error
```

```
❯ echo $DRI_PRIME ${VDPAU_DRIVER:=unset} ${LIBVA_DRIVER_NAME:=unset}
0 unset unset
```

AbhyudayaSharma, 2022-03-29:
Interestingly, I also see this:

```
❯ env -u DRI_PRIME LIBVA_DRIVER_NAME=iHD vainfo --display drm --device /dev/dri/renderD128
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [2]
param: 4, val: 0
libva error: /usr/lib/dri/iHD_drv_video.so init failed
vaInitialize failed with error code 18 (invalid parameter),exit
```

This is the same error we get with ffmpeg.

What I think happens is that ffmpeg uses renderD128 even when tell it to use renderD129

Jexu, 2022-03-29:
You can refere to ffmpeg wiki https://trac.ffmpeg.org/wiki/Hardware/VAAPI to get more input option.
Anyway, is your issue solved with iHD driver?

Jexu, 2022-04-26:
Since the issue is caused by running iHD driver on amd gpu by wrong env variable setting. I will closed it and you could re-open it again if any other problems.

closed 2022-04-26 by Jexu