intel/media-driver #1364: [build] gmmlib commit: 29152fc6 cause media-driver build error

Bin-CI, 2022-03-23:
media-driver commit: https://github.com/intel/media-driver/commit/34d9e7870f250c91d7e5a9c3f0d9e3c54fae1195
gmmlib issue: https://github.com/intel/gmmlib/issues/96

Bin-CI, 2022-03-23:
build error information:
```
20:38:02,669 INFO  - In file included from /opt/media/build/intel-media-driver/media_driver/linux/common/ddi/media_libva_common.h:36:0,
20:38:02,669 INFO  -                  from /opt/media/build/intel-media-driver/media_driver/linux/common/cm/ddi/media_libva_cm.h:30,
20:38:02,669 INFO  -                  from /opt/media/build/intel-media-driver/media_driver/linux/common/cm/hal/cm_def_os.h:41,
20:38:02,669 INFO  -                  from /opt/media/build/intel-media-driver/media_driver/agnostic/common/cm/cm_def.h:30,
20:38:02,669 INFO  -                  from /opt/media/build/intel-media-driver/media_driver/agnostic/common/cm/cm_buffer.h:30,
20:38:02,669 INFO  -                  from /opt/media/build/intel-media-driver/media_driver/agnostic/common/cm/cm_buffer_rt.h:30,
20:38:02,669 INFO  -                  from /opt/media/build/intel-media-driver/media_driver/agnostic/common/cm/cm_buffer_rt.cpp:27:
20:38:02,669 INFO  - /opt/media/build/intel-media-driver/media_driver/linux/common/os/i915/include/mos_bufmgr.h:391:79: error: 'GFX_GMD_ID' has not been declared
20:38:02,669 INFO  -  int mos_query_hw_ip_version(int fd, struct i915_engine_class_instance engine, GFX_GMD_ID *ipVerInfo);
20:38:02,669 INFO  -                                                                                ^~~~~~~~~~
20:38:02,716 INFO  - cd /opt/media/build/intel-media-driver/__cmake_build/cmrtlib/linux && /usr/bin/cmake -E cmake_symlink_library libigfxcmrt.so.7.2.0 libigfxcmrt.so.7 libigfxcmrt.so
20:38:02,722 INFO  - make[2]: Leaving directory '/opt/media/build/intel-media-driver/__cmake_build'
20:38:02,725 INFO  - [  2%] Built target igfxcmrt
20:38:02,766 INFO  - In file included from /opt/media/build/intel-media-driver/media_driver/linux/common/ddi/media_libva_common.h:36:0,
20:38:02,766 INFO  -                  from /opt/media/build/intel-media-driver/media_driver/linux/common/cm/ddi/media_libva_cm.h:30,
20:38:02,766 INFO  -                  from /opt/media/build/intel-media-driver/media_driver/linux/common/cm/hal/cm_def_os.h:41,
20:38:02,766 INFO  -                  from /opt/media/build/intel-media-driver/media_driver/agnostic/common/cm/cm_def.h:30,
20:38:02,766 INFO  -                  from /opt/media/build/intel-media-driver/media_driver/agnostic/common/cm/cm_buffer.h:30,
20:38:02,766 INFO  -                  from /opt/media/build/intel-media-driver/media_driver/agnostic/common/cm/cm_buffer_rt.h:30,
20:38:02,766 INFO  -                  from /opt/media/build/intel-media-driver/media_driver/agnostic/common/cm/cm_state_buffer.h:30,
20:38:02,766 INFO  -                  from /opt/media/build/intel-media-driver/media_driver/agnostic/common/cm/cm_state_buffer.cpp:27:
20:38:02,766 INFO  - /opt/media/build/intel-media-driver/media_driver/linux/common/os/i915/include/mos_bufmgr.h:391:79: error: 'GFX_GMD_ID' has not been declared
20:38:02,766 INFO  -  int mos_query_hw_ip_version(int fd, struct i915_engine_class_instance engine, GFX_GMD_ID *ipVerInfo);
20:38:02,766 INFO  -                                                                                ^~~~~~~~~~
20:38:03,106 INFO  - media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:103: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/cm/cm_state_buffer.cpp.o' failed
20:38:03,106 INFO  - make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/cm/cm_state_buffer.cpp.o] Error 1
20:38:03,246 INFO  - media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:89: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/cm/cm_buffer_rt.cpp.o' failed
20:38:03,246 INFO  - make[2]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/cm/cm_buffer_rt.cpp.o] Error 1
20:38:03,247 INFO  - make[2]: Leaving directory '/opt/media/build/intel-media-driver/__cmake_build'
20:38:03,247 INFO  - CMakeFiles/Makefile2:244: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/all' failed
20:38:03,247 INFO  - make[1]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/all] Error 2
20:38:03,247 INFO  - make[1]: Leaving directory '/opt/media/build/intel-media-driver/__cmake_build'
20:38:03,247 INFO  - Makefile:155: recipe for target 'all' failed
20:38:03,247 INFO  - make: *** [all] Error 2
```

Sherry-Lin, 2022-03-23:
it's related to GmmLib ABI broken by https://github.com/intel/gmmlib/commit/8dcebfefaaacff0cbc262fe050b2b0fd060898d0, please update media code after https://github.com/intel/media-driver/commit/79c20b78a and gmmlib code after https://github.com/intel/gmmlib/commit/29152fc6d39f87983b176664f225d42af039f716

Bin-CI, 2022-03-23:
fixed

closed 2022-03-23 by Bin-CI

Sherry-Lin, 2022-03-23:
Detail in https://github.com/intel/gmmlib/issues/95