intel/media-driver #1366: Video decode fails with "operation failed" on Tiger Lake

lhw, 2022-03-24:
## System information
- CPU information(`cat /proc/cpuinfo | grep "model name" | uniq`): 11th Gen Intel(R) Core(TM) i7-1165G7
- GPU information(`lspci -nn | grep -E 'VGA|isplay`): Intel Corporation TigerLake-LP GT2 [Iris Xe Graphics] [8086:9a49]
- Display server if rendering to display(X or wayland): wayland

## Issue behavior
### Describe the current behavior
The va-api library returns with code 1 operation failed. See the vainfo log below. Usage via mpv and web browsers is also impossible. i.e. with mpv
```
[vaapi] libva: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed
```
but runs fine with the gpu backend which then bypasses the video decoder
```
         ENGINES     BUSY                                                                                        MI_SEMA MI_WAIT
       Render/3D   29.24% |████████████████████████▋                                                           |      0%      0%
         Blitter    0.00% |                                                                                    |      0%      0%
           Video    0.00% |                                                                                    |      0%      0%
    VideoEnhance    0.00% |                                                                                    |      0%      0%
```

### Describe the expected behavior

## Debug information
- What's libva/libva-utils/gmmlib/media-driver version? libva 2.14.0 / libva-utils 2.12.0 / gmmlib 22.1.1 / media-driver 22.2.1 (current packages on Ubuntu 22.04)
- Could you confirm whether GPU hardware exist or not by `ls /dev/dri`? card0  renderD128
- Could you provide vainfo log if possible by `vainfo >vainfo.log 2>&1`?
  ```
  % LIBVA_DRIVERS_PATH=/usr/lib/x86_64-linux-gnu/dri LIBVA_DRIVER_NAME=iHD vainfo
  libva info: VA-API version 1.14.0
  libva info: User environment variable requested driver 'iHD'
  libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
  libva info: Found init function __vaDriverInit_1_14
  libva error: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed
  libva info: va_openDriver() returns 1
  vaInitialize failed with error code 1 (operation failed),exit
  ```
- Could you provide libva trace log if possible? Run cmd `export LIBVA_TRACE=/tmp/libva_trace.log` first then execute the case.
  ```
  [55000.599878][ctx       none]==========va_TraceInitialize
  [55000.599881][ctx       none]=========vaInitialize ret = VA_STATUS_ERROR_UNKNOWN, unknown libva error 
  [55000.599884][ctx       none]==========va_TraceTerminate
  [55000.599901][ctx       none]=========vaTerminate ret = VA_STATUS_SUCCESS, success (no error) 
   ```
- Could you attach dmesg log if it's GPU hang by `dmesg >dmesg.log 2>&1`? /
- Do you want to contribute a patch to fix the issue? (yes/no): no

rathann, 2022-03-28:
Same here on Fedora 36 pre-release with:
```
intel-gmmlib-22.1.0-1.fc36
intel-media-driver-22.1.1-2.fc36.x86_64
libva-2.14.0-1.fc36.x86_64
```
Downgrading to 21.x series makes it work again, so this looks like a regression.

rathann, 2022-03-28:
It looks like a simple rebuild of `intel-media-driver` package fixed things for me. VA-API is working again with:
```
intel-gmmlib-22.1.0-1.fc36.x86_64
intel-media-driver-22.1.1-3.fc36.x86_64
```
Downstream bug report: https://bugzilla.rpmfusion.org/show_bug.cgi?id=6248 .

lhw, 2022-03-29:
I can report the same for Ubuntu 22.04. A simple recompile of both packages solves the problem. So my guess would be that this is not an upstream issue per se. Maybe just indirect due to changes with the ABI. Feel free to close the issue if you consider the problem solved.

rathann, 2022-04-05:
Unfortunately, the issue is back after updating intel-gmmlib from 22.1.0 to 22.1.2 without rebuilding intel-media-driver. Rebuild helps, so there's definitely some ABI incompatibility between minor intel-gmmlib versions.

seb128, 2022-04-05:
the new update fixed the ABI to be compatible with older versions but it means if the media driver was built with the buggy .0 then it needs another rebuild now

rathann, 2022-04-05:
Oh, so 22.1.0 was buggy? Too bad, but thanks for letting us know. Which versions are ABI-compatible, then? 22.1.2 isn't compatible with 22.0.3, either.

seb128, 2022-04-05:
see https://github.com/intel/gmmlib/issues/95 , it states that 22.0.3 to 22.1.1 are incompatible with the old ABI

rathann, 2022-04-05:
Thanks for the link. It explains the issue well. I can confirm that media-driver built with either gmmlib 22.0.2 or 22.1.2 works with either, too.

lhw, 2022-04-08:
I think we can go ahead and close the issue then

closed 2022-04-08 by lhw