intel/media-driver #1367: Unable to build Intel-Media-Driver

ngboonhai, 2022-03-24:
## System information
- Impacted GPU platform: Intel ADL-P 1265UE iGPU 
- System OS: Ubuntu 20.04
- Kernel: 5.16.12-051612-generic 
- Current libva/libva-utils/gmmlib/media-driver version?
libva-drm2:amd64                            2.7.0-2 
libva-wayland2:amd64                        2.7.0-2
libva-x11-2:amd64                           2.7.0-2
libva2:amd64                                2.7.0-2 
intel-media-va-driver:amd64                 20.1.1+dfsg1-1
intel-gmmlib                                21.2.1 

## Issue behavior
In file included from /home/user/media-driver/media_driver/linux/common/ddi/media_libva_putsurface_linux.cpp:46:
/home/user/media-driver/media_driver/linux/common/ddi/meda_libva_putsurface_linux.h:31:10: fatal error: va/va_dricommon.h: No such file or directory
   31 | #include <va/va_dricommon.h>
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.

### Describe the current behavior
When run cmake ../media-driver and output as below detect X11 modules not found, but libva-x11-2 package is installed as per above mentioned.
-- media -- X11 Found
-- Checking for module 'libva-x11'
--   No package 'libva-x11' found
-- LIBVA_DRIVERS_PATH = /usr/local/lib/dri

#Run vainfo and here is the output, iHD_video.so init failed.
libva info: VA-API version 1.10.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /opt/intel/mediasdk/lib64/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_10
libva error: /opt/intel/mediasdk/lib64/iHD_drv_video.so init failed
libva info: va_openDriver() returns 1
vaInitialize failed with error code 1 (operation failed),exit

XinfengZhang, 2022-03-25:
please install libva-X11 in your machine.  media driver build found X11, but libva is without X11.

ngboonhai, 2022-03-25:
XinfengZhang.
          Alright. found libva-x11-dev not proper install, issues resolved.

XinfengZhang, 2022-03-27:
@ngboonhai lets close the issue?

ngboonhai, 2022-03-27:
let, please close this and thank you.

closed 2022-03-27 by ngboonhai