intel/media-driver #1371: Alder Lake init failed

Brampie66, 2022-03-28:
Hi,

I'm trying to get vaapi to work on an intel i7 12700 (alder lake), the pc is running centos8
vainfo:
Found init function _vaDriverInit_1_14
/usr/lib64/dri/iHD_drv_video.so init failed
va_openDriver() returns 1

I've tried and checked the following:
Linux localhost.localdomain 5.17.0-1.el8.elrepo.x86_64  , I updated linux-firmware as well, this dmesg part looks ok to me
[    1.516969] fb0: switching to i915 from simple
[    1.517131] i915 0000:00:02.0: vgaarb: deactivate vga console
[    1.517903] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    1.518405] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adls_dmc_ver2_01.bin (v2.1)
[    1.534970] i915 0000:00:02.0: [drm] GuC firmware i915/tgl_guc_62.0.0.bin version 62.0 submission:disabled
[    1.534983] i915 0000:00:02.0: [drm] GuC SLPC: disabled
[    1.534986] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc_7.9.3.bin version 7.9 authenticated:yes
[    1.539418] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    1.554958] fbcon: i915drmfb (fb0) is primary device
[    1.609166] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[    3.046435] snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])

media_driver is 22.2.2 , libva is 2.14
xorg log:
[     7.085] (==) modeset(0): Backing store enabled
[     7.085] (==) modeset(0): Silken mouse enabled
[     7.215] (II) modeset(0): Initializing kms color map for depth 24, 8 bpc.
[     7.215] (==) modeset(0): DPMS enabled
[     7.215] (II) modeset(0): [DRI2] Setup complete
[     7.215] (II) modeset(0): [DRI2]   DRI driver: iris
[     7.215] (II) modeset(0): [DRI2]   VDPAU driver: va_gl
........
[     7.217] (II) Initializing extension DRI3

I tried the same combo (libva,media-driver,gmmlib,kernel) on a much older J5005 and this works ok.

I was under the impression vaapi should work for this cpu on that kernel, or am I mistaken ?
How can I check to see why vainfo is failing ?

thx
Bram

aloise, 2022-04-05:
I have exactly the same error on Arch:

```
# pacman --query intel-media-driver 
intel-media-driver 22.2.2-1

# Linux aloise-i12700 5.17.1-arch1-1 #1 SMP
Linux pc-i12700 5.17.1-arch1-1 #1 SMP
```

paulmenzel, 2022-08-11:
We are seeing this on the Intel Alder Lake device Dell XPS 13 9315 with

    00:02.0 VGA compatible controller [0300]: Intel Corporation Alder Lake-UP4 GT2 [Iris Xe Graphics] [8086:46aa] (rev 0c)

running Debian sid/unstable currently with *libva2* 2.15.0-1, and *intel-media-driver-non-free* 22.5.1+ds1-1.

```
$ vainfo
libva info: VA-API version 1.15.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_15
libva error: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 1
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit
```

The issue was reported at the end of March, and almost five months have past without a roadmap. It’d be great, if you could give one.

@mrhpearson, I am assuming Lenovo Thinkpads are also affected. Are you aware of this?

dvrogozh, 2022-08-11:
There might be 2 rather than 1 issue here.

1. First issue is why intel i7 12700 does not work? It has ``0x4680`` device ID and it's listed here in the driver code. It should be supported it seems:

https://github.com/intel/media-driver/blob/ccc137c9281eb874c4d2a8c85defb4aae81a75ad/media_driver/linux/gen12/ddi/media_sysinfo_g12.cpp#L456

Patch for it is long available, see https://github.com/intel/media-driver/commit/1fefcd0080355102c16cbb4908625d02b8151efd. I've no idea what's wrong here. Someone needs to debug. In a meanwhile, @Brampie66, @aloise : do you still observe the issue on a more recent driver builds for this device?

2. Second issue is reported by @paulmenzel that ``0x46aa`` is not supported. It's not a surprise this one fails - it's missing in device enumeration list in the driver. Checking other ADLP device ids enlisted in https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/include/drm/i915_pciids.h?h=v5.19#n649 the following are missing in media-driver: ``46a2, 46aa, 462a``

dvrogozh, 2022-08-19:
Second issue (missing device ids) got addressed by https://github.com/intel/media-driver/commit/1350c5ae9aff9e8f1f5f5c61c1262dd747bd0838 (media-driver) and https://github.com/oneapi-src/oneVPL-intel-gpu/commit/e790c7d8bdd7d2ff55aaf252548fdc7e211e0ebf (adding here for consistency).

@paulmenzel : can you, please, help rebuild media-driver and try again on your side under Intel Alder Lake device Dell XPS 13 9315? Please, let us know the behavior.

Brampie66, 2022-08-26:
Hi,

I did some updates and tried again:
[bram@localhost ~]$ export LIBVA_DRIVER_NAME=iHD
[bram@localhost ~]$ vainfo
libva info: VA-API version 1.15.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib64/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_15
libva error: /usr/lib64/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 1
vaInitialize failed with error code 1 (operation failed),exit

dmesg:
[    2.085561] i915 0000:00:02.0: vgaarb: deactivate vga console
[    2.086205] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    2.086684] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adls_dmc_ver2_01.bin (v2.1)
[    2.087228] i915 0000:00:02.0: [drm] GuC firmware i915/tgl_guc_70.1.1.bin is recommended, but only i915/tgl_guc_69.0.3.bin was found
[    2.087229] i915 0000:00:02.0: [drm] Consider updating your linux-firmware pkg or downloading from https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915
[    2.089381] i915 0000:00:02.0: [drm] GuC firmware i915/tgl_guc_69.0.3.bin version 69.0
[    2.089382] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc_7.9.3.bin version 7.9
[    2.103079] i915 0000:00:02.0: [drm] HuC authenticated
[    2.103084] i915 0000:00:02.0: [drm] GuC submission disabled
[    2.103087] i915 0000:00:02.0: [drm] GuC SLPC disabled
[    2.128379] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    2.128912] ACPI: video: Video Device [GFX0] (multi-head: yes  rom: no  post: no)

packages:
linux-firmware.noarch                                                           20220815-138.fc38   
intel-gmmlib.x86_64                                                               1:22.1.7-1.el8 
intel-media-driver.x86_64                                                         22.5.3-1.el8 
libva.x86_64                                                                      1:2.15.0-1.el8
Kernel -ml 5.19.4-1.el8.elrepo.x86_64 

thx
Bram

dvrogozh, 2022-08-26:
I do not see this issue running out-of-the-box media driver on Ubuntu 22.04:
```
$ cat /proc/cpuinfo  | grep name
model name      : 12th Gen Intel(R) Core(TM) i7-12700K

$ uname -a
Linux DUT102-DG2FRD 5.15.0-25-generic #25-Ubuntu SMP Wed Mar 30 15:54:22 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
$ sudo dmesg | grep drm
[    3.666801] systemd[1]: Starting Load Kernel Module drm...
[    3.680434] systemd[1]: modprobe@drm.service: Deactivated successfully.
[    3.680861] systemd[1]: Finished Load Kernel Module drm.
[    4.044046] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    4.048528] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adls_dmc_ver2_01.bin (v2.1)
[    4.198480] i915 0000:00:02.0: [drm] GuC firmware i915/tgl_guc_62.0.0.bin version 62.0 submission:disabled
[    4.198486] i915 0000:00:02.0: [drm] GuC SLPC: disabled
[    4.198487] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc_7.9.3.bin version 7.9 authenticated:yes
[    4.200110] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    4.201143] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    4.201282] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes

$ apt-cache show libva2 | grep Version
Version: 2.14.0-1
$ apt-cache show libigdgmm12 | grep Version
Version: 22.1.2+ds1-1
$ apt-cache show vainfo | grep Version
Version: 2.12.0+ds1-1
$ apt-cache show intel-media-va-driver | grep Version
Version: 22.3.1+dfsg1-1ubuntu1
$ apt-cache show intel-media-va-driver-non-free | grep Version
Version: 22.3.1+ds1-1

$ sudo vainfo
error: XDG_RUNTIME_DIR not set in the environment.
error: can't connect to X server!
libva info: VA-API version 1.14.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_14
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.14 (libva 2.12.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.3.1 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
...
```

You definitely have later kernel and user space stack... I will try on later one on my side.

Can you help to collect output of `strace vainfo`?

From you dmesg log, this line somewhat bothers, thought later on I don't see errors from i915:
```
[ 2.087228] i915 0000:00:02.0: [drm] GuC firmware i915/tgl_guc_70.1.1.bin is recommended, but only i915/tgl_guc_69.0.3.bin was found
[ 2.087229] i915 0000:00:02.0: [drm] Consider updating your linux-firmware pkg or downloading from
```
Did you try to update linux-firmware and regenerate ramfs?

Brampie66, 2022-08-27:
Hi,

I have the non K model, don't know if that makes a difference.

cat /proc/cpuinfo  | grep name
model name	: 12th Gen Intel(R) Core(TM) i7-12700

I updated the firmware, dmesg is now:
[    2.080940] i915 0000:00:02.0: vgaarb: deactivate vga console
[    2.081593] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    2.082057] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adls_dmc_ver2_01.bin (v2.1)
[    2.084650] i915 0000:00:02.0: [drm] GuC firmware i915/tgl_guc_70.1.1.bin version 70.1
[    2.084652] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc_7.9.3.bin version 7.9
[    2.098322] i915 0000:00:02.0: [drm] HuC authenticated
[    2.098331] i915 0000:00:02.0: [drm] GuC submission disabled
[    2.098334] i915 0000:00:02.0: [drm] GuC SLPC disabled

packages:
libva.x86_64                                                                                                                 1:2.15.0-1.el8                                        
libva-utils.x86_64                                                                                                           1:2.15.0-1.el8 
intel-gmmlib.x86_64                                                                                                       1:22.1.7-1.el8
intel-media-driver.x86_64                                                                                                  22.5.3-1.el8

xorg.0.log
[    20.151] (II) modeset(0): glamor X acceleration enabled on Mesa Intel(R) Graphics (ADL-S GT1)
[    20.151] (II) modeset(0): glamor initialized

strace
write(2, "libva info: Found init function "..., 52libva info: Found init function __vaDriverInit_1_15
) = 52
openat(AT_FDCWD, "/etc/igfx_user_feature_next.txt", O_RDONLY) = 5
read(5, "[config]\n\n[report]\nOCA Status=0\n"..., 8191) = 33
read(5, "", 8191)                       = 0
close(5)                                = 0
openat(AT_FDCWD, "/etc/igfx_user_feature.txt", O_RDONLY) = -1 ENOENT (No such file or directory)
getpid()                                = 5055
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332359b0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332359b0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332359b0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332359b0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332359b0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332359b0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332359b0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332359b0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332359b0) = 0
mmap(NULL, 278528, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f71bf2f1000
ioctl(4, DRM_IOCTL_I915_GEM_GET_APERTURE, 0x7fff332351b0) = 0
geteuid()                               = 1000
getuid()                                = 1000
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff33235200) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332351c0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332351c0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332351c0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332351c0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332351c0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332351c0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332351c0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332351c0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332351c0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332351c0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332351c0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332351c0) = 0
ioctl(4, DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM, 0x7fff332351e0) = 0
ioctl(4, DRM_IOCTL_I915_QUERY, 0x7fff332351d0) = 0
ioctl(4, DRM_IOCTL_I915_QUERY, 0x7fff332351d0) = 0
ioctl(4, DRM_IOCTL_I915_QUERY, 0x7fff332351d0) = 0
ioctl(4, DRM_IOCTL_I915_QUERY, 0x7fff332351d0) = 0
ioctl(4, DRM_IOCTL_I915_QUERY, 0x7fff33235030) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332351b0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332351b0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332351b0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332351b0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332351b0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332351b0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332351b0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332351b0) = 0
ioctl(4, DRM_IOCTL_I915_GETPARAM, 0x7fff332351b0) = 0
ioctl(4, DRM_IOCTL_I915_QUERY, 0x7fff33235190) = 0
ioctl(4, DRM_IOCTL_I915_QUERY, 0x7fff33235190) = 0
write(2, "libva error: /usr/lib64/dri/iHD_"..., 57libva error: /usr/lib64/dri/iHD_drv_video.so init failed
) = 57
write(2, "libva info: va_openDriver() retu"..., 38libva info: va_openDriver() returns 1
) = 38
write(2, "vaInitialize failed with error c"..., 62vaInitialize failed with error code 1 (operation failed),exit
) = 62
close(4)                                = 0

thx
Bram

dvrogozh, 2022-08-27:
To completely wave away kernel setup issue you can try to run vainfo under ubuntu 22,04 docker. Something like this:
```
$ docker pull ubuntu:22.04
$ docker run -it --rm --privileged ubuntu:22.04 # don't use --privileged under production environment, but for quick try it should do
> apt-get update
> apt-get install intel-media-va-driver libva-drm2 vainfo
> vainfo
```

> strace

Thank you for the log. Problem is that I don't see any issue in it except that libva ultimately failed to initialize. (/etc/igfx_user_feature.txt don't count - that's just optional driver settings file, it's ok if it's not exist). At the moment the best I can think of is that something is wrong with the driver build. Can you highlight what's your system setup? If you will explain how you've configured system and especially user space - we can try to replicate on our side.

Brampie66, 2022-08-27:
Hi,

I'll try on different hardware and see if I can add some more info.
I have device running v22 on a J5005 and an i3-9100.

thx

Brampie66, 2022-08-28:
Hi,

Using the docker method it does work
root@d3a4debeea55:/# vainfo
error: XDG_RUNTIME_DIR not set in the environment.
error: can't connect to X server!
libva info: VA-API version 1.14.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_14
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.14 (libva 2.12.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.3.1 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats

Brampie66, 2022-08-28:
Hi,

I tried the ssd in a gigabyte brix with a J5005 and here it does work.
libva info: VA-API version 1.15.0
libva info: Trying to open /usr/lib64/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_15
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.15 (libva 2.15.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.5.3 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats

dmesg
[    3.547331] i915 0000:00:02.0: vgaarb: deactivate vga console
[    3.547485] i915 0000:00:02.0: [drm] couldn't get memory information
[    3.547920] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    3.549030] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/glk_dmc_ver1_04.bin (v1.4)

thx

Brampie66, 2022-08-30:
Hi,

I did some more testing,
the drive fails to work on an i3-12100 but does work on an i3-10105.

thx

victornoel, 2022-10-24:
I'm having the same issue with on linux 6.0.2 and intel-media-driver 22.4.4 on archlinux and the following chipset:
```
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-UP4 GT2 [Iris Xe Graphics] (rev 0c)
```

The only particular thing I did was adding those options to the i915 module:
```
options i915 enable_guc=3 enable_fbc=1
```

Here are the kernel logs:
```
➜  ~ sudo dmesg | grep drm        

[    1.278140] ACPI: bus type drm_connector registered
[    1.493164] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    1.493289] i915 0000:00:02.0: [drm] Using Transparent Hugepages
[    1.498541] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/adlp_dmc_ver2_16.bin (v2.16)
[    1.637748] i915 0000:00:02.0: [drm] GuC firmware i915/adlp_guc_70.1.1.bin version 70.1
[    1.637753] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc_7.9.3.bin version 7.9
[    1.660164] i915 0000:00:02.0: [drm] HuC authenticated
[    1.661223] i915 0000:00:02.0: [drm] GuC submission enabled
[    1.661224] i915 0000:00:02.0: [drm] GuC SLPC enabled
[    1.661985] i915 0000:00:02.0: [drm] GuC RC: enabled
[    1.665428] i915 0000:00:02.0: [drm] Protected Xe Path (PXP) protected content support initialized
[    2.811009] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    2.826523] fbcon: i915drmfb (fb0) is primary device
[    2.861338] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[   78.240127] systemd[1]: Starting Load Kernel Module drm...
[   78.250796] systemd[1]: modprobe@drm.service: Deactivated successfully.
[   78.250897] systemd[1]: Finished Load Kernel Module drm.
[   88.639851] i915 0000:00:02.0: [drm] Selective fetch area calculation failed in pipe A
```

Not sure what the last line is about.

Is there anything I can do to help moving forward with this issue?

Nexarian, 2022-11-06:
Seeing something similar on my ` 12th Gen Intel(R) Core(TM) i9-12950HX` (from `cat /proc/cpuinfo`)

Seen on a fresh install of Ubuntu 22.04.

Tried to build the latest relevant packages myself and got this:

```
write(2, "libva info: Trying to open /usr/"..., 81libva info: Trying to open /usr/local/lib/x86_64-linux-gnu/dri/i965_drv_video.so
) = 81
openat(AT_FDCWD, "/usr/local/lib/x86_64-linux-gnu/dri/i965_drv_video.so", O_RDONLY|O_CLOEXEC) = 4
read(4, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(4, "", {st_mode=S_IFREG|0755, st_size=13167576, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 8164888, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 4, 0) = 0x7f078a000000
mmap(0x7f078a011000, 884736, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 4, 0x11000) = 0x7f078a011000
mmap(0x7f078a0e9000, 7184384, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 4, 0xe9000) = 0x7f078a0e9000
mmap(0x7f078a7c3000, 28672, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 4, 0x7c2000) = 0x7f078a7c3000
close(4)                                = 0
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 4
newfstatat(4, "", {st_mode=S_IFREG|0644, st_size=68791, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 68791, PROT_READ, MAP_PRIVATE, 4, 0) = 0x7f078db76000
close(4)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libdrm_intel.so.1", O_RDONLY|O_CLOEXEC) = 4
read(4, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(4, "", {st_mode=S_IFREG|0644, st_size=151496, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 153512, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 4, 0) = 0x7f078d841000
mmap(0x7f078d844000, 94208, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 4, 0x3000) = 0x7f078d844000
mmap(0x7f078d85b000, 40960, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 4, 0x1a000) = 0x7f078d85b000
mmap(0x7f078d865000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 4, 0x23000) = 0x7f078d865000
close(4)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libpciaccess.so.0", O_RDONLY|O_CLOEXEC) = 4
read(4, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(4, "", {st_mode=S_IFREG|0644, st_size=39424, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 41360, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 4, 0) = 0x7f078d836000
mmap(0x7f078d838000, 20480, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 4, 0x2000) = 0x7f078d838000
mmap(0x7f078d83d000, 8192, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 4, 0x7000) = 0x7f078d83d000
mmap(0x7f078d83f000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 4, 0x8000) = 0x7f078d83f000
close(4)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libz.so.1", O_RDONLY|O_CLOEXEC) = 4
read(4, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(4, "", {st_mode=S_IFREG|0644, st_size=108936, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 110776, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 4, 0) = 0x7f078d5e4000
mprotect(0x7f078d5e6000, 98304, PROT_NONE) = 0
mmap(0x7f078d5e6000, 69632, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 4, 0x2000) = 0x7f078d5e6000
mmap(0x7f078d5f7000, 24576, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 4, 0x13000) = 0x7f078d5f7000
mmap(0x7f078d5fe000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 4, 0x19000) = 0x7f078d5fe000
close(4)                                = 0
mprotect(0x7f078d5fe000, 4096, PROT_READ) = 0
mprotect(0x7f078d83f000, 4096, PROT_READ) = 0
mprotect(0x7f078d865000, 4096, PROT_READ) = 0
mprotect(0x7f078a7c3000, 4096, PROT_READ) = 0
munmap(0x7f078db76000, 68791)           = 0
write(2, "libva info: Found init function "..., 52libva info: Found init function __vaDriverInit_1_17
) = 52
ioctl(3, DRM_IOCTL_I915_GEM_GET_APERTURE, 0x7ffc9e55c760) = 0
geteuid()                               = 1000
getuid()                                = 1000
ioctl(3, DRM_IOCTL_I915_GETPARAM, 0x7ffc9e55c770) = 0
ioctl(3, DRM_IOCTL_I915_GETPARAM, 0x7ffc9e55c770) = 0
ioctl(3, DRM_IOCTL_I915_GETPARAM, 0x7ffc9e55c770) = 0
ioctl(3, DRM_IOCTL_I915_GETPARAM, 0x7ffc9e55c770) = 0
ioctl(3, DRM_IOCTL_I915_GETPARAM, 0x7ffc9e55c770) = 0
ioctl(3, DRM_IOCTL_I915_GETPARAM, 0x7ffc9e55c770) = 0
ioctl(3, DRM_IOCTL_I915_GETPARAM, 0x7ffc9e55c770) = 0
ioctl(3, DRM_IOCTL_I915_GETPARAM, 0x7ffc9e55c770) = 0
ioctl(3, DRM_IOCTL_I915_GETPARAM, 0x7ffc9e55c770) = 0
ioctl(3, DRM_IOCTL_I915_GETPARAM, 0x7ffc9e55c770) = 0
ioctl(3, DRM_IOCTL_I915_GETPARAM, 0x7ffc9e55c770) = 0
write(2, "libva error: /usr/local/lib/x86_"..., 79libva error: /usr/local/lib/x86_64-linux-gnu/dri/i965_drv_video.so init failed
) = 79
write(2, "libva info: va_openDriver() retu"..., 39libva info: va_openDriver() returns -1
) = 39
write(2, "vaInitialize failed with error c"..., 66vaInitialize failed with error code -1 (unknown libva error),exit
) = 66
close(3)                                = 0
exit_group(3)                           = ?
+++ exited with 3 +++
```

Nexarian, 2022-11-11:
Got it working! Submitted a patch: https://github.com/intel/media-driver/pull/1544

The fix is that not all PCI IDs are registered yet. I found a bug where this was fixed in MESA awhile ago. I'm not sure if it'll fix other's use of this driver, but it fixed mine with the latest builds.

```
/media-driver$ vainfo --device /dev/dri/renderD128 --display drm
libva info: VA-API version 1.17.0
libva info: Trying to open /usr/local/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_17
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.17 (libva 2.12.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.6.2 (f38f67cf6)
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointFEI
      VAProfileHEVCMain               :	VAEntrypointEncSliceLP
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSlice
      VAProfileHEVCMain10             :	VAEntrypointEncSliceLP
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointEncSliceLP
      VAProfileVP9Profile1            :	VAEntrypointVLD
      VAProfileVP9Profile1            :	VAEntrypointEncSliceLP
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointEncSliceLP
      VAProfileVP9Profile3            :	VAEntrypointVLD
      VAProfileVP9Profile3            :	VAEntrypointEncSliceLP
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointEncSlice
      VAProfileHEVCMain422_10         :	VAEntrypointVLD
      VAProfileHEVCMain422_10         :	VAEntrypointEncSlice
      VAProfileHEVCMain422_12         :	VAEntrypointVLD
      VAProfileHEVCMain422_12         :	VAEntrypointEncSlice
      VAProfileHEVCMain444            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         :	VAEntrypointVLD
      VAProfileHEVCMain444_10         :	VAEntrypointEncSliceLP
      VAProfileHEVCMain444_12         :	VAEntrypointVLD
      VAProfileHEVCSccMain            :	VAEntrypointVLD
      VAProfileHEVCSccMain            :	VAEntrypointEncSliceLP
      VAProfileHEVCSccMain10          :	VAEntrypointVLD
      VAProfileHEVCSccMain10          :	VAEntrypointEncSliceLP
      VAProfileHEVCSccMain444         :	VAEntrypointVLD
      VAProfileHEVCSccMain444         :	VAEntrypointEncSliceLP
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCSccMain444_10      :	VAEntrypointVLD
      VAProfileHEVCSccMain444_10      :	VAEntrypointEncSliceLP
```

victornoel, 2022-11-16:
@Nexarian any idea how I can check what's the device id on my computer to see if it's the same you added or some more are missing?

victornoel, 2022-11-17:
@Nexarian thanks, it looks like my device is in there, but I just realized I am using 22.4.4 of intel media driver so I suspect I should first try the latest version as you recommend :)

For the record:
```
➜  ~ lspci -nn |grep -Ei 'VGA|DISPLAY'
00:02.0 VGA compatible controller [0300]: Intel Corporation Alder Lake-UP4 GT2 [Iris Xe Graphics] [8086:46aa] (rev 0c)
```

victornoel, 2022-11-17:
It indeed works well for the Dell XPS 9315 with Alder Lake :) (something I could have know earlier if I had read a bit more the first comments at the top :P)

```
➜  trunk git:(packages/intel-media-driver) ✗ vainfo 
Trying display: wayland
Trying display: x11
vainfo: VA-API version: 1.16 (libva 2.16.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.6.2 (98ca3f4)
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointFEI
      VAProfileHEVCMain               :	VAEntrypointEncSliceLP
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSlice
      VAProfileHEVCMain10             :	VAEntrypointEncSliceLP
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointEncSliceLP
      VAProfileVP9Profile1            :	VAEntrypointVLD
      VAProfileVP9Profile1            :	VAEntrypointEncSliceLP
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointEncSliceLP
      VAProfileVP9Profile3            :	VAEntrypointVLD
      VAProfileVP9Profile3            :	VAEntrypointEncSliceLP
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointEncSlice
      VAProfileHEVCMain422_10         :	VAEntrypointVLD
      VAProfileHEVCMain422_10         :	VAEntrypointEncSlice
      VAProfileHEVCMain422_12         :	VAEntrypointVLD
      VAProfileHEVCMain422_12         :	VAEntrypointEncSlice
      VAProfileHEVCMain444            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         :	VAEntrypointVLD
      VAProfileHEVCMain444_10         :	VAEntrypointEncSliceLP
      VAProfileHEVCMain444_12         :	VAEntrypointVLD
      VAProfileHEVCSccMain            :	VAEntrypointVLD
      VAProfileHEVCSccMain            :	VAEntrypointEncSliceLP
      VAProfileHEVCSccMain10          :	VAEntrypointVLD
      VAProfileHEVCSccMain10          :	VAEntrypointEncSliceLP
      VAProfileHEVCSccMain444         :	VAEntrypointVLD
      VAProfileHEVCSccMain444         :	VAEntrypointEncSliceLP
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCSccMain444_10      :	VAEntrypointVLD
      VAProfileHEVCSccMain444_10      :	VAEntrypointEncSliceLP
```

Nexarian, 2022-11-17:
@Brampie66 What about you? Were you ever able to resolve the reason you created this issue?

ReshaSD, 2022-12-23:
Hey guys, I see you solved the issue. Can you help me with the same. Sorry, I new on Linux and didn't understand what you've exactly done to solve it.. Here is some info of my system:

```
$ sudo vainfo
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
libva info: VA-API version 1.15.0
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
vaInitialize failed with error code -1 (unknown libva error),exit

$ vainfo
libva info: VA-API version 1.15.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_14
libva error: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 1
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_10
libva error: /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so init failed
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit

$ cat /proc/cpuinfo  | grep name
model name	: 12th Gen Intel(R) Core(TM) i9-12900HX
model name	: 12th Gen Intel(R) Core(TM) i9-12900HX
model name	: 12th Gen Intel(R) Core(TM) i9-12900HX
model name	: 12th Gen Intel(R) Core(TM) i9-12900HX
model name	: 12th Gen Intel(R) Core(TM) i9-12900HX
model name	: 12th Gen Intel(R) Core(TM) i9-12900HX
model name	: 12th Gen Intel(R) Core(TM) i9-12900HX
model name	: 12th Gen Intel(R) Core(TM) i9-12900HX
model name	: 12th Gen Intel(R) Core(TM) i9-12900HX
model name	: 12th Gen Intel(R) Core(TM) i9-12900HX
model name	: 12th Gen Intel(R) Core(TM) i9-12900HX
model name	: 12th Gen Intel(R) Core(TM) i9-12900HX
model name	: 12th Gen Intel(R) Core(TM) i9-12900HX
model name	: 12th Gen Intel(R) Core(TM) i9-12900HX
model name	: 12th Gen Intel(R) Core(TM) i9-12900HX
model name	: 12th Gen Intel(R) Core(TM) i9-12900HX
model name	: 12th Gen Intel(R) Core(TM) i9-12900HX
model name	: 12th Gen Intel(R) Core(TM) i9-12900HX
model name	: 12th Gen Intel(R) Core(TM) i9-12900HX
model name	: 12th Gen Intel(R) Core(TM) i9-12900HX
model name	: 12th Gen Intel(R) Core(TM) i9-12900HX
model name	: 12th Gen Intel(R) Core(TM) i9-12900HX
model name	: 12th Gen Intel(R) Core(TM) i9-12900HX
model name	: 12th Gen Intel(R) Core(TM) i9-12900HX

$ uname -a
Linux stas-ubuntu 5.19.0-26-generic #27-Ubuntu SMP PREEMPT_DYNAMIC Wed Nov 23 20:44:15 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

$ lspci -nn |grep  -Ei 'VGA|DISPLAY'
00:02.0 VGA compatible controller [0300]: Intel Corporation Alder Lake-HX GT1 [UHD Graphics 770] [8086:4688] (rev 0c)
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GA104 [Geforce RTX 3070 Ti Laptop GPU] [10de:24a0] (rev a1)

$ apt-cache show libva2 | grep Version
Version: 2.15.0-1

$ apt-cache show libigdgmm12 | grep Version
Version: 22.2.0+ds1-1

$ apt-cache show vainfo | grep Version
Version: 2.12.0+ds1-1

$ apt-cache show intel-media-va-driver | grep Version
Version: 22.4.3+dfsg1-1

$ apt-cache show intel-media-va-driver-non-free | grep Version
Version: 22.4.3+ds1-1
```
cc @victornoel @Nexarian @Brampie66 @dvrogozh

victornoel, 2022-12-25:
@ReshaSD we "simply" manually built the latest version (tag 22.6.4) of this repository (and I suspect you will also need latest version of https://github.com/intel/gmmlib/, 22.3.2) and installed it on our system. This is not very complex but each distribution will have its own way of doing it cleanly.

Note that on Archlinux, finally the latest version 22.6.4 is now available in the package manager.

You can check if packages for the latest version exist for your distribution. All of this is a bit out of scope of this discussion though, so I would recommend either waiting for the latest version be available in your distribution or investigate how to build the driver manually by yourself. See the README of the repository: https://github.com/intel/media-driver#building

ReshaSD, 2022-12-26:
@victornoel I installed libva 2.16.0 and latest tags from gmmlib and media-driver.
And it helped. Thanks.

paulmenzel, 2022-12-26:
@ReshaSD, it’d be great if you reported the issue to your distribution (Ubuntu?), so they can backport the patches.

woodrow-shen, 2023-01-15:
My Dell xps 9315 also works fine with `vainfo` after manually building libva, gmmlib, media-driver, however, when playing the HD/4K video via Youtube, either on chrome or firefox, I can't see workloads on Video engine from `intel_gpu_top`'s monitor, even I built the intel-gpu-tools from scratch. Have anyone verified if the video decoder really works by checking `intel_gpu_top` on Alder Lake? Thanks.

paulmenzel, 2023-01-15:
@woodrow-shen, is the video engine used, when using mpv or another video player? Anyway, please create a separate issue, and you can check VAAPI things with Mozilla Firefox with the logging options below:

    MOZ_LOG="PlatformDecoderModule:5" nightly

woodrow-shen, 2023-01-15:
@paulmenzel Thanks for reply, fix my words, I can confirm that vaapi can work on firefox and meanwhile I can see the workload from intel_gpu_top as well. Great work.

victornoel, 2023-01-26:
@selfdrivingmenee hello, I have the same laptop and as discussed in the comments above, this works as expected with the latest release version. Multiple people confirmed this, at least for the present issue.

You may be hitting a problem of configuration or maybe not using the latest version. In any case, if you think this is a big, the best is to open a new issue with details of the versions of libva, intel-media-driver and gmmlib :)

selfdrivingmenee, 2023-01-27:
@victornoel Ciao Victor, first of all thank you since I managed to get it to work by manually building it following your advice from your previous post -  I can confirm that the GPU is now working perfectly fine on my XPS13 9315!

Seems then that the issue is not in the driver itself but somewhere in the "distribution chain" for the Dell-provided oem Ubuntu distro. I contacted them to inform of the issue and see if it's only on our side or it's affecting all users with same computer.

victornoel, 2023-01-27:
@selfdrivingmenee hehe, yes, I looked quickly at their ubuntu oem version and was a bit scared by all the hacks they have in there to make things work. I really find it unacceptable that they sell a laptop saying it's linux-certified but in the end requires so much hacking around, either by ubuntu people or manually as in this issue… this is a bit sad :)

I myself am on archlinux and still need to hack around to get the webcam working, all the rest (audio and video mainly) seems ok now.

santomet, 2023-03-09:
Hello guys,
Just bought Dell 7770 with i7-12850HX and I have exactly the same issue. The GPU ID is also `4688`.
However, I got rid of OEM Ubuntu and Installed clean Linux Mint 21.1. 
I do not have problems with Alder Lake GPUs on any other device. Just on this Dell.

So if I understand correctly, it is a Dell-specific problem?

LWLLR, 2023-04-30:
Description:
I'm using Debian12 with an Intel n100 CPU and the following system information:
```
$ cat /etc/debian_version
12.0

$ uname -r
6.1.0-7-amd64

$ lspci -nn |grep -Ei 'VGA|DISPLAY'
00:02.0 VGA compatible controller [0300]: Intel Corporation Alder Lake-N [UHD Graphics] [8086:46d1]

$ ls -l /dev/dri
by-path
card0
renderD128
```

When I run the command `vainfo --display drm --device /dev/dri/renderD128`, the output shows:
```ssh
Trying display: drm
libva info: VA-API version 1.19.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_19
libva error: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 1
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_8
libva error: /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so init failed
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit
```
I would appreciate any help in resolving this issue.

eero-t, 2023-07-04:
> So if I understand correctly, it is a Dell-specific problem?

Initial problem was that driver itself was missing GPU HW IDs for some of the newer ADL models, so it did not recognize the iGPU.  After driver was updated, only problem people had was their Linux distributions shipping too old driver version.  This apparently included Ubuntu version pre-installed on given DELL laptop.

If you're using LTS version of Ubuntu or its derivative, you can install so called HWE (HW enabling) meta package that pulls in newer driver versions.   That provides drivers from newer Ubuntu non-LTS releases, after they've been tested few months in those newer distro versions.

(You can see the available HWE driver packages with `apt search hwe`.)

Maybe MiNT has something similar?

If that does help i.e. current HWE packages do not yet provide new enough drivers, you may need to wait until next Ubuntu release + few months, for  HWE packages in LTS distro to be updated.

PS. While one could always build & install drivers from sources, I would recommend that only for experienced Linux users, or somebody wanting to learn more about debugging. Things installed outside of the distro's own repositories have a tendency of creating problems _much later_, when other packages get updated (and you do not any more remember installing things from other sources).

closed 2024-03-26 by Sherry-Lin