intel/media-driver #1373: [TGL] Hang during decoding H265 corrupted stream with change resolution

Fantast34, 2022-03-29:
## System information
- OS: Debian11, kernel 5.10.0-10-amd64
- libva : dad7abcb958b97b50c318dd42a477e035a42e358
- gmmlib: 29152fc6d39f87983b176664f225d42af039f716
- libva-utils: ec1c44c19a05cdb7006e5468529c71a4fb7a3065
- media-driver: f33d756c2de3174c08fe0e10d8742134b8721f57
- MediaSDK: cf544528845a95bf021ca9334a1611be56ad7b97
- oneVPL: 9b659b0373118e4f9410544aed1689ecc2a2a4e0
- oneVPL-intel-gpu: 01a2422f1aa716456012ef03b32da6aa5d5fc511
- Good: KabyLake system with HD Graphics 630 (rev.4)
- Bad: TigerLake system with Iris XE Graphicx (rev.01)

cmd: sample_decode h265 -i input.h265 -o out.yuv

## Issue behavior
  On Tigerlake system decoder decodes problematic bitstream until corrupted frame (before change resolution) and hang for about 10 seconds. Behavior is the same with MediaSDK and oneVPL sdk. This is very critical delay for our customer. 
  On Kabylake system there is no Hang, decoder reports corrupted frames and continue decoding

### Describe the expected behavior
No Hang
[video.tar.gz](https://github.com/intel/media-driver/files/8372972/video.tar.gz)
[tgl_legacy.log](https://github.com/intel/media-driver/files/8372978/tgl_legacy.log)
[tgl_gen.log](https://github.com/intel/media-driver/files/8372979/tgl_gen.log)
[kabylake_log.txt](https://github.com/intel/media-driver/files/8372996/kabylake_log.txt)
g

Jexu, 2022-03-30:
Normally, iHD/ i915 driver will reset the ctx and recover the decoding within seconds if occuring hang. But 10s seems abnormal.

May i know how many problematic frames in your bitstream and are they in same GOP. Successive error frame decoding may lead above latency.

Fantast34, 2022-03-30:
In our customer application we wait 3-4 times in loop with 3000ms sync.timeout with MFX_WRN_IN_EXECUTION return code, and only after that driver returns GPU_HANG code.

We have several problematic frames in the stream, after each IDR frame with changed resolution placed, so there is only one problematic frame in the GOP. I have attached elementary h265 stream to this issue - video.tar.gz

Fantast34, 2022-03-30:
All logs created by the same version of sample_decode from intel oneVPL repository. As I understand Incopatible video parameters caused by resolution change in encoded stream

Jexu, 2022-03-30:
I know, i will take a look firstly.

Fantast34, 2022-04-13:
Hi Jexu,
have you any updates?

Jexu, 2022-04-13:
We are plan to enhance the error concealment by detecking invalid syntax vs hevc spec and error bistream data. By skiping error frame decoding to avoid hang happen. It is in progress.

Fantast34, 2022-05-07:
Hi Jexu,
Can you told any dates for this issue? It is a blocker for our customer release.

guowillyy, 2022-05-11:
It is an error clip handling issue , only happened on TGL, no issue from Alchemist. So far no plan to fix it on TGL.

Jexu, 2022-05-18:
Your problematic bitstream misses some slice data while slice data size in slice header also equals to remaining slice data. In this case, no matter app or media driver cann't detect the error to skip the decoding. Normally, the gpu can handle this error, but tgl has limitation to do this. The hw has improved it from later generation.
So, we trend to close this issue now.

Jexu, 2022-05-26:
I close this now, feel free to re-open again if having any other issues.

closed 2022-05-26 by Jexu