intel/media-driver #1374: vaRenderPicture fails on multiple launches

pbochenk, 2022-03-30:
Hi, 
I’m from DL Streamer team. We develop libva element which uses vaRenderPicture function, and we ran into some problems.

## Issue behavior
On Intel GPU UHD 630 If I run the vaRenderPicture with batch **1** more then **65** times, surface count will be accumulating for existing context till **65**, and vaRenderPicture returns error with “_**list argument exceeds maximum number**_” message and **VA_STATUS_ERROR_MAX_NUM_EXCEEDED** status.

Is this behavior expected? If yes, how to perform more than 65 operations with the same context?

## System information
media-driver: [intel-media-21.4.1](https://github.com/intel/media-driver/tree/intel-media-21.4.1)
OS: Ubuntu 20.04
GPU: Intel GPU UHD 630

LhGu, 2022-11-25:
@pbochenk  We could not support more than 65 input source in the same vpp context. We think 65 sources in one vpp context is the worst condition in vpp. You could create other vpp context for other stream.

LhGu, 2022-11-28:
Close it now, and please feel free to re-open this issue if you want to discuss more.

closed 2022-11-28 by LhGu