intel/media-driver #1375: Clarification on the releases versionning

kwizart, 2022-04-01:
As a (Fedora) package maintainer, I'm asking for clarification of the release versioning. 

intel-media-driver has started the 22.3 branch which should stand for 3rd quarter release of intel-media driver but intel-gmmlib is still updated at 22.1.x. 

Does the quarter release still hold ? It's difficult to follow between released and target quarter release date ...
(specially targetting f36 and at least 22.2 (pre-)release if nearly pending).

Thanks for any insight.

Sherry-Lin, 2022-04-14:
Thanks for raising this question.
Media version naming version is Year.minor.micro.
Minor version would be bumped up when
1. New quarter as you mentioned
2. New Key platform upstream
3. New Key feature upstream
Micro version would be bumped up for each release including bi-weekly rolling release and quarterly release.

https://github.com/intel/media-driver/releases shows all quarterly release.

Sherry-Lin, 2023-05-18:
You could check the detail information in https://github.com/intel/media-driver/wiki#media-release-model. I'm going to close this issue. Please feel free to re-open this issue if any concerns. Thanks.

closed 2023-05-18 by Sherry-Lin