intel/media-driver #1377: Cannot create UYVY using vaCreateSurfaces (missing a few lines to have support it seems)

MathieuGirard, 2022-04-04:
The switch in the following is missing:
```
case VA_FOURCC_UYVY:
  expected_fourcc = VA_FOURCC_UYVY;
  break;
```

https://github.com/intel/media-driver/blob/c77763f837f3014925263748ae46111b9a86f75a/media_driver/linux/common/ddi/media_libva.cpp#L2589-L2732

I tested adding the lines and the rest of the code seems to work.

FurongZhang, 2022-04-04:
We can add the code as you mentioned. But, may I know what the usage is, here? Need to check whether our codec, vpp can support this.

MathieuGirard, 2022-04-05:
Hi,
According to the nedia-driver documentation the UYVY is supposed to be supported as input for AVC encoding and video processing (VEBOX) (see https://github.com/intel/media-driver/blob/master/docs/media_features.md).

I tested both use cases:
- OneVPL AVC does not accept an UYVY input (invalid video parameters). I tried the 4 possible combinaison for memory types (video, system) and low power.
- Video processing context is sucessfully created with a "In" with a UYVY FourCC and a "Out" with a NV12 FOURCC. We were attempting to create the input surface to further test it.

The setup was a NUC (11th Gen i7-1165G7) ,with an Irix Xe graphics (9a49) on OneVPL (2022.0.0) with a 5.12.9 Linux kernel (Ubuntu 20.04).

Regards,
Mathieu

MathieuGirard, 2022-04-05:
Addadum: We were also looking into using DRM to display UYVY surfaces. Here again the basic support was looking good (could enumerate de format from DRM) but surface creation support was missing.

LhGu, 2022-04-06:
We need to check which features support this UYVY format.

MathieuGirard, 2022-04-12:
On tigerlake:
- the AVC refused UYVY input at context creation.
- the video processing engine accepted UYVY at context creation and simple tests generated the expected colors.
- DRM seemed to be able to enumerate UYVY color format with the screen we used to test.

Regards

MathieuGirard, 2022-04-12:
According to the documentation JPEG encoding should be possible with a UYVY output (https://github.com/intel/media-driver/blob/master/docs/media_features.md#jpeg-decoding-format-support).

But we did not had the environment to test this up with OneVPL 2022.0.0.

LhGu, 2022-04-25:
Hi， @MathieuGirard . As I know, you don't need to add any code in media_driver to allow UYVY surface. The code location you add, just some work around code for some tests, we can infer from the code annotation(**#if 1 //added for having MDF sanity test pass, will be removed after MDF formal patch checked in** ).  
I think that it actually just support  VA_RT_FORMAT_***, not VA_FOURCC_*** for real usage.

If you want to make **expected_fourcc** to be VA_FOURCC_UYVY, you can just pass down  VASurfaceAttrib    *attrib_list in the 
vaCreateSurfaces.  some sample code as below:
    VASurfaceAttrib    surface_attrib;
    surface_attrib.type =  VASurfaceAttribPixelFormat;
    surface_attrib.flags = VA_SURFACE_ATTRIB_SETTABLE;
    surface_attrib.value.type = VAGenericValueTypeInteger;
    surface_attrib.value.value.i = VA_FOURCC_UYVY;

    va_status = vaCreateSurfaces(va_dpy,
                                 format,
                                 width,
                                 height,
                                 p_surface_id,
                                 1,
                                 &surface_attrib,
                                 1);

You can just use VA_RT_FORMAT_YUV422 for var format, and this **surface_attrib.value.value.i**    value will overwrite  **expected_fourcc** in the follwing code.

LhGu, 2022-04-25:
In addition, I know that this format just support input  surface and platform should be more than ICL.  VPP just support Color Format Conversion, Scaling, Blending, Detail(Sharpness), Mirror/Rotation.

MathieuGirard, 2022-04-26:
Passing the correct RT format seems to work. We already past the attribute. VPP seems to do color convention.

Thanks,
Mathieu

closed 2022-04-26 by MathieuGirard

FurongZhang, 2022-04-27:
Thanks so much @LhGu and @MathieuGirard to close this!